networkupstools/nut #551: Who suppose to run `nut-monitor poweroff` on modern debian dists?

zapadinsky, 2018-04-25:
The call to `init.d/halt` was removed from sysvinit scripts before 2005 (see the [shutdown.c file in repo](http://git.savannah.nongnu.org/cgit/sysvinit.git/tree/src/shutdown.c#n301)). The recent Ubuntu 17.10 dist doesn't have `init.d/halt` at all. The `nut-monitor poweroff` (i.e. `nut-client poweroff` and `nut-server poweroff`) was called from `init.d/halt`, so it is not called during shutdown procedure now. As the result `upsdrvctl shutdown` is not executed and UPS doesn't power off after shutdown.

zapadinsky, 2018-04-26:
Why isn't `scripts/systemd/nutshutdown` just call `ups-server poweroff`and has a separate branch of code doing the same thing instead. `scripts/systemd/nutshutdown` was created in 2011 but not included in any packages older than 2.7.4-5.1. The UPS shutdown functionality is broken for years and may only be fixed in very recent not yet released Debian 10 and Ubuntu 18.

jimklimov, 2018-04-26:
To make it clear, are you asking NUT community about why did the Debian and Ubuntu communities choose to do something in a particular way? 

No idea, can only make some guesses ;)

Probably stems from linux migration from init scripts to systemd, for starters...

zapadinsky, 2018-04-26:
The integration with systemd is a part of system tool development, just like drivers etc. The deb package maintainer and the author of `nutshutdown` script @aquette is also the most active committer here. I don't know how this bug remained unnoticed for so long.