networkupstools/nut #552: [HCL] CyberPower BR1000ELCD supported by usbhid-ups

Amunak, 2018-04-30:
After some tinkering with udev rules to assign correct permissions I managed to get my UPS working with NUT.

It's CyberPower [BR1000ELCD](https://www.cyberpower.com/eu/en/product/sku/BR1000ELCD). There are other pretty much identical models (with only different capacity / max load; different outlets or different certifications) that most likely have the exact same capabilities: models `BR(700|1000|1200)ELCD` and the same with `-FR` at the end.

upsc output:
```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 7200
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 12.9
battery.voltage.nominal: 12
device.mfr: CPS
device.model: BR1000ELCD
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: BR1000ELCD
driver.parameter.productid: 0501
driver.parameter.synchronous: no
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 0
input.transfer.low: 0
input.voltage: 238.0
input.voltage.nominal: 230
output.voltage: 239.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: CPS
ups.model: BR1000ELCD
ups.productid: 0501
ups.realpower.nominal: 600
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

upsrw output:
```
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

[input.transfer.high]
High voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 0

[input.transfer.low]
Low voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 0

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30
```
upscmd output:
```
Instant commands supported on UPS [nutdev1]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

I did not have a chance to test it yet.

Amunak, 2018-06-10:
Just tested shutdown sequence, it works! (yay)

Though the `battery.charge.low` option cannot *really* be set (as in, the upsrw command returns OK, but it doesn't actually change anything on the UPS) and 10% seems way too low to be safe. It *does work* but I wouldn't trust it if your load is over about 20% (or your shutdown takes more than, say, 30 seconds). But that's not a NUT issue, just something I observed and it's good to keep it in mind.

rickettm, 2021-01-06:
I have also confirmed this UPS works (although mine was shipped with a half-dead battery :( )

closed 2022-04-25 by jimklimov

bangcrash, 2023-02-14:
@Amunak Apologies for digging up and old thread - I have the BR700ELCD and the ups.load attribute never reports anything other than 0.  Does this report correctly with your UPS? Thanks

Amunak, 2023-02-14:
> I have the BR700ELCD and the ups.load attribute never reports anything other than 0. Does this report correctly with your UPS? Thanks

Yep, I get regular values for the load, though I'm not sure about the accuracy. It likes to be even at 0 sometimes for me, even if I'm probably at about 10-30% actual load most of the time. Try to run it like this: `$ watch -n 0.5 upsc nutdev1` and watch for the values to change.

It also counts (somewhat understandably) only the battery-backed plugs. I think it's more accurate when running on battery power but I don't entirely remember. Try plugging in some higher load maybe?

commit df28dbbd (2022-04-25): driver.list.in: Add HCL info for CyberPower Systems BR1000ELCD (USB)

Closes: #552