networkupstools/nut #553: Hotplug USB device doesn't load driver

VasiliPupkin256, 2018-05-07:
The nut driver is normally loaded during boot, if there was no UPS connected at boot time the driver will be terminated and will not start automatically if you plug it later. It can be fixed by adding `RUN+="/bin/systemctl restart nut-server.service"` to the 52-nut-usbups udev rules file.

jimklimov, 2018-05-07:
For clarity, which NUT version (and OS) is this about? :)

At face value, restarting the `nut-server` is orthogonal to restarting a driver. It might however make visible the driver that was not available earlier (at least, if you configure new sections in `ups.conf` you should both start a new driver daemon and restart the `nut-server` which wraps `upsd` daemon), and/or such restart would quickly remove an existing known driver's "data stale" condition by starting life from scratch.

I hope my work on #330 does a good step toward automating much of these restarts (as well as allowing HW/SW dependencies to be tracked as for other OS services), and would welcome some real-life testing and feedback there ;)

VasiliPupkin256, 2018-05-08:
My usecase: I've set everything up and turned off computer to do final cable management, when I've turned it on I realized that I forget to plug UPS USB cable, I've plugged it in but NUT didn't catch it and didn't load a driver for it.

Will the #330 solve this issue?