networkupstools/nut #554: [HCL] Salicru SPS ONE 2000 supported by nutdrv_qx

vanmaal, 2018-05-13:
I managed to get it working using the suggested parameters in the driver page for USB UPS
Description can be found here:
https://www.salicru.com/en/ups/sps-one.html 
I have the 2000 VA model, the others use the same ViewPower software so they should use the same driver, but I cannot guarantee that.
`# upsc salicru
Init SSL without certificate database
battery.charge: 70
battery.voltage: 24.30
battery.voltage.high: 25.50
battery.voltage.low: 21.50
battery.voltage.nominal: 24.0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.offdelay: 120
driver.parameter.ondelay: 120
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.2
driver.version.data: Voltronic-QS 0.01
driver.version.internal: 0.06
input.current.nominal: 8.0
input.frequency: 50.1
input.frequency.nominal: 50
input.voltage: 229.7
input.voltage.fault: 229.7
input.voltage.nominal: 230
output.voltage: 229.7
ups.beeper.status: enabled
ups.delay.shutdown: 120
ups.delay.start: 120
ups.firmware.aux: PMV
ups.load: 6
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665`

`# upsrw salicru
[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: RANGE
Option: "12-540" SELECTED

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: RANGE
Option: "60-599940" SELECTED`

`# upscmd -l salicru
Instant commands supported on UPS [salicru]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.quick - Start a quick battery test`

Shutdown worked as expected when reaching LOWBATT:
`May 13 20:43:22 oriolnas upsmon[1535]: UPS salicru@127.0.0.1 battery is low
May 13 20:43:22 oriolnas upsd[1532]: Client upsmon@127.0.0.1 set FSD on UPS [salicru]
May 13 20:43:22 oriolnas upsmon[1535]: Executing automatic power-fail shutdown
May 13 20:43:22 oriolnas upsmon[1535]: Auto logout and shutdown proceeding`

javier-izquierdo-visualfy, 2019-07-25:
@vanmaal Could you send me the configuration? I can't get it to work, thanks.

vanmaal, 2019-07-25:
@javier-izquierdo-visualfy Ooops... Unfortunately this machine is out of reach for me now... Hopefully next week will get it online again, so if you still need it I'll get it for you. I assume you need ups.conf? Another config file...?

javier-izquierdo-visualfy, 2019-07-26:
@vanmaal Any other that may be necessary, in principle only the ups.conf.

closed 2022-04-25 by jimklimov

sshaikh, 2025-02-17:
Not able to get this to work with my 900VA either. Blazer works.

jimklimov, 2025-02-18:
@sshaikh : which NUT version?

sshaikh, 2025-02-18:
`uspc --help` returns 2.8.0