networkupstools/nut #555: MInuteman UPS

wkleinhenz, 2018-05-22:
Currently i was able to acquire a minuteman Enterprise plus E1500RM2U UPS, if possible i was wonder if there is a plan to support this brand/model and if so how I can help

Thanks

wkleinhenz, 2018-05-22:
I gave that driver a try but it doesnt seems to be picking anything up, im going to unplug the other ups(cyberpower) and figure out the port the Minuteman one is using and double check

wkleinhenz, 2018-05-23:
sorry im not able to find the drivers/usbhid-ups file but attached below is the output of `lsusb -vvv` ill keep looking for the other bit

thanks

[minuteman ups lsusb output.txt](https://github.com/networkupstools/nut/files/2033185/minuteman.ups.lsusb.output.txt)

clepple, 2018-05-23:
That path is if you are building from source. If it's a Debian/Ubuntu-derived distribution, the driver gets installed to `/lib/nut/usbhid-ups`. If not, let us know which distribution (I don't know the other common paths off the top of my head).

clepple, 2018-05-23:
Also, thanks for posting the lsusb output. `wDescriptorLength    1028` is a good indication that `usbhid-ups` can be adapted to work with this UPS.

clepple, 2018-05-24:
The standard HID PDC IDs look good, and the vendor-specific IDs look similar to Tripp-Lite.

Next step is to rebuild the driver. Instructions for Ubuntu are here: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

Once you can build and run the driver from source (the `-x explore` thing is a good test), then we can add the USB vendor/product IDs into the `tripplite-hid.c` file, and see how well that matches.

clepple, 2018-05-25:
I added the USB VID:PID to a branch. You can get that code with `git fetch` and then `git checkout minuteman-HID`, then rebuild.

At that point, you should be able to run the driver without the `-x explore` option. Logs (with `-DD`) would help with figuring out the line/battery problem (although if it stays stuck at around 80% battery, there may be an issue with the charging circuitry)

wkleinhenz, 2018-05-25:
ok i complied that branch ill run the command tomorrow when i get a chance and post the logs but for the time being using the nut web gui seems to show the battery level at 100%

clepple, 2018-05-29:
(Just stumbled across your update - note that Github doesn't notify via email for edits, so it's probably best to add a new comment if you add something like a new log file.)

If you monitor the `upsc` output, is it still switching between battery and line (`ups.status` going between `OL` and `OB`), or is that no longer the case? The log you just posted shows `UPS.PowerSummary.PresentStatus.ACPresent` as 1, so if it is switching, we might need a higher debug level to see which other variable is affecting it.

wkleinhenz, 2018-05-29:
I took some more, the messaged still happens, it seems at some point the connection times out or something. would and increase in the poll_time or Max Dead time help?

Sorry had to use pastebin, the file wouldn't take 
https://pastebin.com/C4TDBts4

clepple, 2018-05-30:
Polling less frequently might help, but the "busy" error is worrisome (that's why the driver reconnects). Details on the polling times here: http://new.networkupstools.org/docs/man/usbhid-ups.html#_usb_polling_and_interrupt_transfers

The polling frequency doesn't explain why the UPS occasionally returns ACPresent as 0. Looks like flaky firmware.

MAXDEAD and others are for when you get "data stale" messages from `upsc` and `upsmon`.

clepple, 2018-05-30:
Potentially stupid question, but are there any other packages that might be polling the UPS at the same time? UPower might be, though it might also not know about this USB VID:PID combination.

wkleinhenz, 2018-05-30:
I dont believe so, tomorrow ill try a fresh install just in case, is there anything else i could check

wkleinhenz, 2018-06-11:
Sorry i haven't messaged in a bit, but i rebuilt NUT and have it working with the ups and am seeing the same issue but it seems a little less common. What would be the best approach for trying to figure this issue out?

charlesnw1, 2020-03-28:
Hello,

I also have a minuteman UPS. 

lsusb shows it as:
Bus 001 Device 005: ID 05dd:a011 Delta Electronics, Inc.

Any updates on this issue? Will support be added to NUT upstream? Is best bet to run the minuteman branch?