networkupstools/nut #560: support for PowerWalker VFI 2000 TGS

Links2004, 2018-06-13:
Hi,
I have a PowerWalker VFI 2000 TGS UPS and I am able to get some basic values via `upsc`,
but many values are missing, like:

- Input Frequency
- Input Voltage
- Output Frequency
- Output Voltage
- correct battery Voltage (54.8V not 55V)

what is needed to do to map values?

```
# upsc PowerWalker-hid
Init SSL without certificate database
battery.charge: 92
battery.runtime: 31503
battery.type: PbAc
battery.voltage: 55
battery.voltage.nominal: 48
device.mfr: PPC
device.model: HID UPS
device.serial: 170909-54810007     
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: no
driver.parameter.vendorid: 06da
driver.version: 2.7.4
driver.version.data: Phoenixtec/Liebert HID 0.3
driver.version.internal: 0.41
ups.load: 0
ups.mfr: PPC
ups.model: HID UPS
ups.productid: ffff
ups.serial: 170909-54810007     
ups.status: OL CHRG
ups.vendorid: 06da
```

```
# usbhid-ups -D -a PowerWalker-hid

Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	debug level is '1'
   0.000309	upsdrv_initups...
   0.010152	Using subdriver: Phoenixtec/Liebert HID 0.3
   0.010160	144 HID objects found
   0.010346	Path: UPS.BatterySystem.Battery.BatteryID, Type: Feature, ReportID: 0x20, Offset: 0, Size: 8, Value: 1
   0.010578	Path: UPS.BatterySystem.Battery.ffff0090, Type: Feature, ReportID: 0x21, Offset: 0, Size: 8, Value: 1
   0.010841	Path: UPS.BatterySystem.Battery.PresentStatus.Present, Type: Feature, ReportID: 0x27, Offset: 0, Size: 8, Value: 1
   0.010866	Path: UPS.BatterySystem.Battery.PresentStatus.Present, Type: Input, ReportID: 0x27, Offset: 0, Size: 8, Value: 1
   0.010870	Path: UPS.BatterySystem.Battery.PresentStatus.ffff0074, Type: Feature, ReportID: 0x27, Offset: 8, Size: 8, Value: 0
   0.010874	Path: UPS.BatterySystem.Battery.PresentStatus.ffff0074, Type: Input, ReportID: 0x27, Offset: 8, Size: 8, Value: 0
   0.011064	Path: UPS.BatterySystem.Battery.Test, Type: Feature, ReportID: 0x24, Offset: 0, Size: 8, Value: 6
   0.011375	Path: UPS.BatterySystem.Battery.ffff00e3, Type: Feature, ReportID: 0xe3, Offset: 0, Size: 8, Value: 0
   0.011551	Path: UPS.BatterySystem.Battery.ffff0045, Type: Feature, ReportID: 0x25, Offset: 0, Size: 32, Value: 2.4192e+06
   0.011576	Path: UPS.BatterySystem.BatterySystemID, Type: Feature, ReportID: 0x20, Offset: 8, Size: 8, Value: 1
   0.011582	Path: UPS.BatterySystem.Charger.ChargerID, Type: Feature, ReportID: 0x20, Offset: 16, Size: 8, Value: 1
   0.011589	Path: UPS.BatterySystem.Charger.PresentStatus.ffff0079, Type: Feature, ReportID: 0x27, Offset: 16, Size: 8, Value: 0
   0.011594	Path: UPS.BatterySystem.Charger.PresentStatus.ffff0079, Type: Input, ReportID: 0x27, Offset: 16, Size: 8, Value: 0
   0.011600	Path: UPS.BatterySystem.Charger.PresentStatus.InternalFailure, Type: Feature, ReportID: 0x27, Offset: 24, Size: 8, Value: 0
   0.011617	Path: UPS.BatterySystem.Charger.PresentStatus.InternalFailure, Type: Input, ReportID: 0x27, Offset: 24, Size: 8, Value: 0
   0.011621	Path: UPS.BatterySystem.Charger.PresentStatus.Used, Type: Feature, ReportID: 0x27, Offset: 32, Size: 8, Value: 1
   0.011640	Path: UPS.BatterySystem.Charger.PresentStatus.Used, Type: Input, ReportID: 0x27, Offset: 32, Size: 8, Value: 1
   0.011783	Path: UPS.Flow.[1].ConfigFrequency, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 50
[... 110 lines cut ...]
   0.022131	Path: UPS.PowerSummary.PresentStatus.InternalFailure, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 0
   0.022137	Path: UPS.PowerSummary.PresentStatus.InternalFailure, Type: Input, ReportID: 0x02, Offset: 0, Size: 8, Value: 0
   0.022142	Path: UPS.PowerSummary.PresentStatus.NeedReplacement, Type: Feature, ReportID: 0x02, Offset: 8, Size: 8, Value: 0
   0.022148	Path: UPS.PowerSummary.PresentStatus.NeedReplacement, Type: Input, ReportID: 0x02, Offset: 8, Size: 8, Value: 0
   0.022153	Path: UPS.PowerSummary.PresentStatus.Overload, Type: Feature, ReportID: 0x02, Offset: 16, Size: 8, Value: 0
   0.022158	Path: UPS.PowerSummary.PresentStatus.Overload, Type: Input, ReportID: 0x02, Offset: 16, Size: 8, Value: 0
   0.022162	Path: UPS.PowerSummary.PresentStatus.OverTemperature, Type: Feature, ReportID: 0x03, Offset: 8, Size: 8, Value: 0
   0.022167	Path: UPS.PowerSummary.PresentStatus.OverTemperature, Type: Input, ReportID: 0x03, Offset: 8, Size: 8, Value: 0
   0.022271	Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 92
   0.022277	Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x06, Offset: 0, Size: 8, Value: 92
   0.022283	Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x06, Offset: 8, Size: 32, Value: 31492
   0.022288	Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x06, Offset: 8, Size: 32, Value: 31492
   0.022294	Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x07, Offset: 8, Size: 16, Value: 54.8
   0.022525	Path: UPS.ffff0010.[1].ffff00f0, Type: Feature, ReportID: 0xf0, Offset: 0, Size: 8, Value: 0
   0.022786	Path: UPS.ffff0010.[1].ffff00f1, Type: Feature, ReportID: 0xf1, Offset: 0, Size: 8, Value: 0
   0.023017	Path: UPS.ffff0010.[1].ffff00ff, Type: Feature, ReportID: 0xff, Offset: 0, Size: 8, Value: 0
   0.023042	Detected a UPS: PPC/HID UPS
   0.023402	upsdrv_initinfo...
   0.023408	upsdrv_updateinfo...
   0.273702	Got 0 HID objects...
   0.273717	Quick update...
   0.273798	upsdrv_updateinfo...
   0.524625	Got 0 HID objects...
   0.524640	Quick update...
```

Links2004, 2018-06-17:
where able to get read working.
interesting is that the latest git uses mge-hid and not Liebert

https://github.com/Links2004/nut/commit/b6bd034e09589549afad6ec9ff7522f13654861b

clepple, 2018-06-18:
Are you sure that it is necessary to comment some of those out? That will likely break support for MGE hardware.

Links2004, 2018-06-19:
will test, but the commented stuff where from type strong `%s` but needed to be float `%.1f` the naming of the parameters is the same

Keltere, 2021-03-18:
There are any updates regarding this? Why hasn't been merged yet?

jimklimov, 2021-03-18:
@Keltere : this was not a pull request so technically can't merge - no such button here. And probably got lately overlooked as an old issue discussion. Also without testing on varied hardware, just looking at the commit, I agree with @clepple that even if this change fixes support for one vendor/model, it might be breaking it for established other(s), at least until proven it works on both types of devices. In the short discussion above I do not see confirmations of that.

Does PowerWalker use the same USB HID ID's as MGE? If not, a certainly less conflictey way would be to define a new usbhid-ups subdriver by copying the file content from that commit (or maybe compiling it twice with ifdef's into different objects), so defining two different mapping tables for two different situations (and original not broken).

I believe for logically similar mappings in snmp-ups, it was possible to define same field as implemented by different OIDs, so the driver would ask in order until it gets a reply. Not sure if there is something similar for USB but may be worth investigating (or implementing) to make one driver that handles both device types without regression - and avoid maintenance of almost identical separate sources.

jimklimov, 2022-04-08:
I've updated PR #564 to err on the safe side, adding the mappings to liebert-hid instead (the support of Vendor ID 0x06da in mge-hid subdriver seems to be a later accident, and might pertain to AEG devices if at all). That said, liebert mappings seem to be smaller than mge mappings, so additional PRs (tested against HW) to transplant served but ignored data into the liebert-hid.c would be welcome.