networkupstools/nut #562: Eaton Network Card-MS Outlet Group

kmlucy, 2018-06-18:
I have an Eaton PW9130L1500R-XL2U with a Network Card-MS. I am using the netxml-ups driver. The ``ups.conf`` config I am using is here:
```
[ups1]
        driver = netxml-ups
        port = http://<FQDN>:80
        subscribe
        shutdown_duration = 600
        login=<USERNAME>
        password=<PASSWORD>
```

Everything works correctly, but I am trying to figure out how to set NUT to use a specific outlet group. On my management card's webui, it shows this client being configured on the ``Master`` segment:
[image]

I need to set it to be on Group1 so that only that load segment gets powered off when the server shuts down.

Is there any way to accomplish this?

clepple, 2018-06-19:
I am not very familiar with this driver, and it doesn't look like we have any public protocol documentation for it. The only thing related to outlet groups seems to be this:

https://github.com/networkupstools/nut/blob/master/drivers/netxml-ups.c#L730

So it looks like it is currently hardcoded, but could probably be extended. @aquette or @jimklimov, any pointers?

jimklimov, 2018-06-19:
I believe this goes as `outlet.$N.nutvarname` management, e.g. `outlet.1.delay.shutdown` etc.

Note that for `upsrw` over netxml you may need to use `-u admin -p yourpasswd`, at least that's what I see in Eaton's IPP Unix (1.40) scripts that is based on older NUT, you can download it freely to inspect the scripts and setup templates, if that helps for now...

clepple, 2018-06-20:
Ah, this is starting to sound familiar. The [clone driver](https://networkupstools.org/docs/man/clone.html) seems to be intended to handle the case where the shutdown command from `upsmon` gets routed to only one outlet control.

kmlucy, 2018-06-21:
That doesn't sound like quite what I need. My UPS can be configured such that once all servers on a outlet group have shut down, it powers off that outlet group. In order for that to work, the clients need to report which outlet group they are on, similar to how the driver reports the shutdown duration.

closed 2018-11-06 by kmlucy

Rogue136198, 2022-10-31:
Sorry to necro this but wat this ever resolved? If so can you share how to do this?

kmlucy, 2022-10-31:
No, I gave up and used Eaton's software, IPP.