networkupstools/nut #567: dummy-ups consumes much CPU

jimklimov, 2018-06-29:
We have a system that acts as a sort of NUT proxy, to alleviate actual power devices during testing, and to provide some virtual devices - both roles implemented by dummy-ups from 42ity fork of NUT (should be on par with github master). The system was restarted about a week ago, and today I've noticed that the dummy-ups drivers that serve "dev" files consume a lot of CPU (50-70% of a core each) despite the files ending with a `TIMER 60` line.

Restarting the driver in debug mode did not help - it behaves as it should, reading the file and then quickly and quietly refreshing the state ("I'm not stale") with no load on the system.

While `strace`ing the driver with high load, which ran for a week now, I see it doing the following loop:

````
...
10:48:36 select(8, [5 6 7], NULL, NULL, {0, 352}) = 1 (in [6], left {0, 350})
10:48:36 read(6, "", 512)               = 0
10:48:36 gettimeofday({1530262116, 859989}, NULL) = 0
10:48:36 select(8, [5 6 7], NULL, NULL, {0, 246}) = 1 (in [6], left {0, 243})
10:48:36 read(6, "", 512)               = 0
10:48:36 gettimeofday({1530262116, 860095}, NULL) = 0
10:48:36 select(8, [5 6 7], NULL, NULL, {0, 140}) = 1 (in [6], left {0, 138})
10:48:36 read(6, "", 512)               = 0
10:48:36 gettimeofday({1530262116, 860208}, NULL) = 0
10:48:36 select(8, [5 6 7], NULL, NULL, {0, 27}) = 1 (in [6], left {0, 25})
10:48:36 read(6, "", 512)               = 0
10:48:36 gettimeofday({1530262116, 860312}, NULL) = 0
10:48:36 select(8, [5 6 7], NULL, NULL, {0, 0}) = 1 (in [6], left {0, 0})
10:48:36 read(6, "", 512)               = 0
10:48:36 gettimeofday({1530262116, 860416}, NULL) = 0
````
(many screenfuls of this spit out quickly)

then a sleep:
````
10:48:36 rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
10:48:36 rt_sigaction(SIGCHLD, NULL, {SIG_DFL, [], 0}, 8) = 0
10:48:36 rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
10:48:36 nanosleep({1, 0},
````
followed by another burst like above

with occasional occurrences of

````
11:10:31 gettimeofday({1530263431, 513516}, NULL) = 0
11:10:31 select(8, [5 6 7], NULL, NULL, {0, 577422}) = 1 (in [6], left {0, 577419})
11:10:31 read(6, "", 512)               = 0
11:10:31 gettimeofday({1530263498, 611545}, NULL) = 0
11:11:38 select(8, [5 6 7], NULL, NULL, {0, 0}) = 2 (in [6 7], left {0, 0})
11:11:38 read(7, "PING\nPING\nPING\nPING\nPING\nPING\nPI"..., 512) = 55
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 write(7, "PONG\n", 5)          = 5
11:11:38 read(6, "", 512)               = 0
11:11:38 gettimeofday({1530263498, 612809}, NULL) = 0
11:11:38 rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
11:11:38 rt_sigaction(SIGCHLD, NULL, {SIG_DFL, [], 0}, 8) = 0
11:11:38 rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
11:11:38 nanosleep({1, 0},
````

````

clepple, 2018-06-29:
If you still have the `strace` output, or if you can grab this from `/proc/fd`, what is fd#6 connected to?

If it is the dev file, then maybe the code is not handling EOF correctly?

>Those listed in readfds will be watched to see if characters become available for reading (more precisely, to see  if  a  read  will  not  block;  in  particular,  a file descriptor is also ready on end-of-file), ...

I'm not sure why the `PING`s are not getting flushed.

jimklimov, 2018-06-29:
Some unspecified socket; the dev-file is fd#4:

````
# ls -la /proc/60?/fd/*
lrwx------ 1 root root 64 Jun 29 14:37 /proc/600/fd/0 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/600/fd/1 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/600/fd/2 -> /dev/null
lr-x------ 1 root root 64 Jun 29 14:37 /proc/600/fd/3 -> /dev/urandom
lrwx------ 1 root root 64 Jun 29 14:37 /proc/600/fd/4 -> socket:[2029786205]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/600/fd/5 -> socket:[2029782541]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/600/fd/6 -> socket:[2029786213]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/600/fd/8 -> socket:[2041948453]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/602/fd/0 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/602/fd/1 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/602/fd/2 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/602/fd/3 -> socket:[2029779593]
lr-x------ 1 root root 64 Jun 29 14:37 /proc/602/fd/4 -> /etc/nut/Eaton__UPS_9PX_snmp_pw.dev
lrwx------ 1 root root 64 Jun 29 14:37 /proc/602/fd/5 -> socket:[2029779623]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/602/fd/6 -> socket:[2029812746]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/602/fd/7 -> socket:[2041957512]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/603/fd/0 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/603/fd/1 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/603/fd/2 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/603/fd/3 -> socket:[2029787142]
lr-x------ 1 root root 64 Jun 29 14:38 /proc/603/fd/4 -> /etc/nut/Eaton__UPS_9PX_netxml.dev
lrwx------ 1 root root 64 Jun 29 14:37 /proc/603/fd/5 -> socket:[2029787169]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/603/fd/6 -> socket:[2029812750]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/603/fd/7 -> socket:[2041962657]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/604/fd/0 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/604/fd/1 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/604/fd/2 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/604/fd/3 -> socket:[2029779599]
lr-x------ 1 root root 64 Jun 29 14:37 /proc/604/fd/4 -> /etc/nut/Eaton__UPS_9PX_snmp_mge.dev
lrwx------ 1 root root 64 Jun 29 14:37 /proc/604/fd/5 -> socket:[2029779625]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/604/fd/6 -> socket:[2029812748]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/604/fd/7 -> socket:[2041937804]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/605/fd/0 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/605/fd/1 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/605/fd/2 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/605/fd/3 -> socket:[2029782520]
lr-x------ 1 root root 64 Jun 29 14:37 /proc/605/fd/4 -> /etc/nut/Eaton__EPDU_MI_0U_(309W_32A_3P)18XC13_9XC19_combo_MI_38U-A__snmp-ups__2.7.4.1__01.dev
lrwx------ 1 root root 64 Jun 29 14:37 /proc/605/fd/5 -> socket:[2029782543]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/605/fd/6 -> socket:[2029812756]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/605/fd/7 -> socket:[2041964523]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/606/fd/0 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/606/fd/1 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/606/fd/2 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/606/fd/3 -> socket:[2029786199]
lr-x------ 1 root root 64 Jun 29 14:37 /proc/606/fd/4 -> /etc/nut/Eaton__EPDU_MI_38U-A__snmp-ups__2.7.4__01.dev
lrwx------ 1 root root 64 Jun 29 14:37 /proc/606/fd/5 -> socket:[2029786228]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/606/fd/6 -> socket:[2029812754]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/606/fd/7 -> socket:[2041964527]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/607/fd/0 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/607/fd/1 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/607/fd/2 -> /dev/null
lrwx------ 1 root root 64 Jun 29 14:37 /proc/607/fd/3 -> socket:[2029784179]
lr-x------ 1 root root 64 Jun 29 14:37 /proc/607/fd/4 -> /etc/nut/Eaton__EPDU_MI_0U_(309W_32A_3P)18XC13_9XC19__snmp-ups__2.7.4.1__01.dev
lrwx------ 1 root root 64 Jun 29 14:37 /proc/607/fd/5 -> socket:[2029784200]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/607/fd/6 -> socket:[2029812752]
lrwx------ 1 root root 64 Jun 29 14:37 /proc/607/fd/7 -> socket:[2041964592]
````

jimklimov, 2018-06-29:
As another data point: despite all this, `upsc DEVICENAME` passes well, quickly and producing some reasonably looking `key: value` results.