networkupstools/nut #569: support for PowerWalker VI 1500 RT HID

goldie83, 2018-07-03:
Hi
I have a PowerWalker VI 1500 RT HID, trying to use the usbhid-ups driver, but I do not see any commands:
```
$ upsc pw
Init SSL without certificate database
battery.charge: 100
battery.runtime: 5282
battery.type: PbAc
battery.voltage: 41
battery.voltage.nominal: 36
device.mfr: PPC
device.model: HID UPS
device.serial: 716317C00120
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.vendorid: 06da
driver.version: 2.7.2
driver.version.data: Liebert HID 0.3
driver.version.internal: 0.38
ups.load: 18
ups.mfr: PPC
ups.model: HID UPS
ups.productid: ffff
ups.serial: 716317C00120
ups.status: OL CHRG
ups.vendorid: 06da
```
Output from `/lib/nut/usbhid-ups -DD -a pw`
```
   0.000000     debug level is '2'
   0.002392     upsdrv_initups...
   0.003650     Checking device (174C/1153) (001/006)
   0.003821     - VendorID: 174c
   0.003868     - ProductID: 1153
   0.003909     - Manufacturer: unknown
   0.003949     - Product: unknown
   0.003989     - Serial Number: unknown
   0.004029     - Bus: 001
   0.004067     Trying to match device
   0.004117     Device does not match - skipping
   0.004174     Checking device (06DA/FFFF) (001/005)
   0.010156     - VendorID: 06da
   0.010204     - ProductID: ffff
   0.010245     - Manufacturer: PPC
   0.010285     - Product: HID UPS
   0.010324     - Serial Number: 716317C00120
   0.010364     - Bus: 001
   0.010403     Trying to match device
   0.010563     Device matches
   0.011565     HID descriptor length 1393
   0.033784     Report Descriptor size = 1393
   0.036523     Using subdriver: Liebert HID 0.3
   0.037062     Path: UPS.BatterySystem.Battery.BatteryID, Type: Feature, ReportID: 0x20, Offset: 0, Size: 8, Value: 1
   0.037555     Path: UPS.BatterySystem.Battery.ffff0090, Type: Feature, ReportID: 0x21, Offset: 0, Size: 8, Value: 3
   0.038061     Path: UPS.BatterySystem.Battery.PresentStatus.Present, Type: Feature, ReportID: 0x27, Offset: 0, Size: 8, Value: 1
   0.038141     Path: UPS.BatterySystem.Battery.PresentStatus.Present, Type: Input, ReportID: 0x27, Offset: 0, Size: 8, Value: 1
   0.038217     Path: UPS.BatterySystem.Battery.PresentStatus.ffff0074, Type: Feature, ReportID: 0x27, Offset: 8, Size: 8, Value: 0
   0.038292     Path: UPS.BatterySystem.Battery.PresentStatus.ffff0074, Type: Input, ReportID: 0x27, Offset: 8, Size: 8, Value: 0
   0.038801     Path: UPS.BatterySystem.Battery.Test, Type: Feature, ReportID: 0x24, Offset: 0, Size: 8, Value: 1
   0.040554     Path: UPS.BatterySystem.Battery.ffff00e3, Type: Feature, ReportID: 0xe3, Offset: 0, Size: 8, Value: 1
   0.041067     Path: UPS.BatterySystem.Battery.ffff0045, Type: Feature, ReportID: 0x25, Offset: 0, Size: 32, Value: 2.4192e+06
   0.041138     Path: UPS.BatterySystem.BatterySystemID, Type: Feature, ReportID: 0x20, Offset: 8, Size: 8, Value: 1
   0.041206     Path: UPS.BatterySystem.Charger.ChargerID, Type: Feature, ReportID: 0x20, Offset: 16, Size: 8, Value: 1
   0.041677     Path: UPS.BatterySystem.Charger.ffff0094, Type: Feature, ReportID: 0x26, Offset: 0, Size: 8, Value: 4
   0.041757     Path: UPS.BatterySystem.Charger.PresentStatus.ffff0079, Type: Feature, ReportID: 0x27, Offset: 16, Size: 8, Value: 0
   0.041832     Path: UPS.BatterySystem.Charger.PresentStatus.ffff0079, Type: Input, ReportID: 0x27, Offset: 16, Size: 8, Value: 0
[... 146 lines cut ...]
   0.101824     Path: UPS.PowerSummary.PresentStatus.Overload, Type: Feature, ReportID: 0x02, Offset: 16, Size: 8, Value: 0
   0.101898     find_nut_info: unknown info type: load.off.delay
   0.101942     find_nut_info: unknown info type: load.on.delay
   0.101983     find_nut_info: unknown info type: load.off.delay
   0.102055     upsdrv_initinfo...
   0.102118     upsdrv_updateinfo...
   0.184313     Got 2 HID objects...
   0.184405     Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x06, Offset: 0, Size: 8, Value: 100
   0.184503     Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x06, Offset: 8, Size: 32, Value: 2927
   0.184576     Quick update...
   0.184642     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.184707     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
   0.184772     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
   0.184872     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
   0.185113     dstate_init: sock /var/run/nut/usbhid-ups-pokoje open on fd 5
   0.185200     upsdrv_updateinfo...
   0.435446     libusb_get_interrupt: Connection timed out
   0.435542     Got 0 HID objects...
   0.435586     Quick update...
   0.435656     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.435724     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
   0.435790     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
   0.435855     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
   2.187000     upsdrv_updateinfo...
```

mdillmann, 2019-07-08:
I also own this device.
In the current git-version the MGE-HID driver claims this vendor/product id and actually delivers more values (eg support for the outlets) and command support:

```
upsc pw
Init SSL without certificate database
battery.charge: 100
battery.charger.status: floating
battery.energysave: no
battery.runtime: 6244
battery.type: PbAc
device.mfr: PPC
device.model: HID UPS Sinewave 1500VA UPS
device.serial: 716318C00018    
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: no
driver.parameter.vendorid: 06da
driver.version: 2.7.4-708-g241848c3
driver.version.data: MGE HID 1.44
driver.version.internal: 0.43
input.frequency: 50.0
input.frequency.nominal: 50
input.transfer.boost.low: 207
input.transfer.high: 276
input.transfer.low: 184
input.transfer.trim.high: 243
input.voltage: 233.0
input.voltage.nominal: 230
outlet.1.delay.shutdown: 2147483647
outlet.1.delay.start: 3
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.delay.shutdown: 2147483647
outlet.2.delay.start: 6
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 2
outlet.2.status: on
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 233.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 05649-0499
ups.load: 10
ups.load.high: 110
ups.mfr: PPC
ups.model: HID UPS Sinewave 1500VA UPS
ups.power: 152
ups.power.nominal: 1500
ups.productid: ffff
ups.realpower: 96
ups.realpower.nominal: 1350
ups.serial: 716318C00018    
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL CHRG
ups.temperature: 34.0
ups.test.interval: 2419200
ups.test.result: Done and passed
ups.timer.shutdown: -1
ups.timer.start: -1
ups.type: offline / line interactive
ups.vendorid: 06da
```

BUT:
the online/onbatt status fluctuates:
```
root@pihole:~/nut# upsc pw ups.status
Init SSL without certificate database
OL CHRG
root@pihole:~/nut# upsc pw ups.status
Init SSL without certificate database
OB CHRG
root@pihole:~/nut# upsc pw ups.status
Init SSL without certificate database
OB CHRG
root@pihole:~/nut# upsc pw ups.status
Init SSL without certificate database
OL CHRG
```
During "quick poll" all is good, only a full-hid-scan sets it as offline.

The issue is, that the firmware of this device sends a "UPS.PowerConverter.Input.[3].PresentStatus.Used" during normal operations (AC present) and the original MGE sees this as on bat:
```
1201.398937	[D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
1201.398985	[D5] hu_find_infoval: found online (value: 1)
1201.399027	[D5] process_boolean_info: online
1201.399065	[D4] Entering libusb_get_report
1201.399603	[D3] Report[get]: (2 bytes) => 3b 01
1201.399659	[D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
1201.399704	[D5] Unit = 00000000, UnitExp = 0
1201.399740	[D5] Exponent = 0
1201.399784	[D2] Path: UPS.PowerConverter.Input.[3].PresentStatus.Used, Type: Feature, ReportID: 0x3b, Offset: 0, Size: 8, Value: 1
1201.399830	[D5] hu_find_infoval: found !online (value: 1)
1201.399870	[D5] process_boolean_info: !online
1201.399925	[D3] Report[buf]: (6 bytes) => 01 01 00 01 00 01
1201.399968	[D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
1201.400009	[D5] Unit = 00000000, UnitExp = 0
1201.400048	[D5] Exponent = 0
1201.400091	[D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
1201.400135	[D5] process_boolean_info: !dischrg
1201.400190	[D3] Report[buf]: (6 bytes) => 01 01 00 01 00 01
1201.400233	[D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
1201.400276	[D5] Unit = 00000000, UnitExp = 0
1201.400314	[D5] Exponent = 0
```

Can mge-hid.c be changed so that the "UPS.PowerConverter.Input.[3].PresentStatus.Used" is NOT applied/considered as offline for the specific manufacturer/model `ups.mfr: PPC
ups.model: HID UPS Sinewave 1500VA UPS `?

I could test, but lack coding skills to do that on my own.

cuthulino, 2022-09-07:
I got the same problem as @mdillmann.
onbatt and online status are fluctuating.
Not possible to send notifications.

I tried to filter the events with upssched timer but no luck because when it is onbatt, sometimes it reports an online event between.

Is there a workaround or can we excpect a fix here?

jimklimov, 2022-09-08:
@cuthulino : just in case, which version of NUT are you running? If it is a packaged one (e.g. old v2.7.4) then probably no fix is expected until distros update their recipes to deliver a newer NUT release. What you have packaged - is set in stone. If you're (open to) running a custom build, there's a chance :)

As I re-read this issue (and thanks to @mdillmann for analysis above), it sounded reminiscent of `onlinedischarge` feature added before NUT v2.8.0 (hence the question - maybe the new release does happen to address this behavior?), although upon closer reading into the code of that change at #811 and #1359 - maybe it does not and just is similar... similar enough to propose another toggle to either completely disable looking at that report as @mdillmann proposed, or to ignore `!online` from `UPS.PowerConverter.Input.[3].PresentStatus.Used`  if `UPS.PowerSummary.PresentStatus.ACPresent` says we are online (and for good measure, if we are not discharging at the moment).

CC @aquette about whether/how this can be done in a sane manner? Move from `mge_onbatt_info` static mapping to a function like `eaton_converter_online_fun` from `eaton_converter_online_info` dynamic mapping, that would consider recent (timed to not expire) readings from those several data sources, and/or the user toggle to ignore some of the data sources?

Also generally, why would it flip? Is it a firmware bug to nag the vendor about, or does it reflect inner workings of that UPS way too honestly (e.g. some line-interactive "Power Converter" device really going "offline" so the UPS load is fed by the wall power directly, and this intimate knowledge is misinterpreted by NUT driver)?

jimklimov, 2022-09-08:
@mdillmann  @cuthulino  : Also as for a local PoC fix, you can try to comment away (move under nearby `#if 0`) the mapping line https://github.com/networkupstools/nut/blob/master/drivers/mge-hid.c#L1251 and rebuild NUT locally. It should then ignore/not-query that information, so we can see if that ignorance generally helps with these devices.

jimklimov, 2022-09-08:
And as a side note: curious how many mappings apply to "PowerWalker VI" series - here different versions of NUT covered it with Liebert and MGE mapping support, and in #1512 the device has a completely different vendor and product ID and so gets covered as a CyberPower (CPS) unit...

cuthulino, 2022-09-09:
@jimklimov you are right. I am really running the old 2.7.4 from the packages.

I will try the 2.8.0 in the next days and report back, if the error persists.

warp735, 2023-05-31:
I have the same Problem with 2.8.0

jimklimov, 2023-05-31:
@warp735 : Same model? NUT 2.8.0 release or more recent (e.g. custom build from github master)?

warp735, 2023-05-31:
@jimklimov 
sorry, just saw that it's still 2.7 (use Unraid) Yes, it's the same model and the same error.

desertwitch, 2023-07-26:
@warp735 has since confirmed his VI 1000 RT HID working with NUT 2.8.0 on UNRAID
see https://forums.unraid.net/topic/60217-plugin-nut-v2-network-ups-tools/?do=findComment&comment=1288104

cuthulino, 2023-07-26:
I just upgraded to bookworm and can confirm it is working with the 1500 hid.

jimklimov, 2024-12-18:
Testing desired with NUT v2.8.2 or newer (ideally a build of current git master) to check if:
* reasonable values are reported (fitting in range, etc.)
* commands and variable settings work
* OL/OB status does not fluctuate anymore