networkupstools/nut #571: ./configure --with-openssl fails with OpenSSL 1.1, SSL_library_init now a macro

Luo2Jie2, 2018-07-10:
When running ./configure --with-openssl on Debian Stretch.  The ./configure output ends with

   checking for OpenSSL version via pkg-config... 1.1.0f found
   checking for OpenSSL cflags...
   checking for OpenSSL ldflags... -lssl `-lcrypto`
   checking for openssl/ssl.h... yes
   checking for SSL_library_init... no
   configure: error: "OpenSSL not found, required for SSL support"

File ssl.h which defines SSL_library_init is present, but configure can't see the 
symbol SSL_library_init.

Quoting https://bugs.archlinux.org/task/53809  << in openssl 1.1, 
SSL_library_init is a preprocessor macro, not an actual symbol; so 
libesmtp's ./configure erroneously thinks that libssl isn't available, >>

Roger

Luo2Jie2, 2018-07-12:
Workaround (not very elegant):  
Look in script configure for the two lines which declare
   if test "x$ac_cv_func_SSL_library_init" = xyes; then :
and replace them with
   if test xyes = xyes; then :

Roger

zykh, 2018-07-12:
not commenting on the issue, but workaround for workaround: why not enabling caching (`./configure ... -C`) and override the cached value of the `ac_cv_func_SSL_library_init` var?

Luo2Jie2, 2018-07-12:
I never use the configure cache - I always recalculate everything everytime. 
Perhaps a bad habit, but I was once caught out by stuff hanging around in the 
cache.

Roger

Luo2Jie2, 2018-07-16:
It looks as if this is a duplicate of issue "openssl 1.1 support #429", April 2017, 
https://github.com/networkupstools/nut/issues/429 and covered by 
"Add support for openssl-1.1.0 #504", Nov 2017,
https://github.com/networkupstools/nut/pull/504

Roger

closed 2021-11-14 by jimklimov