networkupstools/nut #572: NSS on Debian Stretch with libnss3: Can not initialize SSL context

Luo2Jie2, 2018-07-10:
I am unable to get 2.7.4 to work with NSS support as provided by 
libnss3 2:3.26.2 on Debian stretch. Currently NSS supports two 
database formats identified by prefixes "sql:" for the new database 
and "dbm:" for the legacy database, but there is no way of specifying 
them in 2.7.4.

I created the NSS database in directory /etc/nut with command
   certutil -N -d dbm:NSS_db --empty-password

I copied over public key and certificate from a working NUT+OpenSSL 
installation and checked them as follows:  List certificates:
  root@gold /etc/nut # certutil -L -d dbm:NSS_db
  Certificate Nickname       Trust Attributes
                                          SSL,S/MIME,JAR/XPI
  gold                                 Cu,u,u

Check certificates:
  root@gold /etc/nut → certutil -K -d dbm:NSS_db
  certutil: Checking token "NSS Certificate DB" in slot
            "NSS User Private Key and Certificate Services"
  < 0> rsa      df7b376946c8cfe59d74095dfc4b882d081b981b   gold

File upsd.conf is
  LISTEN 0.0.0.0 3493
  CERTPATH /etc/nut/NSS_db
  CERTIDENT gold sekret

Command systemctl start nut-server produces the message

Jul 10 15:02:51 gold upsd[15961]: Connected to UPS [heartbeat]: dummy-ups-heartbeat
Jul 10 15:02:51 gold upsd[15961]: Connected to UPS [Eaton]: usbhid-ups-Eaton
Jul 10 15:02:51 gold upsd[15961]: listening on 0.0.0.0 port 3493
Jul 10 15:02:51 gold upsd[15962]: Startup successful
Jul 10 15:02:51 gold upsd[15962]: Can not initialize SSL context

It looks as if legacy NSS_Init in netssl.c line 442 is unable 
to access the libnss3 NSS_db database.

Roger

Luo2Jie2, 2018-08-12:
On Sun, 12 Aug 2018, stratege1401 wrote:

> i think we have a related bug. : #563

Agreed - #563 and #572 look like the same libnss3 problem.  Roger

Gooberpatrol66, 2021-12-12:
I believe I am having this same problem.

trallnag, 2024-01-31:
Same here. Hmm, I wonder if the OpenSSL integration works better. Too bad on Ubuntu / Debian nut is compiled without OpenSSL support.