networkupstools/nut #573: Protocol error, Ubuntu 18.04, nut 2.7.4

howardtopher, 2018-07-11:
Currently rebuilding my system from Ubuntu 16.04 to Ubuntu 18.04.  Nut is unable to communicate with my Cyberpower UPS on 18.04, but it works fine on 16.04.  I installed nut through the default Ubuntu repositories (2.7.4-5).  I've copied configuration files over directly from the old system to the new one.

Debug mode when running usbhid-ups results in finding a matching device, but then having a protocol error when trying to communicate with it.

~# /lib/nut/usbhid-ups -a howardtopherups -DDD
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	debug level is '3'
   0.002225	upsdrv_initups...
   0.052708	Checking device (0764/0501) (002/003)
   0.071042	- VendorID: 0764
   0.071098	- ProductID: 0501
   0.071133	- Manufacturer: CP1500PFCLCD
   0.071170	- Product: CRCA102#BJ1
   0.071213	- Serial Number: CPS
   0.071249	- Bus: 002
   0.071285	- Device release number: 0001
   0.071321	Trying to match device
   0.071364	Device matches
   0.071483	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.072256	Unable to get HID descriptor (error sending control message: Protocol error)
   0.072891	HID descriptor length (method 1) -1
   0.075009	HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 ae 01
   0.075475	HID descriptor length (method 2) 430
   0.075899	HID descriptor length 430
   0.077447	Unable to get Report descriptor: Protocol error
   0.078237	No appropriate HID device found
   0.078258	No matching HID UPS found

clepple, 2018-07-12:
What if you install `libusb-1.0-0-dev`, then switch to the `libusb-1.0+0.1` branch and rebuild ?

See #300 for all of the gory details of that branch. You may also want to check the other bugs tagged CPS: https://github.com/networkupstools/nut/labels/CyberPower%20%28CPS%29

howardtopher, 2018-07-12:
There's definitely a difference in that branch, but still doesn't work.

clepple, 2018-07-13:
Do you see any kernel error messages around the same time in `dmesg` or `journalctl`?