networkupstools/nut #575: 2.7.4 upsclient.c, netssl.c use deprecated TLSv1_client_method()

Luo2Jie2, 2018-07-12:
On Debian stretch, after ./configure --with-openssl, make produces the following warning: 

libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../include -I../include -g -O2 -Wall -Wsign-compare -MT upsclient.lo -MD -MP -MF .deps/upsclient.Tpo -c upsclient.c  -fPIC -DPIC -o .libs/upsclient.o
upsclient.c: In function 'upscli_init':
upsclient.c:322:2: warning: 'TLSv1_client_method' is deprecated [-Wdeprecated-declarations]
  ssl_method = TLSv1_client_method();
  ^~~~~~~~~~
In file included from /usr/include/openssl/ct.h:13:0,
                 from /usr/include/openssl/ssl.h:61,
                 from upsclient.h:25,

The same warning appears at netssl.c:392.  I understand from 
https://users-cs.au.dk/sortie/sortix/release/nightly/man/man3/TLSv1_server_method.3.html
that version flexible TLS_server_method() is now preferred.  Quoting:

TLS_method(), TLS_server_method(), TLS_client_method()

These are the general-purpose version-flexible SSL/TLS methods. The actual protocol version used will be negotiated to the highest version mutually supported by the client and the server. The supported protocols are TLSv1, TLSv1.1 and TLSv1.2. Applications should use these methods and avoid the version-specific methods ....

Roger

opoplawski, 2020-05-18:
This should be resolved by #504

jimklimov, 2021-09-21:
I believe PR #1043 also addresses the subject, allowing newer TLS to be used where available.

jimklimov, 2021-11-14:
Is this issue (warning) still reproducible on any modern systems with NUT master branch code base?
And conversely, does any user of older systems (before openssl-1.1.0) have problems building with openssl support there?

In short, can the issue be closed? (...or forever hold your silence...)

Luo2Jie2, 2021-11-14:
On Sun, 14 Nov 2021, Jim Klimov wrote:

> Is this issue (warning) still reproducible on any modern systems with NUT master branch code base?
> And conversely, does any user of older systems (before openssl-1.1.0) have problems building with openssl support there?
> 
> In short, can the issue be closed? (...or forever hold your silence...)

If the issue has disappeared with the most recent build, then for me the issue 
can be closed.  Roger

jimklimov, 2021-11-14:
Well... I haven't seen this message in various builds during the "fightwarn" effort, so I guess it was resolved before last autumn (which matches those older PRs) :)

closed 2021-11-14 by jimklimov