networkupstools/nut #577: New Device: Tripp Lite OMNI1500LCDT supported by usbhid-ups

jackmanlabs, 2018-07-15:
I'd like to request help getting this device recognized by nut:
```
root@dorito:~# lsusb | grep 09ae
Bus 002 Device 013: ID 09ae:3016 Tripp Lite 
```

I used this configuration, hoping that the description was simply missing or mislabled in the compatibility database:
```
[tripplite]
        driver = usbhid-ups
        port = /dev/ttyS0
        desc = "Tripp Lite OMNI1500LCDT"
        vendorid = 09ae
```

I assumed this device would be supported by usbhid-ups because this device seemed like it would be a logical relative: https://networkupstools.org/ddl/Tripp_Lite/OMNI1000LCD.html

Per the usbhid-ups documentation, the port is an ignored value.

This is the output I get when I try to start the driver:
```
root@dorito:~# upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
No matching HID UPS found
Driver failed to start (exit status=1)
```

For good measure, here is my relevant udev configuration:
```
root@dorito:~# cat /etc/udev/rules.d/90-nut-ups.rules 
ACTION=="add", \
SUBSYSTEM=="usb", \
ATTR{idVendor}=="09ae", ATTR{idProduct}=="3016", \
MODE="0660", GROUP="nut"
```

Link to the HCL: http://www.networkupstools.org/stable-hcl.html?manufacturer=Tripp%20Lite&device-type=ups&connection=USB

Link to other Tripp Lite devices: https://networkupstools.org/ddl/Tripp_Lite/

I would be interested in helping as much as possible. Please let me know what information I can provide or work I might plausibly be able to perform. Thank you.

clepple, 2018-07-15:
The Tripp Lite devices with protocol 3016 have been nothing but trouble, especially with newer motherboards. (Older, pre-USB3 chipsets and ARM-based boards have had better results, but still not the kind of reliability you'd want for shutting down a critical system.)

Here is some background: https://github.com/networkupstools/nut/pull/122 (we don't plan to merge that approach, given the impact on non-Tripp-Lite devices, and the potential for races with other power software like DeviceKit Power)

Given the "no matching HID UPS found", you may not have hit the same error yet, but I suspect that it is similar hardware in the SMART*LCDT series as your OMNI1500LCDT. (Note that the protocol for the OMNI1000LCD (no T) is 2005 - see ups.productid in the variable list - so that one probably doesn't apply here.) Tripp Lite has not replied to questions about the 3016 protocol line.

To see if you have the same USB stability issues I was seeing with the SMART*LCDT, try running `lsusb -vvv -d 09ae:3016` a few times as root, and see if it can dump the HID Report Descriptor reliably (the long one, not the shorter HID Descriptor). You may need to run the NUT driver once, or something like `usbhid-dump`, to detach the kernel's HID driver so that the libusb backend in NUT can operate on the UPS.

The udev rule should change the permissions on the `/dev/bus/usb/` node for your UPS - can you check that? For the `lsusb` output you posted before, it would be `/dev/bus/usb/002/013`. You may need to do some combination of restarting udev and re-plugging the USB cable to get the permissions fixed.

jackmanlabs, 2018-07-16:
Hey! I appreciate your thorough and respectful reply. I will follow up with your suggestions when I get the chance, but I also understand that I shouldn't expect much, too. When I get a chance to fiddle around a bit, I will reply to this thread with the results. You may mark this closed if you desire. Have a great day!

robbiet480, 2018-07-20:
@clepple See #122 for important update on my situation