networkupstools/nut #585: tripplite_usb, Tripp Lite 3005 protocol

mbastiaan, 2018-07-25:
I have a couple of SMX500RT1U UPS's which use the 3005 protocol and the driver gives a couple of "debug arrays" with values, but the implementation of the 3005 protocol isn't that great at the moment. Based on the current data in the driver and the actual values I think I've managed to map most of the arrays as can be seen in the table below.

Debug array|1|2|3|4|5|6|7
-|-|-|-|-|-|-|-
debug.D: | xx | Input Voltage | xx | Battery Voltage x 10 | xx | xx | xx
debug.L: | Load % | xx | xx | xx | xx | xx | xx
debug.M: | xx | Input Voltage Minimum | xx | Input Voltage Maximum | xx | xx | xx
debug.P: ASCII? | Rating? | Rating? | Rating? | Rating? | Rating? | xx | xx
debug.S: | Low Battery (0 Low, 1 OK)? | Battery Test Status (0 OK, 4 Unknown) | xx | Status (0 Online, 1 On Battery, 4 Off) | Battery Charge | xx | xx	
debug.T: | Temperature C | xx | Frequency x 10 | Frequency x 10 | Freq Nominal | xx | xx
debug.V: | Input Voltage Nominal (certain value corresponds to certain voltage) | xx | Battery Voltage Nominal | Switchable Load Banks | xx | xx | xx

I expect things like the measured voltages to take 2 spaces in the array, so Input Voltage would be D1 and D2, like the frequency already does.
Is there anyone that can confirm if the table is correct or maybe has some additions?

MikeMark, 2018-08-01:
@mbastiaan 
From my investigation, that all looks correct. I'm testing on a TRIPP LITE SMART500RT1U.

Data from that unit:
```
Power on:
# upsc trippups
battery.voltage.nominal: 0
device.mfr: Tripp Lite
device.model: TRIPP LITE SMART500RT1U
device.type: ups
driver.name: tripplite_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.2
driver.version.internal: 0.20
input.voltage.nominal: 120
ups.debug.0: 30 05 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 00 75 00 89 0d 00 00 '.u.....'
ups.debug.L: 13 00 06 07 58 58 0d '....XX.'
ups.debug.load_banks: 0
ups.debug.M: 00 71 00 79 0d 00 00 '.q.y...'
ups.debug.P: 30 30 35 30 30 58 0d '00500X.'
ups.debug.S: 01 04 00 00 64 00 0d '....d..'
ups.debug.T: 2a 00 02 57 01 58 0d '...W.X.'
ups.debug.V: 02 00 0c 01 58 58 0d '....XX.'
ups.delay.shutdown: 64
ups.firmware: F29AF.D
ups.firmware.aux: protocol 3005
ups.id: 65535
ups.mfr: Tripp Lite
ups.model: TRIPP LITE SMART500RT1U
ups.power.nominal: 500
ups.productid: 0001
ups.status: 
ups.vendorid: 09ae

OFF:

# upsc trippups@localhost
battery.voltage.nominal: 0
device.mfr: Tripp Lite
device.model: TRIPP LITE SMART500RT1U
device.type: ups
driver.name: tripplite_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.2
driver.version.internal: 0.20
input.voltage.nominal: 120
ups.debug.0: 30 05 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 00 00 00 7d 0d 00 00 '.......'
ups.debug.L: 13 00 06 07 58 58 0d '....XX.'
ups.debug.load_banks: 0
ups.debug.M: 00 00 00 79 0d 00 00 '...y...'
ups.debug.P: 30 30 35 30 30 58 0d '00500X.'
ups.debug.S: 01 04 00 01 41 00 0d '....A..'
ups.debug.T: 2c 00 00 00 01 58 0d '.....X.'
ups.debug.V: 02 00 0c 01 58 58 0d '....XX.'
ups.delay.shutdown: 64
ups.firmware: F29AF.D
ups.firmware.aux: protocol 3005
ups.id: 65535
ups.mfr: Tripp Lite
ups.model: TRIPP LITE SMART500RT1U
ups.power.nominal: 500
ups.productid: 0001
ups.status: 
ups.vendorid: 09ae

[... 308 lines cut ...]
driver.parameter.port: auto
driver.version: 2.7.2
driver.version.internal: 0.20
input.voltage.nominal: 120
ups.debug.0: 30 05 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 00 74 00 80 0d 00 00 '.t.....'
ups.debug.L: 12 00 06 07 58 58 0d '....XX.'
ups.debug.load_banks: 0
ups.debug.M: 00 70 00 79 0d 00 00 '.p.y...'
ups.debug.P: 30 30 35 30 30 58 0d '00500X.'
ups.debug.S: 01 04 00 00 33 00 0d '....3..'
ups.debug.T: 32 00 02 57 01 58 0d '2..W.X.'
ups.debug.V: 02 00 0c 01 58 58 0d '....XX.'
ups.delay.shutdown: 64
ups.firmware: F29AF.D
ups.firmware.aux: protocol 3005
ups.id: 65535
ups.mfr: Tripp Lite
ups.model: TRIPP LITE SMART500RT1U
ups.power.nominal: 500
ups.productid: 0001
ups.status: 
ups.vendorid: 09ae

```

mbastiaan, 2018-08-03:
@MikeMark Thanks for the info.
I've also added debug.0, debug.F and debug.U to the table.

MikeMark, 2018-08-03:
@mbastiaan Thanks for the thanks.
Your additions look correct to me. Keep up the good work!

jimklimov, 2022-04-10:
CC @dzomaya