networkupstools/nut #586: HID UPS INNRT3000VA - shutdown.recover does nothing.

kvaster, 2018-07-27:
I have simpe setup - one PC and innova rt 3000 ups (ippon). I'm trying to set up server power up after power loss. Main problem is that shutdown.recover does not work - UPS does not respond to it.

`
/lib64/nut/usbhid-ups -a mainups -k
Network UPS Tools - Generic HID driver 0.42 (2.7.4-480-g14301bdb)
USB communication driver 0.33
Using subdriver: MGE HID 1.43
Initiating UPS shutdown
`

Nothing happens after this command.

ups.conf:
`
[mainups]
    driver = "usbhid-ups"
    port = "auto"
    vendorid = "06DA"
    productid = "FFFF"
    desc = "IPPON RT 3K"
`
I'm using latest git version:
`
 * Starting UPS drivers ...
Network UPS Tools - UPS driver controller 2.7.4-480-g14301bdb
Network UPS Tools - Generic HID driver 0.42 (2.7.4-480-g14301bdb)
USB communication driver 0.33
Using subdriver: MGE HID 1.43                                                                                                                                                                                 [ ok ]
`

upsc output:
`
battery.charge: 93
battery.charger.status: floating
battery.energysave: no
battery.runtime: 3409
battery.type: PbAc
device.mfr: PPC
device.model: HID UPS INNRT3000VA 
device.serial: R411D8Y00079   
device.type: ups
driver.name: usbhid-ups
driver.parameter.offdelay: 15
driver.parameter.ondelay: 20
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: FFFF
driver.parameter.synchronous: no
driver.parameter.vendorid: 06DA
driver.version: 2.7.4-480-g14301bdb
driver.version.data: MGE HID 1.43
driver.version.internal: 0.42
input.bypass.frequency: 49.9
input.bypass.voltage: 236.0
input.frequency: 49.9
input.frequency.nominal: 50
input.transfer.high: 276
input.transfer.low: 120
input.voltage: 236.0
input.voltage.nominal: 230
outlet.1.delay.shutdown: 2147483647
outlet.1.delay.start: 5
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.delay.shutdown: 2147483647
outlet.2.delay.start: 5
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 2
outlet.2.status: on
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.frequency: 49.9
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 15
ups.delay.start: 20
ups.firmware: 05517-1200
ups.load: 9
ups.load.high: 102
ups.mfr: PPC
ups.model: HID UPS INNRT3000VA 
ups.power: 271
ups.power.nominal: 3000
ups.productid: ffff
ups.realpower: 213
ups.realpower.nominal: 2700
ups.serial: R411D8Y00079   
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL CHRG
ups.temperature: 27.0
ups.test.interval: 2419200
ups.test.result: No test initiated
ups.timer.shutdown: 0
ups.timer.start: 0
ups.type: online
ups.vendorid: 06da
`

clepple, 2018-07-27:
If the UPS was still seeing line power (`ups.status: OL`) when you sent the `-k` command (which expands to either `shutdown.return`, or the combination of `load.on.delay` and `load.off.delay`), you may be seeing a variation of the FAQ 51: https://networkupstools.org/docs/FAQ.html

Some CPS units appear to round the `ups.delay.shutdown` and `ups.delay.start` values to minutes. There may be a similar issue with your brand of UPS. Try setting them to 60 and 120, respectively, and check the `ups.timer.*` variables to see if they change.

If neither of those ideas work, you can try running the driver with `-DD -k` to see if there are any USB errors that are not being logged to stderr. Please do not post the entire log inline - compressing it and attaching it is probably best, or you can excerpt the beginning and end.

kvaster, 2018-07-28:
There are no any usb errors. Also I've tried to launch power off without power on ups. Still nothing happens. Will try to discharge ups to 'low battery'...

```
ups.status: OB DISCHRG
ups.timer.shutdown: 0
ups.timer.start: 0
```

```
  0.048921     Initiating UPS shutdown
   0.048926     [D1] upsdrv_shutdown...
   0.048930     [D1] instcmd(shutdown.return, [NULL])
   0.048935     [D2] find_nut_info: unknown info type: shutdown.return
   0.048939     [D1] setvar(ups.start.auto, yes)
   0.049218     [D1] instcmd(load.on.delay, 120)
   0.049357     [D1] instcmd(load.off.delay, 60)
   0.049499     [D1] upsdrv_cleanup...
```

kvaster, 2018-07-30:
I've changed connection type from usb to rs232 and now ups is working as expected via blazer_ser driver. It shows much less info, but it works.

kvaster, 2018-11-16:
Hello again. Due to problems with rs232 (unreliable connection errors with my setup - connection lost messages) I decided to investigate problem with USB further.

Part of 'upsc' command output:
```
device.model: HID UPS INNRT3000VA
driver.version: 2.7.4-526-g2e6816be
driver.version.data: MGE HID 1.43
driver.version.internal: 0.42
outlet.1.delay.shutdown: 2147483647
outlet.1.delay.start: 5
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.delay.shutdown: 2147483647
outlet.2.delay.start: 5
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 2
outlet.2.status: on
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
```

As you can see outlet.switchable is 'no' for 'main' outlet (global?), but it is yes for outlet 1 and 2.

Output of 'upscmd -l' cmd:
```
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
outlet.1.load.off - Turn off the load on outlet 1 immediately
outlet.1.load.on - Turn on the load on outlet 1 immediately
outlet.2.load.off - Turn off the load on outlet 2 immediately
outlet.2.load.on - Turn on the load on outlet 2 immediately
```

Global load.off/on and load.off/on.delay commands are available and they produce OK reault, but they do really nothing... Maybe it's cause of global switchable = off ?

Basically 'shutdown.return' issues 'load.on.delay' following by 'load.off.delay' command in usbhid-ups.c.

As a result I've tried patching mge-hid.c (mge_hid2nut[]):
```
{ "outlet.1.load.off.delay", 0, 0, "UPS.OutletSystem.Outlet.[2].DelayBeforeShutdown", NULL, DEFAULT_OFFDELAY, HU_TYPE_CMD, NULL },
{ "outlet.1.load.on.delay", 0, 0, "UPS.OutletSystem.Outlet.[2].DelayBeforeStartup", NULL, DEFAULT_ONDELAY, HU_TYPE_CMD, NULL },
{ "outlet.2.load.off.delay", 0, 0, "UPS.OutletSystem.Outlet.[3].DelayBeforeShutdown", NULL, DEFAULT_OFFDELAY, HU_TYPE_CMD, NULL },
{ "outlet.2.load.on.delay", 0, 0, "UPS.OutletSystem.Outlet.[3].DelayBeforeStartup", NULL, DEFAULT_ONDELAY, HU_TYPE_CMD, NULL },
```

And replaced shutdown.recover command with following code:
```
usleep(1000000);
instcmd("outlet.2.load.on.delay", dstate_getinfo("ups.delay.start"));
usleep(1000000);
instcmd("outlet.2.load.off.delay", dstate_getinfo("ups.delay.shutdown"));
usleep(1000000);
instcmd("outlet.1.load.on.delay", dstate_getinfo("ups.delay.start"));
usleep(1000000);
instcmd("outlet.1.load.off.delay", dstate_getinfo("ups.delay.shutdown"));
```

This patches makes my ups working correct with shutting down power and reenabling it after.

It seems that this ups need to control outlet 1 and 2 directly. Also we need some time befire issuing first command - driver is not initialized yet ? And also we need delays between commands. In my experiments commands reported OK but only shutdown was accepted (don't know why). And all commands are working now after adding delays between them.

mdillmann, 2019-07-10:
Looks like the exact same problem I have. My UPS is named differently but vendorid/productid are the same.
Did you experience any negative side-effects from your modifications ? Because I would need to apply the same hack ...

Thanks,
-Markus

kvaster, 2019-07-10:
I have no negative impact. It's just working.

AlexanderHamza, 2021-09-26:
Hi, @kvaster I have exactly the same issue with PowerWalker VFI 2000 RT. Where I can find "shutdown.recover" command in order to change it as you suggested?

kvaster, 2021-09-27:
It's not command - it's an ugly patch for NUT, cause I don't know how to implement this properly...

[power-patch.txt](https://github.com/networkupstools/nut/files/7238206/power-patch.txt)

AlexanderHamza, 2021-09-27:
@kvaster anyway, the ugly patch is better than nothing:) Thank you.