networkupstools/nut #589: Eaton Ellipse ECO - usbhid-ups

alesnav, 2018-08-09:
Hello there,

I'm trying to use nut running on FreeBSD 11.1-RELEASE-p11 to connect to a Eaton Ellipse ECO 800 which is shown as fully supported in HCL, but it does not work in my case.

[image]

```
 # /usr/local/libexec/nut/usbhid-ups -u uucp -DDD -a eatontest
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.000490     upsdrv_initups...
   2.263739     Checking device (0463/FFFF) (/dev/usb//dev/ugen0.4)
   2.992547     - VendorID: 0463
   2.992569     - ProductID: ffff
   2.992600     - Manufacturer: EATON
   2.992605     - Product: Ellipse ECO
   2.992608     - Serial Number: 000000000
   2.992611     - Bus: /dev/usb
   2.992614     - Device release number: 0100
   2.992616     Trying to match device
   2.992634     Device matches
   2.992640     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   3.067951     HID descriptor, method 1: (9 bytes) => 09 21 10 10 21 01 22 8d 03
   3.067966     HID descriptor length (method 1) 909
   3.067973     HID descriptor, method 2: (9 bytes) => 09 21 10 10 21 01 22 8d 03
   3.067976     HID descriptor length (method 2) 909
   3.067979     HID descriptor length 909
   5.176835     Unable to get Report descriptor: Input/output error
   5.176889     No appropriate HID device found
   5.176928     No matching HID UPS found
```
`ups.conf` file (I have tried with `maxretry`, `retrydelay` and `usb_set_altinterface` with the same results
):

```
# cat /usr/local/etc/nut/ups.conf
[eatontest]
driver = usbhid-ups
port = /dev/ugen0.4
desc = "Ellipse ECO"
vendorid = 0463
productid = ffff
```

I also did the tests using two different USB cables to avoid a cable problem, getting the same result.

The output of `usbconfig` commands:

```
 # usbconfig -u 0 -a 4 dump_device_desc
ugen0.4: <EATON Ellipse ECO> at usbus0, cfg=0 md=HOST spd=LOW (1.5Mbps) pwr=ON (20mA)

  bLength = 0x0012
  bDescriptorType = 0x0001
  bcdUSB = 0x0200
  bDeviceClass = 0x0000  <Probed by interface class>
  bDeviceSubClass = 0x0000
  bDeviceProtocol = 0x0000
  bMaxPacketSize0 = 0x0008
  idVendor = 0x0463
  idProduct = 0xffff
  bcdDevice = 0x0100
  iManufacturer = 0x0001  <retrieving string failed>
  iProduct = 0x0002  <Ellipse ECO>
  iSerialNumber = 0x0004  <000000000>
  bNumConfigurations = 0x0006

# usbconfig -u 0 -a 4 dump_curr_config_desc
ugen0.4: <EATON Ellipse ECO> at usbus0, cfg=0 md=HOST spd=LOW (1.5Mbps) pwr=ON (20mA)

 Configuration index 0

    bLength = 0x0009
    bDescriptorType = 0x0002
    wTotalLength = 0x0022
    bNumInterfaces = 0x0001
    bConfigurationValue = 0x0001
    iConfiguration = 0x0000  <no string>
    bmAttributes = 0x00a0
    bMaxPower = 0x000a

    Interface 0
      bLength = 0x0009
      bDescriptorType = 0x0004
      bInterfaceNumber = 0x0000
      bAlternateSetting = 0x0000
      bNumEndpoints = 0x0001
      bInterfaceClass = 0x0003  <HID device>
      bInterfaceSubClass = 0x0000
      bInterfaceProtocol = 0x0000
      iInterface = 0x0000  <no string>

      Additional Descriptor

      bLength = 0x09
      bDescriptorType = 0x21
      bDescriptorSubType = 0x10
       RAW dump:
       0x00 | 0x09, 0x21, 0x10, 0x10, 0x21, 0x01, 0x22, 0x8d,
       0x08 | 0x03

     Endpoint 0
        bLength = 0x0007
        bDescriptorType = 0x0005
        bEndpointAddress = 0x0081  <IN>
        bmAttributes = 0x0003  <INTERRUPT>
        wMaxPacketSize = 0x0008
        bInterval = 0x0014
        bRefresh = 0x0000
        bSynchAddress = 0x0000

# usbconfig -u 0 -a 4 do_request 0x81 0x06 0x2200 0 0x100
REQUEST = <0x05 0x84 0x09 0x04 0xa1 0x01 0x09 0x10 0xa1 0x00 0x09 0x12 0xa1 0x00 0x09 0x5a 0x85 0x20 0x75 0x08 0x95 0x01 0x15 0x00 0x26 0xff 0x00 0x65 0x00 0x55 0x00 0xb1 0x82 0xc0 0x09 0x14 0xa1 0x00 0x09 0x02 0xa1 0x02 0x06 0xff 0xff 0x09 0x74 0x85 0x03 0xb1 0x83 0x09 0x74 0x81 0x83 0x09 0x75 0xb1 0x83 0x09 0x75 0x81 0x83 0xc0 0xc0 0xc0 0x05 0x84 0x09 0x1e 0xa1 0x84 0x09 0x43 0x85 0x0d 0x75 0x10 0x27 0xff 0xff 0x00 0x00 0x66 0x21 0xd1 0x55 0x07 0xb1 0x83 0x09 0x42 0x75 0x08 0x26 0xff 0x00 0x66 0x01 0xf0 0x55 0x00 0xb1 0x83 0x09 0x40 0x85 0x12 0x67 0x21 0xd1 0xf0 0x00 0x55 0x07 0xb1 0x82 0x09 0x1f 0x85 0x0b 0x65 0x00 0x55 0x00 0xb1 0x03 0xc0 0x06 0xff 0xff 0x09 0x1c 0xa1 0x00 0x09 0x1e 0xa1 0x81 0x09 0x98 0x85 0x18 0x75 0x10 0x27 0xff 0xff 0x00 0x00 0xb1 0x83 0x09 0x1f 0x85 0x17 0x75 0x08 0x26 0xff 0x00 0xb1 0x03 0xc0 0x09 0x1e 0xa1 0x82 0x09 0x98 0x85 0x18 0x75 0x10 0x27 0xff 0xff 0x00 0x00 0xb1 0x83 0x09 0x1f 0x85 0x17 0x75 0x08 0x26 0xff 0x00 0xb1 0x03 0xc0 0x09 0x1e 0xa1 0x83 0x09 0x98 0x85 0x18 0x75 0x10 0x27 0xff 0xff 0x00 0x00 0xb1 0x83 0x09 0x1f 0x85 0x17 0x75 0x08 0x26 0xff 0x00 0xb1 0x03 0xc0 0x09 0x1e 0xa1 0x84 0x09 0x98 0x85 0x18 0x75 0x10 0x27 0xff 0xff 0x00 0x00 0xb1 0x83 0x09 0x1f 0x85 0x17 0x75 0x08 0x26 0xff 0x00 0xb1 0x03 0xc0 0x09 0x1e 0xa1 0x85 0x09><Z u&eUttuuCu'f!UBu&fU@g!UeUu'u&u'u&u'u&u'u&>
```

Thanks,
Best regards

aquette, 2018-08-09:
Can you please try adding "-u root" to check if there's a privileges issue?

alesnav, 2018-08-09:
Yes, sure, I tested it too. The main difference is that if I use root as user, all USB ports are tested, so it seems that permissions works fine based on `/usr/local/etc/devd/nut-usb.conf` conditions:

```
#  various models  - usbhid-ups
notify 100 {
        match "system"          "USB";
        match "subsystem"       "DEVICE";
        match "type"            "ATTACH";
        match "vendor"          "0x0463";
        match "product"         "0xffff";
        action "chgrp uucp /dev/$cdev; chmod g+rw /dev/$cdev";
};
```
Thanks.

This is the output when using root user:

```
 # /usr/local/libexec/nut/usbhid-ups -u root -DDD -a eatontest
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.000401     upsdrv_initups...
   2.276172     Checking device (0463/FFFF) (/dev/usb//dev/ugen0.4)
   3.008139     - VendorID: 0463
   3.008169     - ProductID: ffff
   3.008203     - Manufacturer: EATON
   3.008208     - Product: Ellipse ECO
   3.008212     - Serial Number: 000000000
   3.008216     - Bus: /dev/usb
   3.008220     - Device release number: 0100
   3.008224     Trying to match device
   3.008291     Device matches
   3.008301     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   3.082913     HID descriptor, method 1: (9 bytes) => 09 21 10 10 21 01 22 8d 03
   3.082933     HID descriptor length (method 1) 909
   3.082941     HID descriptor, method 2: (9 bytes) => 09 21 10 10 21 01 22 8d 03
   3.082945     HID descriptor length (method 2) 909
   3.082949     HID descriptor length 909
   5.198286     Unable to get Report descriptor: Input/output error
   5.198339     Checking device (0E0F/0002) (/dev/usb//dev/ugen0.3)
   5.207737     - VendorID: 0e0f
   5.207751     - ProductID: 0002
   5.207756     - Manufacturer: unknown
   5.207760     - Product: VMware Virtual USB Hub
   5.207764     - Serial Number: unknown
   5.207768     - Bus: /dev/usb
   5.207772     - Device release number: 0100
   5.207775     Trying to match device
   5.207786     Device does not match - skipping
   5.207805     Checking device (0E0F/0003) (/dev/usb//dev/ugen0.2)
   5.230346     - VendorID: 0e0f
   5.230364     - ProductID: 0003
   5.230369     - Manufacturer: VMware
   5.230372     - Product: VMware Virtual USB Mouse
   5.230377     - Serial Number: unknown
   5.230381     - Bus: /dev/usb
   5.230385     - Device release number: 0102
   5.230389     Trying to match device
   5.230397     Device does not match - skipping
   5.230422     Checking device (0000/0000) (/dev/usb//dev/ugen0.1)
   5.230499     - VendorID: 0000
   5.230509     - ProductID: 0000
   5.230513     - Manufacturer: (0x15ad)
   5.230516     - Product: UHCI root HUB
   5.230520     - Serial Number: unknown
   5.230524     - Bus: /dev/usb
   5.230528     - Device release number: 0100
   5.230532     Trying to match device
   5.230537     Device does not match - skipping
   5.230549     Checking device (0000/0000) (/dev/usb//dev/ugen1.1)
   5.230582     - VendorID: 0000
   5.230586     - ProductID: 0000
   5.230590     - Manufacturer: (0x15ad)
   5.230594     - Product: EHCI root HUB
   5.230598     - Serial Number: unknown
   5.230601     - Bus: /dev/usb
   5.230605     - Device release number: 0100
   5.230609     Trying to match device
   5.230613     Device does not match - skipping
   5.230623     No appropriate HID device found
   5.230633     No matching HID UPS found
```

clepple, 2018-08-10:
This could be a corner case in VMware's USB stack, plus maybe the driver not knowing how to handle the generic "Input/output error".

Not sure whether it is easier for you to reproduce this on a physical box, or to rebuild NUT using a branch that is aware of libusb-1.0: 
https://github.com/networkupstools/nut/tree/libusb-1.0+0.1 (there may be some details in the related issue: https://github.com/networkupstools/nut/issues/300 but a quick read says maybe not applicable to your situation)

Either way, we will be moving to either that branch (or one substantially like it) for the next NUT release.

clepple, 2018-08-10:
Also, I'm short on time to look this up, but is the `usbconfig` command sending the same USB request as "Method 1" here? https://github.com/networkupstools/nut/blob/v2.7.4/drivers/libusb.c#L305

alesnav, 2018-08-12:
Hmmm, @clepple yes, it seems that it is related to VMware. I have just test with another VM running CentOS 7.5 and has the same result, plus a clear message in `dmesg`:

```
[   15.926550] usb 2-2.1: new low-speed USB device number 5 using uhci_hcd
[   18.667607] usb 2-2.1: New USB device found, idVendor=0463, idProduct=ffff
[   18.667612] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[   18.667615] usb 2-2.1: Product: Ellipse ECO
[   18.667617] usb 2-2.1: Manufacturer: EATON
[   18.667619] usb 2-2.1: SerialNumber: 000000000
[   22.903907] usbhid 2-2.1:1.0: can't add hid device: -71
[   22.904086] usbhid: probe of 2-2.1:1.0 failed with error -71
```

Will try with a Raspberry Pi to avoid VMware problems.

alesnav, 2018-08-15:
I have test it with a fresh Raspberry Pi using the same config files and it does work without any problem.

It should be a VMware problem...

```
root@raspberrypi:~# upsc eaton@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.runtime: 2087
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse ECO 800
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: no
driver.parameter.vendorid: 0463
driver.version: 2.7.4
driver.version.data: MGE HID 1.39
driver.version.internal: 0.41
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: on
outlet.2.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.power: 25
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02
ups.load: 13
ups.mfr: EATON
ups.model: Ellipse ECO 800
ups.power.nominal: 800
ups.productid: ffff
ups.serial: 000000000
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 0463
root@raspberrypi:~#
```

Thank you for your help.

closed 2018-08-15 by alesnav

GreaseMonkey88, 2020-02-26:
I have the same problem. I read somewhere that it is not possible in ESXi to passthrough USB HID devices.