networkupstools/nut #590: Support CyberPower RMCARD205

C4Wiz, 2018-08-13:
can you add support for this card please?

username115, 2019-01-20:
I would like to second this request

dogsbody, 2019-12-22:
+1

This card supports SNMP v1 & v3

The MIB file is [CyberPower_MIB_v2.9.MIB.TXT](https://github.com/networkupstools/nut/files/3993029/CyberPower_MIB_v2.9.MIB.TXT) which was downloaded (and renamed for uploading) from https://www.cyberpower.com/uk/en/product/sku/rmcard205#downloads

aquette, 2019-12-23:
Please look ar #416 to prépare the needed data. I'll do my best in january
ti look ar this.

tnagels, 2020-01-06:
@aquette I'd love to help with this as I am also in need of the update. I don't understand what you mean by referring to [#416](https://github.com/networkupstools/nut/issues/416), there is a ton of information there. What exactly do you expect from me/us?

aquette, 2020-01-07:
hi @tnagels 
look for the instructions around gen-snmp-subdriver.sh to provide me a base.
I'll then turns this into a branch like in https://github.com/networkupstools/nut/issues/416#issuecomment-554307433, so that you can test and we can progress

Update: note that we already have https://github.com/networkupstools/nut/blob/master/drivers/cyberpower-mib.c
I'd be first interested in seeing the results from this.
Also note that NUT 2.7.4 (last official release) only have version 0.1 of this driver, while the GIT version has 0.2, so it would be better to use GIT (in which case, use the above referenced comment to get and compile)

tnagels, 2020-01-07:
@aquette I will try to check this later this week on a separate system. The setup I am using now does not allow me to do this (it is in a managed container I have little control over).

Meanwhile, I'll also check the driver you referenced by just comparing the MIB entries with what I get when doing an snmp walk. If they match up there is a decent chance the driver will work i guess.

Thomas

aquette, 2020-01-07:
@tnagels ack. it's possible that this current driver needs some update and completion though. Time (and your tests) will tell.

freecableguy, 2020-02-22:
Would also appreciate seeing RMCARD205 support added.

cybrnook, 2020-05-07:
@aquette I have one of these cards in my UPS at home, so have free access to it anytime. However, trying to read above, and your reference to the link for #416 , I don't even know where to start to help you and the community. Just a normal Joe, so would need really step by step to do this.

foundbobby, 2020-06-01:
Is anyone on this thread actively working on providing the information @aquette requested to support RMCARD205? Would really like to see this card supported in NUT/homeassistant.io.

I don't have a lot of free time but can take a look if this isn't already in progress.

tnagels, 2020-06-30:
> @aquette I will try to check this later this week on a separate system. The setup I am using now does not allow me to do this (it is in a managed container I have little control over).

Unfortunately I'm not getting round to testing this. So while I'm still interested in this, unfortunately I won't be part of the solution.

cybrnook, 2020-06-30:
> Is anyone on this thread actively working on providing the information @aquette requested to support RMCARD205? Would really like to see this card supported in NUT/homeassistant.io.
> 
> I don't have a lot of free time but can take a look if this isn't already in progress.

I have the card, but need a simple guide to follow to get the info. Step by step :-(

DaftHonk, 2020-07-06:
Compiling from git with the 'cyberpower MIB 0.2' gives me:

````
root@nut:/usr/local/ups/sbin# upsc myups@localhost
Init SSL without certificate database
battery.charge: 100
battery.runtime: 417600
battery.runtime.elapsed: 0
device.mfr: CYBERPOWER
device.model: OR2200LCDRT2U
device.serial: 
device.type: ups
driver.name: snmp-ups
driver.parameter.pollfreq: 15
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.1.123
driver.parameter.snmp_version: v1
driver.parameter.synchronous: no
driver.version: 2.7.4-747-g3d88a972
driver.version.data: cyberpower MIB 0.2
driver.version.internal: 1.12
input.frequency: 60
input.voltage: 120
output.voltage: 120
ups.firmware: BFAD101*781
ups.load: 16
ups.mfr: CYBERPOWER
ups.model: OR2200LCDRT2U
ups.serial: 
ups.status: OL
````

and:

````
root@nut:/usr/local/ups/sbin# snmpwalk -c [private] -v1 [ip] .1.3.6.1.4.1.3808.1.1.1
iso.3.6.1.4.1.3808.1.1.1.1.1.1.0 = STRING: "OR2200LCDRT2U"
iso.3.6.1.4.1.3808.1.1.1.1.1.2.0 = STRING: "RMCARD205"
iso.3.6.1.4.1.3808.1.1.1.1.2.1.0 = STRING: "BFAD101*781"
iso.3.6.1.4.1.3808.1.1.1.1.2.3.0 = ""
iso.3.6.1.4.1.3808.1.1.1.1.2.4.0 = STRING: "1.2.0"
iso.3.6.1.4.1.3808.1.1.1.1.2.5.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.1.2.6.0 = INTEGER: 2000
iso.3.6.1.4.1.3808.1.1.1.1.2.7.0 = INTEGER: 1320
iso.3.6.1.4.1.3808.1.1.1.1.2.8.0 = INTEGER: 167
iso.3.6.1.4.1.3808.1.1.1.1.2.9.0 = STRING: "TALHZ2002868"
iso.3.6.1.4.1.3808.1.1.1.2.1.1.0 = INTEGER: 2
iso.3.6.1.4.1.3808.1.1.1.2.1.2.0 = Timeticks: (0) 0:00:00.00
iso.3.6.1.4.1.3808.1.1.1.2.1.3.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.2.1.4.0 = INTEGER: 0
iso.3.6.1.4.1.3808.1.1.1.2.2.1.0 = Gauge32: 100
iso.3.6.1.4.1.3808.1.1.1.2.2.2.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.2.2.3.0 = Gauge32: 0
iso.3.6.1.4.1.3808.1.1.1.2.2.4.0 = Timeticks: (696000) 1:56:00.00
iso.3.6.1.4.1.3808.1.1.1.2.2.5.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.2.2.8.0 = INTEGER: 24
iso.3.6.1.4.1.3808.1.1.1.2.2.9.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.3.1.1.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.3.2.1.0 = Gauge32: 1200
iso.3.6.1.4.1.3808.1.1.1.3.2.2.0 = Gauge32: 1200
iso.3.6.1.4.1.3808.1.1.1.3.2.3.0 = Gauge32: 1200
iso.3.6.1.4.1.3808.1.1.1.3.2.4.0 = Gauge32: 600
iso.3.6.1.4.1.3808.1.1.1.3.2.5.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.3.2.6.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.4.1.1.0 = INTEGER: 2
iso.3.6.1.4.1.3808.1.1.1.4.1.2.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.4.1.3.0 = STRING: "47~63 Hz"
iso.3.6.1.4.1.3808.1.1.1.4.2.1.0 = Gauge32: 1200
iso.3.6.1.4.1.3808.1.1.1.4.2.2.0 = INTEGER: 600
iso.3.6.1.4.1.3808.1.1.1.4.2.3.0 = Gauge32: 16
[... 73 lines cut ...]
iso.3.6.1.4.1.3808.1.1.1.5.2.18.0 = INTEGER: 10
iso.3.6.1.4.1.3808.1.1.1.6.2.1.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.6.2.2.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.6.2.3.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.6.2.5.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.6.2.6.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.7.2.2.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.7.2.3.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.7.2.4.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.7.2.6.0 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.7.2.7.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.7.2.8.0 = NULL
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.1.1 = INTEGER: 1
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.1.2 = INTEGER: 2
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.1.3 = INTEGER: 3
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.1.4 = INTEGER: 4
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.1.5 = INTEGER: 5
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.2.1 = INTEGER: 3
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.2.2 = INTEGER: 3
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.2.3 = INTEGER: 3
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.2.4 = INTEGER: 3
iso.3.6.1.4.1.3808.1.1.1.8.1.1.1.2.5 = INTEGER: 3
````

This is more information than provided by MIB 0.1 as it provides input and output voltages as well as system load.

berrmirt, 2020-07-23:
I have this card and would be glad to assist in anyway possible.  I note that on the CP website, they have MIB v2.9: [here](https://www.cyberpowersystems.com/product/software/mib-files/mib-v29/)

nesys, 2020-08-29:
Hi All,

any update on this? any luck with RMCARD205 with NUT? thanks

overhacked, 2020-09-20:
I have a pull request submitted to support this card: https://github.com/networkupstools/nut/pull/686

cybrnook, 2020-09-21:
> I have a pull request submitted to support this card: #686

Geez, you've had that PR open for a year and a half, and it's still not accepted?

DaftHonk, 2020-09-21:
There is also another PR that corrects some fields: #793

eliotg, 2020-10-05:
I have the RMCARD205, but am installing nut from apt vs. compiling so @DaftHonk 's solution doesn't work for me--even though I'm clearly experiencing the exact same problem (and am glad for hope to fix it!). 

I've copied CyberPower's MIB from @berrmirt 's link, and placed it in nut's search path: /usr/share/snmp/mibs. I've messed around with the other files and /etc/nut/ups.conf but nothing I do allows it to access the OEM MIB. General internet searches on "nut ups custom MIB" and such aren't proving useful.

While it would be great to get better support this popular device built in, my more urgent question is how to get nut to pick up a custom MIB?  I'm so close, and I need to get this figured out before I ship equipment to a mountain top across the Pacific!

KenwoodFox, 2020-12-13:
Is this in the next NUT release? When can we expect to see it in production and if not, can we find this config compiling from git?

DaftHonk, 2020-12-15:
> can we find this config compiling from git?

I successfully compiled from the `master` branch on amd64 with support for this card.

jimklimov, 2020-12-16:
Before a next official release (for distro packaging, etc), there is some unfinished business to take care of :(

Currently the master branch should be in decent shape, getting better week to week, with lots of changes since last 2.7.4 released a few years ago... so at the moment it is your best shot at supporting the card if this support does work for you (got none to check locally) and if distro-provided packages are not a blocking requirement :)

username115, 2021-02-19:
I just built master at commit 9defa7a9cc97491b4d595fe1233996b948ab730d on Debian amd64 and installed it over the debian package per the instructions on the wiki here.

The driver initializes with some warnings, but I am able to query the RMCARD 205 now. Thank you!

KenwoodFox, 2021-05-07:
> I just built master at commit [9defa7a](https://github.com/networkupstools/nut/commit/9defa7a9cc97491b4d595fe1233996b948ab730d) on Debian amd64 and installed it over the debian package per the instructions on the wiki here.
> 
> The driver initializes with some warnings, but I am able to query the RMCARD 205 now. Thank you!

where in the instructions is this? Id like to do exactly what you've done do you think you would be willing to direct to some more docs on what you did?

jimklimov, 2021-05-07:
Well, beside generic docs in the sources, there are compact specifics in
the wiki, e.g.
https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

Hope this helps, Jim

scyto, 2021-11-11:
I am also getting this with an RMCARD 205
Hardware Version1.1
Firmware Version1.1.6

```
"No matching MIB found for sysOID '.1.3.6.1.4.1.3808.1.1.1'!
Please report it to NUT developers, with an 'upsc' output for your device.
Going back to the classic MIB detection method.
Detected OR700LCDRM1U on host 192.168.1.71 (mib: cyberpower 0.1)
Network UPS Tools - UPS driver controller 2.7.4
```

It seems 'the old' method is detecting the UPS correctly unclear if this is right.  I have same 

I am running Network UPS Tools - UPS driver controller 2.7.4 (this is the home assitant nut UPS package) and seems be latest with this project.

Here is the request USPC output
```
root@a0d7b954-nut:/# upsc AV-Rack-Top@localhost
Init SSL without certificate database
battery.runtime: 82800.00
battery.runtime.elapsed: 0.00
device.mfr: CYBERPOWER
device.model: OR700LCDRM1U
device.serial: 
device.type: ups
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.1.71
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: cyberpower MIB 0.1
driver.version.internal: 0.97
ups.firmware: BF00701ICM1
ups.mfr: CYBERPOWER
ups.model: OR700LCDRM1U
ups.serial: 
ups.status: OL
```

C4Wiz, 2021-11-11:
Update your cards firmware

scyto, 2021-11-12:
@C4Wiz thanks, i just tried that, made no difference.  Card appears to report the same set of info on both SNMPv1 and v3.  I get all the details i expect to get from a APC management card so i think i have figured out configuration.

I note someone
[RMCARD205 OIDs + MIB WALK output.xlsx]
 was asking for OID / MIB data? 
If so maybe this is useful?
(i dont really know what i am doing)

(https://github.com/networkupstools/nut/files/7523910/RMCARD205.OIDs.%2B.MIB.WALK.output.xlsx)

scyto, 2021-11-12:
also i don't know what this is in the USPC output
```
battery.runtime: 82800.00
```
but it certainly isn't number of seconds remaining as the UPS has only 23 minutes of runtime (1,380 seconds...)

DaftHonk, 2021-11-12:
> ```
> battery.runtime: 82800.00
> ```
> 
> but it certainly isn't number of seconds remaining as the UPS has only 23 minutes of runtime (1,380 seconds...)
> 
> the mib walk i did returns the right value
> 
> ```
> Name/OID	 upsAdvanceBatteryRunTimeRemaining.0	 Value (TimeTicks)	 23 minutes  (138000)
> ```

Runtime is in 1/60 of a second (82800/60 = 1380). Not sure why, but that's the way it is.

scyto, 2021-11-12:
> Runtime is in 1/60 of a second (82800/60 = 1380). Not sure why, but that's the way it is.

thanks never thought to work in base 60 below the second level, lol.
I now need to figure out how (if) its possible to make home assistant see that as 60th of seconds.

Or just give up given how little of this device is exposed :-(

i got as far as creating a .c and .h file based on the mib and my live data (as per custom driver instructions)
then it got a bit complex as I am basically a non-coder, but i might try and keep going using WSL as compilation environment...

scyto, 2021-11-12:
i note from upsc i seem to be using driver.version.data: cyberpower MIB 0.1
how can i force the later version?

scyto, 2021-11-12:
@DaftHonk  i suspect the code in 

well for me it seems my NUT server is using the MIB0.1
yet https://github.com/networkupstools/nut/blob/master/drivers/cyberpower-mib.c clearly shows its been bumped to 0.3

i think i need to go pull a docker container of a later build and stop relying on the one in home assistant....

scyto, 2021-11-12:
cyberpower helpfully supplied me with this if someone is able to translate this into the .h and .c files... i also suggested to them they need to contribute....

[CyberPower_UPS related OID.pdf](https://github.com/networkupstools/nut/files/7524509/CyberPower_UPS.related.OID.pdf)

DaftHonk, 2021-11-12:
I built my own container for HA from master. Clone `https://github.com/hassio-addons/addon-nut.git` into the `addons` folder and change `nut/Dockerfile` to:

```
ARG BUILD_FROM=ghcr.io/hassio-addons/debian-base/amd64:4.2.0
FROM ${BUILD_FROM} as builder

RUN apt-get update \
    && apt-get install -y --no-install-recommends \
       build-essential \
       git \
       python \
       autoconf \
       automake \
       libtool \
       libneon27-dev \
       libsnmp-dev \
       libusb-dev \
       libltdl-dev

RUN git clone https://github.com/networkupstools/nut.git \
    && cd nut \
    && ./autogen.sh \
    && ./configure \
            --prefix=/usr \
            --libexecdir=/usr/lib/nut \
            --without-wrap \
            --with-user=root \
            --with-group=root \
            --disable-static \
            --with-serial \
            --with-usb \
            --without-avahi \
            --with-snmp \
            --with-neon \
            --without-powerman \
            --without-ipmi \
            --without-freeipmi \
            --with-libltdl \
            --without-cgi \
            --with-drvpath=/usr/lib/nut \
            --datadir=/usr/share/nut \
            --sysconfdir=/etc/nut \
            --with-statepath=/var/run/nut \
            --with-altpidpath=/var/run/nut \
    && make DESTDIR=/app install \
    && make DESTDIR=/app install-conf

FROM ${BUILD_FROM}

RUN \
    apt-get update \
    && apt-get install -y --no-install-recommends \
       usbutils \
       libtool \
       openssl \
       libsnmp30 \
       libneon27-gnutls \
       libfreeipmi17 \
       libipmimonitoring6 \
    && apt-get clean \
    && rm -rf /var/lib/apt/lists/*

# Copy root filesystem
COPY rootfs /

COPY --from=builder /app /

# Build arguments
ARG BUILD_ARCH
ARG BUILD_DATE
ARG BUILD_DESCRIPTION
ARG BUILD_NAME
ARG BUILD_REF
ARG BUILD_REPOSITORY
ARG BUILD_VERSION

# Labels
LABEL \
    io.hass.name="${BUILD_NAME}" \
    io.hass.description="${BUILD_DESCRIPTION}" \
    io.hass.arch="${BUILD_ARCH}" \
    io.hass.type="addon" \
    io.hass.version=${BUILD_VERSION} \
    maintainer="Dale Higgs <dale3h@addons.community>" \
    org.opencontainers.image.title="${BUILD_NAME}" \
    org.opencontainers.image.description="${BUILD_DESCRIPTION}" \
    org.opencontainers.image.vendor="Home Assistant Community Add-ons" \
    org.opencontainers.image.authors="Dale Higgs <dale3h@addons.community>" \
    org.opencontainers.image.licenses="MIT" \
    org.opencontainers.image.url="https://addons.community" \
    org.opencontainers.image.source="https://github.com/${BUILD_REPOSITORY}" \
    org.opencontainers.image.documentation="https://github.com/${BUILD_REPOSITORY}/blob/main/README.md" \
    org.opencontainers.image.created=${BUILD_DATE} \
    org.opencontainers.image.revision=${BUILD_REF} \
    org.opencontainers.image.version=${BUILD_VERSION}
```
You can likely update to the latest base image, I haven't gotten around to it yet and it's been working just fine. I haven't changed any other values or labels as I haven't published it anywhere. It should then show up as a local addon and you can build it. You may also need to change it obviously if you are using a different arch.

As far as I know, these changes in this thread are only in master and will not be released until v2.7.5, therefore they are not in any of the images or packages.

scyto, 2021-11-12:
@DaftHonk  thanks so much, now i understand, i saw it merged and just assumed it was in 2.7.4 - my bad, i can stop banging my head against the wall (and y'all)

It looks like the latest file in master has bunch more OIDS configured, is anyone planning to add more (like the environment sensors OIDs).

Thanks for giving me breadcrumb trail on how to override the add-on containers in hass.

scyto, 2021-11-12:
@DaftHonk just FYI, yeah base image needs to be 4.2.0, if it is 5.2.2 (or whatever latest is) then libsnmp30 fails to be found, i just reverted to 4.2.0 and build completes. 

I built the image, i am little confused how to make the add-on store see it?
[image]

DaftHonk, 2021-11-12:
You need to create a copy of the nut add-on in your `addons` folder then replace the Dockerfile. The add-on store should see it the n and you can build it from the store the same way you install a normal add-on.

This is getting a bit off topic; I believe HA has some decent docs on developing addons that should help.

scyto, 2021-11-12:
Changes look great.
Building required a lot of tweaking of the dockerfile, i think because frenck is in middle of refactoring.
for anyone else who needs it here is one for testing (note i had to change statepath, altpidpath and snmp libraries

![image](https://user-images.githubusercontent.com/11369180/141429090-d6eacce4-9a37-4998-a8e3-86abe2b41e9d.png
[Dockerfile,txt](https://github.com/networkupstools/nut/files/7525735/Dockerfile.txt)
)

jimklimov, 2021-11-14:
While the actual RMCARD support was added in NUT (master branch, not releases so far), this discussion changed into a lot of good points about setup of updated NUT with Home Assistant and Docker, as well as a copy of CyberPower OID PDF for people to follow up with.

Would someone better versed with HA please aggregate this trove of knowledge into a PR for new documentation files in NUT?

UntouchedWagons, 2021-12-20:
What's the status of supporting the RMCARD205? I'm interested in getting one for my CyberPower UPS and find communicating with it over USB to be rather unreliable.

scyto, 2021-12-20:
Works if you run a Custom install of nutup add-on.
So once a new version of the add-on with a new version of nutups it will work out of the box.
the card does need to be on the latest firmwares

dcglabs, 2022-04-14:
Latest RMCARD FW (1.3.5 - https://www.cyberpowersystems.com/product/software/firmware/rmcard205305-firmware-v135/ ) changed the ups.status OID when on ECO mode. It looks like prior firmware were setting status to onLine (2) when in ECO mode.
Changelog for 1.3.5 ( https://dl4jz3rbrsfum.cloudfront.net/documents/CyberPower_RN_RMCARD205_305-Firmware-v1.3.5.pdf )

- Corrects the OID value of the UPS Output Status when UPS is in ECO Mode.

Looking into MIB 2.9 - there is a specific code of 8 for onECO mode.
`upsBaseOutputStatus OBJECT-TYPE
   SYNTAX INTEGER  {
      unknown(1),
      onLine(2),
      onBattery(3),
      onBoost(4),
      onSleep(5),
      off(6),
      rebooting(7),
      onECO(8),
      onBypass(9),
      onBuck(10),
      onOverload(11)
   }`

However the current CPS SNMP driver does not have status option for ups status code 8:
`static info_lkp_t cyberpower_power_status[] = {
	{ 2, "OL", NULL, NULL },
	{ 3, "OB", NULL, NULL },
	{ 4, "OL BOOST", NULL, NULL },
	{ 5, "OFF", NULL, NULL },
	{ 7, "OL", NULL, NULL },
	{ 1, "NULL", NULL, NULL },
	{ 6, "OFF", NULL, NULL },
	{ 0, NULL, NULL, NULL }
} ;`

Thus when ups is on ECO mode using RMCARD FW 1.3.5 the ups.status is blank/unknown

jimklimov, 2022-04-19:
@dcglabs : hi, can you please take a look at PR #1381 - and if a local build of that branch would fix the status reading for your UPS?

dcglabs, 2022-04-20:
> @dcglabs : hi, can you please take a look at PR #1381 - and if a local build of that branch would fix the status reading for your UPS?

Will check it out.

MatejKovacic, 2022-09-22:
Sorry for my ignorance, but I wonder if it is possible to send commands to UPS device through NUT with HomeAssistand NUT integration? For instance, I would like to have a toggle button in HomeAssistant to toggle the beeper on my Samurai UPS (basically, I would like to sent command `upscmd samurai beeper.toggle`).
My rather funny story with this UPS is here: https://telefoncek.si/2022/09/2022-09-21-priklop-ups-na-homeassistant/ (use Google Translate).

MatejKovacic, 2022-09-23:
Sorry for my ignorance, but how can I call these scripts? Is there any quick example you can share with me?

P. S. I added some additional info on my blog post, for instance how to use `upsrw` and how to create a "virtual" sensor in HA to show UPS load in Watts not in percent only.

jimklimov, 2022-09-23:
They are binaries, consumers of libnutclient that render HTML on stdout. Extensively documented.

scyto, 2023-10-01:
> @dcglabs : hi, can you please take a look at PR #1381 - and if a local build of that branch would fix the status reading for your UPS?

I think this got into master and I am running it via a production package on my proxmox system, the status shows as empty will this prohibit things from working?

```
root@pve1:~# upsc StudyShelf@localhost
Init SSL without certificate database
battery.charge: 100
battery.current: 1.70
battery.runtime: 1140
battery.runtime.elapsed: 0
battery.voltage: 27.40
battery.voltage.nominal: 24
device.contact: alex
device.description: UPS SNMP Card
device.location: Study
device.mfr: CYBERPOWER
device.model: PR750LCD
device.serial: PQ6MS2000186
device.type: ups
driver.name: snmp-ups
driver.parameter.authProtocol: MD5
driver.parameter.mibs: cyberpower
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.1.73
driver.parameter.privProtocol: DES
driver.parameter.secLevel: authPriv
driver.parameter.snmp_version: v3
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: cyberpower MIB 0.51
driver.version.internal: 1.21
input.frequency: 60
input.voltage: 116.10
output.current: 1.70
output.frequency: 60
output.voltage: 116.10
ups.delay.reboot: 0
ups.delay.shutdown: 180
ups.delay.start: 0
ups.firmware: CR01102FBE1
ups.load: 42
ups.mfr: CYBERPOWER
ups.model: PR750LCD
ups.serial: PQ6MS2000186
ups.status: 
```

jimklimov, 2023-10-01:
The empty `ups.status` may be a problem for monitoring, and may be due to an issue fixed after NUT 2.8.0 release (so likely not reflected in packages yet), see #1432 - at least this was the hypothesis explored for #2077 (closed due to a workaround to use USB monitoring instead).

If you are able to build NUT from source, to run a driver from the build workspace and dump the data it collects - see e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests - it would be helpful to know if that table ordering issue is in fact the problem behind empty `ups.status`.

scyto, 2023-10-01:
> If you are able to build NUT from source,

I am not a programmer (like at all) but I did recently build my own linux kernels with patches and some dkms stuff - so can follow instructions :-)

my plan is to build this and test it in a debian userspace WSL2 machine - any reason that shouldn't work?
if that won't work i can spin up a debian barebones VM instead...

jimklimov, 2023-10-01:
WSL2 should be okay, more so for SNMP driver tests (USB/serial would be more problematic).

scyto, 2023-10-02:
Ok I am a little confused.

In my WSL i installed all the pre-reqs.

When i tried the `replaceing a nut deployment` method i noted it compiled everything and placed it in /usr/local/ups  i figured out i had to stop the running version and explicitly call the new version

I was out of ideas once i hit this

```
alex@BIGRIG:/usr/local/ups/sbin$ sudo ./upsd -F -D
Network UPS Tools upsd 2.8.0-2558-gae492a8da
   0.000000     fopen /var/state/ups/upsd.pid: No such file or directory
   0.000013     Could not find PID file '/var/state/ups/upsd.pid' to see if previous upsd instance is already running!
   0.000050     stat /usr/local/ups/etc/upsd.conf: No such file or directory
```
and after i copied good *.conf files i got this
```
alex@BIGRIG:/usr/local/ups/sbin$ sudo ./upsd -F -D
Network UPS Tools upsd 2.8.0-2558-gae492a8da
alex@BIGRIG:/usr/local/ups/sbin$ sudo ./upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0-2558-gae492a8da
Can't start /usr/local/ups/bin/snmp-ups: No such file or directory
```
so i did a make clean and tried the cihelper by running ` ./ci_build.sh inplace`

it looks like it might not have compiled SNMP - not sure:
```
* build SNMP drivers:   no
* build SNMP drivers with statically linked lib(net)snmp:       no
```
i think the ci helper ran a bunch of tests on the compiled items (and i guess i was supposed to edit something before i ran the cihelper?)

did it clean up after itself - or are the files still somewhere and runnable....

apologies for being high maitnerance here....  i feel so close to having got this right, but also so far....

should i

scyto, 2023-10-02:
i deleted my confused post about compiling, tl;dr `winner winner chicken dinner`

```
alex@BIGRIG:~/src/nut$ upsc studyshelf@localhost
Init SSL without certificate database
battery.charge: 100
battery.current: 1.30
battery.date: 03/01/2023
battery.runtime: 1860
battery.runtime.elapsed: 0
battery.voltage: 27.40
battery.voltage.nominal: 24
device.contact: alex
device.description: UPS SNMP Card
device.location: Study
device.mfr: CYBERPOWER
device.model: PR750LCD
device.serial: PQ6MS2000186
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: snmp-ups
driver.parameter.authProtocol: MD5
driver.parameter.mibs: cyberpower
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.1.73
driver.parameter.privProtocol: DES
driver.parameter.secLevel: authPriv
driver.parameter.snmp_version: v3
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.0-2558-gae492a8da
driver.version.data: cyberpower MIB 0.55
driver.version.internal: 1.30
input.frequency: 60
input.transfer.reason: noTransfer
input.voltage: 115.50
output.current: 1.30
output.frequency: 60
output.voltage: 115.50
ups.delay.reboot: 0
ups.delay.shutdown: 180
ups.delay.start: 0
ups.firmware: CR01102FBE1
ups.id: Shelf UPS
ups.load: 34
ups.mfr: CYBERPOWER
ups.model: PR750LCD
ups.serial: PQ6MS2000186
ups.status: OL
ups.temperature: 34
ups.test.date: 09/10/2023
ups.test.result: Ok
```
nice!

jimklimov, 2023-10-02:
Cool, so that (or some later :) ) fix nailed this issue! :)

jimklimov, 2023-10-02:
For deb packages, probably the most straightforward way at the moment would be to take Debian NUT package sources, which layer some recipes and patches on top of a NUT (release) tarball, and feed your tarball there (made on master branch checkout with `make dist`). Roll that somehow into a package then with `debuild` IIRC.

scyto, 2023-10-02:
ok, i will trey that, i already found the source, i see it has some git diff patches, i will see what i can figure out....  i also found the debian rules file.

I am in middle of filing bug with the maintainer of the package.... dunno how much love it gets, we will see.... but at least i can tell them its already fixed upstream :-)

alicancakil, 2025-03-25:
hello! what's the status on this?

I see `rmcard205` on this list: https://networkupstools.org/stable-hcl.html

Does it work properly now? 🤔

jimklimov, 2025-03-25:
According to comments above, should work with v2.8.1 or newer.

jimklimov, 2025-10-01:
@kwinz :  Yes, probably a new issue is needed. SysOID `.1.3.6.1.4.1.3808.1.1.5` is not known to drivers yet, the supported one currently is nearby (`.1` not `.5`): 
```
drivers/cyberpower-mib.c:#define CYBERPOWER_SYSOID                      ".1.3.6.1.4.1.3808.1.1.1"
```

I am sure there was some discussion about this a few months back, but can't find any issue or PR.

I think your best shot is to generate, test and contribute a new subdriver using this OID and further ones under it, see https://networkupstools.org/docs/developer-guide.chunked/new-drivers.html#snmp-subdrivers

So far it fell back to IETF standard MIB ("tripplite" uses it as an essential alias) to get at least something.