networkupstools/nut #594: install fails for Debian/Ubuntu: duplicate nutdrv_qx.8

clepple, 2018-08-15:
`make install` on Buildbot (part of `make distcheck` fails due to a duplicate `nutdrv_qx.8` in the install rule.

First seen in commit 8b75b0360559a917bdc375b6440d8441a2729395

 * [Ubuntu Xenial](http://buildbot.networkupstools.org/public/nut/builders/Ubuntu-xenial-x64/builds/288): [build log](
http://buildbot.networkupstools.org/public/nut/builders/Ubuntu-xenial-x64/builds/288/steps/compile/logs/stdio)
 * [Debian stretch](
http://buildbot.networkupstools.org/public/nut/builders/Debian-x64-gcc/builds/890): [build log](
http://buildbot.networkupstools.org/public/nut/builders/Debian-x64-gcc/builds/890/steps/compile/logs/stdio)

clepple, 2018-08-15:
@jimklimov I did not have a chance to look into the history of each of the commits that were squashed, but I tagged you since yours was the first name I saw. Feel free to reassign.

jimklimov, 2018-08-16:
I'll try to take a look when less in vacation mode...

Note there was also a PR that day which should have fixed delivery of man
pages broken some months before (which did fix us a packaging issue on OBS
when our 42ity fork was remerged to be closer to upstream/master, so our
equivalent solution was replaced by upstream's). It is a more likely cause
of disruption and error induction, at face value at least.

zykh, 2018-08-17:
...and guess who is the one to blame?

nutdrv_qx's manpages appear in two lists: serial and USB drivers, in order to be built in both cases, when we ./configure for only one of them... but this obviously means that, if we ./configure for both, we end up with dupes.

We were previously saved by automake, which splits the list of files to be installed in different calls if they are too much (from [this commit](http://git.savannah.gnu.org/cgit/automake.git/commit/?id=24abcf7fe42b7a6a03f10289643d818eae7e7951) through the `am__install_max` var, [later](http://git.savannah.gnu.org/cgit/automake.git/commit/?id=92a09f69728ee6db2438add8835ee683e921c9e3) lowered to 40), so the two nutdrv_qx entries ended up in different calls to install.
The 8b75b03 commit, by adding more manpages before nutdrv_qx's ones, pushed also the first occurrence to the second call to install, which already contained the other one.

Probably, $(sort)'ing the list of files to be built should solve the issue.
I'll look at it, tomorrow.

zykh, 2018-08-19:
https://github.com/networkupstools/nut/tree/issue-594 (with two additional, unrelated, fixes)
good to merge?

clepple, 2018-08-20:
@zykh thanks for tracking this down. Looks good to me!

closed 2018-08-20 by zykh