networkupstools/nut #598: Communications with UPS lost

kosfango, 2018-08-22:
Hi All,

I have FreeBSD installed:
FreeBSD server 11.1-BETA2 FreeBSD 11.1-BETA2 #0 r320065
and nut-2.7.4_8 prot port collections

I see in logs a lot of lines:
[skip]
Aug 22 18:47:27 server blazer_usb[77549]: Communications with UPS re-established
Aug 22 18:47:27 server upsd[77551]: UPS [Ippon] data is no longer stale
Aug 22 18:47:28 server upsmon[1374]: Communications with UPS ippon@localhost established
Aug 22 18:47:42 server blazer_usb[77549]: Communications with UPS lost: status read failed!
Aug 22 18:47:44 server upsd[77551]: Data for UPS [Ippon] is stale - check driver
Aug 22 18:47:48 server upsmon[1374]: Poll UPS [ippon@localhost] failed - Data stale
Aug 22 18:47:48 server upsmon[1374]: Communications with UPS ippon@localhost lost
Aug 22 18:47:53 server blazer_usb[77549]: Communications with UPS re-established
Aug 22 18:47:53 server upsd[77551]: UPS [Ippon] data is no longer stale
Aug 22 18:47:54 server upsmon[1374]: Communications with UPS ippon@localhost established

When communication is lost:
# upsc ippon@localhost
Error: Data stale

When all is fine I see next:
# upsc ippon@localhost
battery.charge: 100
battery.voltage: 13.10
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/cuau1
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 2.0
input.frequency: 49.9
input.frequency.nominal: 50
input.voltage: 225.4
input.voltage.fault: 225.8
input.voltage.nominal: 220
output.voltage: 225.8
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 13
ups.productid: 5161
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
ups.vendorid: 0665

but sometimes I see more shortly output:
# upsc ippon@localhost
battery.voltage: 13.10
battery.voltage.high: -1.08
battery.voltage.low: -0.87
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/cuau1
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.frequency: 49.9
input.voltage: 227.6
input.voltage.fault: 227.6
output.voltage: 227.6
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 14
ups.productid: 5161
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
ups.vendorid: 0665

My configs:
[ups.conf]
[Ippon]
driver    = "blazer_usb"
port      = "/dev/cuau1"
desc      = "Ippon Back Power Pro 600"

[upsd.conf]
LISTEN 127.0.0.1
LISTEN 192.168.1.1
MAXAGE 25

[upsmon.conf]
RUN_AS_USER root
MONITOR ippon@localhost 1 monmaster bsd master
MINSUPPLIES 1
SHUTDOWNCMD "/sbin/shutdown -h +0"
NOTIFYCMD "/usr/local/sbin/upssched"
POLLFREQ 5
POLLFREQALERT 5
HOSTSYNC 15
DEADTIME 25
POWERDOWNFLAG /etc/killpower
NOTIFYFLAG ONLINE      SYSLOG+EXEC
NOTIFYFLAG ONBATT      SYSLOG+EXEC
NOTIFYFLAG LOWBATT     SYSLOG
NOTIFYFLAG SHUTDOWN    SYSLOG
NOTIFYFLAG COMMOK     SYSLOG
NOTIFYFLAG COMMBAD    SYSLOG
RBWARNTIME 43200
NOCOMMWARNTIME 300
FINALDELAY 10

Any ideas?

Thanks

zykh, 2018-08-22:
> but sometimes I see more shortly output:

That's because, on startup, the driver can't always get a reply to the `F<cr>` megatec query from the device.

> Debug output:

Can you post also the output without altering the default `pollinterval` / `-i` option?

> Any ideas?

hmm... a possible source of problems could be that you're on FreeBSD with its libusb 0.1 (which doesn't return `-errno` on error like real libusb 0.1, and as the driver expects, but almost always `-1`), and using NUT with [this, wrong, patch](https://svnweb.freebsd.org/ports?view=revision&revision=452496) (which further distances the driver from a chance to get the real error), and if the driver can't identify the real error happened while reading, it can't react properly.
If possible, try the driver compiled (without patches) from this branch: https://github.com/networkupstools/nut/tree/libusb-compat-1.0

I would also give a chance to the other USB subdrivers (you're currently using the `cypress` one), starting from the `phoenix` one (again, it will not work exactly as expected because of the aforementioned FreeBSD's libusb 0.1/wrong patch combo, but, nonetheless, it should perform an interrupt read -which should fail- before sending the command to the device, and this could have some, maybe positive, effects).

Also, while it should not make a great difference, consider trying the `nutdrv_qx` driver.

kosfango, 2018-08-26:
> Can you post also the output without altering the default pollinterval / -i option?

Thank you for reply. I did new tests as you recommended:

/usr/local/libexec/nut/blazer_usb -a Ippon -DDD
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
   0.000000     debug level is '3'
   0.002744     Checking device (0A12/0001) (/dev/usb//dev/ugen2.2)
   0.002856     - VendorID: 0a12
   0.002865     - ProductID: 0001
   0.002871     - Manufacturer: unknown
   0.002876     - Product: unknown
   0.002892     - Serial Number: unknown
   0.002902     - Bus: /dev/usb
   0.002907     - Device release number: 1958
   0.002911     Trying to match device
   0.002922     Device does not match - skipping
   0.002945     Checking device (0665/5161) (/dev/usb//dev/ugen3.2)
   0.034945     - VendorID: 0665
   0.034965     - ProductID: 5161
   0.034972     - Manufacturer: Cypress Semiconductor
   0.034979     - Product: USB to Serial
   0.034985     - Serial Number: unknown
   0.034992     - Bus: /dev/usb
   0.034999     - Device release number: 0002
   0.035005     Trying to match device
   0.035024     Device matches
   0.035045     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.035083     Trying megatec protocol...
   0.037947     send: Q1
   0.406942     read: (229.4 229.4 229.4 013 49.9 13.1 25.0 00001001
   0.407040     Status read in 1 tries
   0.407049     Supported UPS detected with megatec protocol
   0.409927     send: F
   0.598929     read: #220.0 002 12.00 50.0
   0.598979     Ratings read in 1 tries
   0.601918     send: I
   1.582867     read: Unknown error
   1.582894     blazer_vendor: short reply
   1.582902     Vendor information read 1 failed
   1.585889     send: I
   2.566860     read: Unknown error
   2.566887     blazer_vendor: short reply
   2.566896     Vendor information read 2 failed
   2.569851     send: I
   3.550866     read: Unknown error
   3.550893     blazer_vendor: short reply
   3.550901     Vendor information read 3 failed
   3.550908     Vendor information unavailable
   3.550915     No values provided for battery high/low voltages in ups.conf

   3.550939     Using 'guestimation' (low: 10.400000, high: 13.000000)!
   3.550951     Battery runtime will not be calculated (runtimecal not set)
   3.553837     send: Q1
   4.619863     read: Unknown error
   4.619888     blazer_status: short reply
   4.619897     Communications with UPS lost: status read failed!
   4.620281     dstate_init: sock /var/db/nut/blazer_usb-Ippon open on fd 6
   4.622781     send: Q1
   5.603881     read: Unknown error
   5.603908     blazer_status: short reply
   5.603922     Communications with UPS lost: status read failed!
   6.625708     send: Q1
   7.605867     read: Unknown error
   7.605895     blazer_status: short reply
   7.605903     Communications with UPS lost: status read failed!
   8.626637     send: Q1
   9.606860     read: Unknown error
   9.606885     blazer_status: short reply
   9.606895     Communications with UPS lost: status read failed!
[... 118 lines cut ...]
   3.936936     No values provided for battery high/low voltages in ups.conf

   3.936958     Using 'guestimation' (low: -0.866667, high: -1.083333)!
   3.936971     Battery runtime will not be calculated (runtimecal not set)
   3.939862     send: Q1
   4.128880     read: P1H001
   4.128905     blazer_status: short reply
   4.128912     Communications with UPS lost: status read failed!
   4.129307     dstate_init: sock /var/db/nut/blazer_usb-Ippon open on fd 6
   4.131857     send: Q1
   5.112745     read: Unknown error
   5.112773     blazer_status: short reply
   5.112781     Communications with UPS lost: status read failed!
  19.133332     send: Q1
  20.113745     read: Unknown error
  20.113771     blazer_status: short reply
  20.113780     Communications with UPS lost: status read failed!
  34.133785     send: Q1
  35.114745     read: Unknown error
  35.114772     blazer_status: short reply
  35.114781     Communications with UPS lost: status read failed!
  49.139339     send: Q1
  50.355747     read: Unknown error
  50.355774     blazer_status: short reply
^C

kosfango, 2018-08-26:
> hmm... a possible source of problems could be that you're on FreeBSD with its libusb 0.1

I forgot to get more helpful info:

1. Many year ago all worked fine. However, about 2 years ago it broken. Sometimes it worked fine, sometimes I had "error data stale", but a couple weeks ago NUT lost my UPS forever. I used com port cable to connect my UPS. Probably the cable has become unusable, so I changed com to usb and reconfigured NUT, but I gotten the same issue.

2. I have another server with FreeBSD Stable+NUT+new Ippon with LCD like 600 back power office.
After a few hours system lost UPS (OS doesn't see UPS device) until re-plug usb cable. Reboot doesn't have effect. I used usbhid-ups driver.

Analyzing both  cases, I suspect that maybe something is broken in the OS itself

kosfango, 2018-08-26:
> If possible, try the driver compiled (without patches) from this branch: https://github.com/networkupstools/nut/tree/libusb-compat-1.0

I did git clone -b libusb-compat-1.0, then ./autogen.sh && ./configure --prefix=/opt/nut && make && make install, then I created symlink to nut.conf and tried to start...The same issue...

kosfango, 2018-08-26:
> I would also give a chance to the other USB subdrivers (you're currently using the cypress one), starting from the phoenix

Tried...The same issue

kosfango, 2018-08-28:
> I did git clone -b libusb-compat-1.0, then ./autogen.sh && ./configure --prefix=/opt/nut && make && make install, then I created symlink to nut.conf and tried to start...The same issue...

# /opt/nut/bin/blazer_usb -DDD -a Ippon
Network UPS Tools - Megatec/Q1 protocol USB driver 0.17 (2.7.4-605-g10d40386)
   0.000000     [D1] debug level is '3'
   0.002161     [D2] nut_libusb_open: checking device 1 of 8.
   0.002289     [D2] nut_libusb_open: - VendorID: 0000.
   0.002299     [D2] nut_libusb_open: - ProductID: 0000.
   0.002305     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.002311     [D2] nut_libusb_open: - Product: EHCI root HUB.
   0.002316     [D2] nut_libusb_open: - Serial Number: unknown.
   0.002325     [D2] nut_libusb_open: - Bus: 004.
   0.002331     [D2] nut_libusb_open: - Device release number: 0100.
   0.002336     [D2] nut_libusb_open: trying to match device...
   0.002357     [D2] nut_libusb_open: device does not match - skipping.
   0.002393     [D2] nut_libusb_open: checking device 2 of 8.
   0.002463     [D2] nut_libusb_open: - VendorID: 0000.
   0.002484     [D2] nut_libusb_open: - ProductID: 0000.
   0.002491     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.002496     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.002505     [D2] nut_libusb_open: - Serial Number: unknown.
   0.002510     [D2] nut_libusb_open: - Bus: 003.
   0.002516     [D2] nut_libusb_open: - Device release number: 0100.
   0.002521     [D2] nut_libusb_open: trying to match device...
   0.002529     [D2] nut_libusb_open: device does not match - skipping.
   0.002549     [D2] nut_libusb_open: checking device 3 of 8.
   0.002614     [D2] nut_libusb_open: - VendorID: 0000.
   0.002623     [D2] nut_libusb_open: - ProductID: 0000.
   0.002629     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.002645     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.002652     [D2] nut_libusb_open: - Serial Number: unknown.
   0.002657     [D2] nut_libusb_open: - Bus: 001.
   0.002663     [D2] nut_libusb_open: - Device release number: 0100.
   0.002668     [D2] nut_libusb_open: trying to match device...
   0.002675     [D2] nut_libusb_open: device does not match - skipping.
   0.002696     [D2] nut_libusb_open: checking device 4 of 8.
   0.002760     [D2] nut_libusb_open: - VendorID: 0000.
   0.002769     [D2] nut_libusb_open: - ProductID: 0000.
   0.002775     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.002791     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.002797     [D2] nut_libusb_open: - Serial Number: unknown.
   0.002803     [D2] nut_libusb_open: - Bus: 002.
   0.002808     [D2] nut_libusb_open: - Device release number: 0100.
   0.002814     [D2] nut_libusb_open: trying to match device...
   0.002821     [D2] nut_libusb_open: device does not match - skipping.
   0.002841     [D2] nut_libusb_open: checking device 5 of 8.
   0.002908     [D2] nut_libusb_open: - VendorID: 0000.
   0.002918     [D2] nut_libusb_open: - ProductID: 0000.
   0.002932     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.002938     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.002944     [D2] nut_libusb_open: - Serial Number: unknown.
   0.002950     [D2] nut_libusb_open: - Bus: 000.
   0.002955     [D2] nut_libusb_open: - Device release number: 0100.
   0.002963     [D2] nut_libusb_open: trying to match device...
   0.002971     [D2] nut_libusb_open: device does not match - skipping.
   0.002991     [D2] nut_libusb_open: checking device 6 of 8.
   0.018350     [D2] nut_libusb_open: - VendorID: 12d1.
   0.018390     [D2] nut_libusb_open: - ProductID: 1001.
   0.018398     [D2] nut_libusb_open: - Manufacturer: HUAWEI Technology.
   0.018407     [D2] nut_libusb_open: - Product: HUAWEI Mobile.
   0.018415     [D2] nut_libusb_open: - Serial Number: unknown.
   0.018423     [D2] nut_libusb_open: - Bus: 004.
   0.018431     [D2] nut_libusb_open: - Device release number: 0000.
   0.018444     [D2] nut_libusb_open: trying to match device...
   0.018456     [D2] nut_libusb_open: device does not match - skipping.
   0.018496     [D2] nut_libusb_open: checking device 7 of 8.
   0.050397     [D2] nut_libusb_open: - VendorID: 0665.
   0.050419     [D2] nut_libusb_open: - ProductID: 5161.
   0.050428     [D2] nut_libusb_open: - Manufacturer: Cypress Semiconductor.
   0.050437     [D2] nut_libusb_open: - Product: USB to Serial.
[... 48 lines cut ...]
  16.707440     [D3] read: Timeout
  16.707468     [D2] blazer_status: short reply
  16.707478     [D2] Status read 2 failed
  17.688423     [D3] flush: Timeout
  17.690758     [D3] send: D
  18.671437     [D3] read: Timeout
  18.671467     [D2] blazer_status: short reply
  18.671476     [D2] Status read 3 failed
  18.671485     [D2] Trying zinto protocol...
  19.652433     [D3] flush: Timeout
  19.654690     [D3] send: Q1
  20.715438     [D3] read: Timeout
  20.715465     [D2] blazer_status: short reply
  20.715475     [D2] Status read 1 failed
  21.696458     [D3] flush: Timeout
  21.698623     [D3] send: Q1
  22.679429     [D3] read: Timeout
  22.679457     [D2] blazer_status: short reply
  22.679467     [D2] Status read 2 failed
  23.660423     [D3] flush: Timeout
  23.662548     [D3] send: Q1
  24.843423     [D3] read: Timeout
  24.843450     [D2] blazer_status: short reply
  24.843460     [D2] Status read 3 failed
  24.843467     No supported UPS detected

kosfango, 2018-08-28:
One of successfuly attempts:

/opt/nut/bin/blazer_usb -DDD -a Ippon
Network UPS Tools - Megatec/Q1 protocol USB driver 0.17 (2.7.4-605-g10d40386)
   0.000000     [D1] debug level is '3'
   0.002521     [D2] nut_libusb_open: checking device 1 of 8.
   0.002669     [D2] nut_libusb_open: - VendorID: 0000.
   0.002682     [D2] nut_libusb_open: - ProductID: 0000.
   0.002690     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.002698     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.002705     [D2] nut_libusb_open: - Serial Number: unknown.
   0.002713     [D2] nut_libusb_open: - Bus: 003.
   0.002720     [D2] nut_libusb_open: - Device release number: 0100.
   0.002728     [D2] nut_libusb_open: trying to match device...
   0.002741     [D2] nut_libusb_open: device does not match - skipping.
   0.002772     [D2] nut_libusb_open: checking device 2 of 8.
   0.002870     [D2] nut_libusb_open: - VendorID: 0000.
   0.002882     [D2] nut_libusb_open: - ProductID: 0000.
   0.002890     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.002898     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.002905     [D2] nut_libusb_open: - Serial Number: unknown.
   0.002913     [D2] nut_libusb_open: - Bus: 000.
   0.002920     [D2] nut_libusb_open: - Device release number: 0100.
   0.002928     [D2] nut_libusb_open: trying to match device...
   0.002937     [D2] nut_libusb_open: device does not match - skipping.
   0.002963     [D2] nut_libusb_open: checking device 3 of 8.
   0.003040     [D2] nut_libusb_open: - VendorID: 0000.
   0.003052     [D2] nut_libusb_open: - ProductID: 0000.
   0.003060     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.003068     [D2] nut_libusb_open: - Product: EHCI root HUB.
   0.003076     [D2] nut_libusb_open: - Serial Number: unknown.
   0.003083     [D2] nut_libusb_open: - Bus: 004.
   0.003091     [D2] nut_libusb_open: - Device release number: 0100.
   0.003098     [D2] nut_libusb_open: trying to match device...
   0.003107     [D2] nut_libusb_open: device does not match - skipping.
   0.003131     [D2] nut_libusb_open: checking device 4 of 8.
   0.003206     [D2] nut_libusb_open: - VendorID: 0000.
   0.003217     [D2] nut_libusb_open: - ProductID: 0000.
   0.003225     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.003233     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.003240     [D2] nut_libusb_open: - Serial Number: unknown.
   0.003248     [D2] nut_libusb_open: - Bus: 002.
   0.003255     [D2] nut_libusb_open: - Device release number: 0100.
   0.003263     [D2] nut_libusb_open: trying to match device...
   0.003271     [D2] nut_libusb_open: device does not match - skipping.
   0.003296     [D2] nut_libusb_open: checking device 5 of 8.
   0.003371     [D2] nut_libusb_open: - VendorID: 0000.
   0.003382     [D2] nut_libusb_open: - ProductID: 0000.
   0.003390     [D2] nut_libusb_open: - Manufacturer: Intel.
   0.003397     [D2] nut_libusb_open: - Product: UHCI root HUB.
   0.003405     [D2] nut_libusb_open: - Serial Number: unknown.
   0.003413     [D2] nut_libusb_open: - Bus: 001.
   0.003420     [D2] nut_libusb_open: - Device release number: 0100.
   0.003427     [D2] nut_libusb_open: trying to match device...
   0.003436     [D2] nut_libusb_open: device does not match - skipping.
   0.003461     [D2] nut_libusb_open: checking device 6 of 8.
   0.018818     [D2] nut_libusb_open: - VendorID: 12d1.
   0.018840     [D2] nut_libusb_open: - ProductID: 1001.
   0.018850     [D2] nut_libusb_open: - Manufacturer: HUAWEI Technology.
   0.018858     [D2] nut_libusb_open: - Product: HUAWEI Mobile.
   0.018865     [D2] nut_libusb_open: - Serial Number: unknown.
   0.018873     [D2] nut_libusb_open: - Bus: 004.
   0.018881     [D2] nut_libusb_open: - Device release number: 0000.
   0.018889     [D2] nut_libusb_open: trying to match device...
   0.018900     [D2] nut_libusb_open: device does not match - skipping.
   0.018940     [D2] nut_libusb_open: checking device 7 of 8.
   0.050470     [D2] nut_libusb_open: - VendorID: 0665.
   0.050499     [D2] nut_libusb_open: - ProductID: 5161.
   0.050510     [D2] nut_libusb_open: - Manufacturer: Cypress Semiconductor.
   0.050524     [D2] nut_libusb_open: - Product: USB to Serial.
[... 101 lines cut ...]
  32.657941     [D2] blazer_status: short reply
  33.676914     [D3] flush: Timeout
  33.679257     [D3] send: Q1
  34.659908     [D3] read: Timeout
  34.659940     [D2] blazer_status: short reply
  35.677899     [D3] flush: Timeout
  35.680186     [D3] send: Q1
  36.741910     [D3] read: Timeout
  36.741943     [D2] blazer_status: short reply
  37.722910     [D3] flush: Timeout
  37.725110     [D3] send: Q1
  38.885907     [D3] read: Timeout
  38.885938     [D2] blazer_status: short reply
  39.866906     [D3] flush: Timeout
  39.869040     [D3] send: Q1
  41.029913     [D3] read: Timeout
  41.029945     [D2] blazer_status: short reply
  42.010908     [D3] flush: Timeout
  42.013959     [D3] send: Q1
  42.994909     [D3] read: Timeout
  42.994940     [D2] blazer_status: short reply
  44.012911     [D3] flush: Timeout
  44.015893     [D3] send: Q1
  44.996923     [D3] read: Timeout
  44.996951     [D2] blazer_status: short reply

zykh, 2018-09-03:
Thanks for the logs -- so the errors are actually timeouts... we could try increasing the allowed time, but I'm not so sure this will solve the issue.

Before the problems with the COM port (I guess nothing has been updated/changed in your system concurrently a couple of weeks ago, causing regressions), did you ever try using the device with USB?
Perhaps, the USB issues have always been there, just unnoticed, and they are not related to the problems with the COM port... or, maybe, the device is just dying (you know... nothing lasts forever, even cold november rain...).
Either way, at first glance, I'm inclined to think that the problem is the device itself, not the OS, but nonetheless, do you have a chance to test it with a Linux-ish system?

Also, I think the other problem you mentioned with usbhid-ups is not related: feel free to open a new issue just for that if you still experience it.

dmilith, 2021-06-30:
Same on FreeBSD 12.2… Wonder if anyone tried on 13.x?

kosfango, 2021-06-30:
> Same on FreeBSD 12.2… Wonder if anyone tried on 13.x?

If you read this thread, I had two different issues, fist with "classic ippon" the second with usbhid-ups based ippon with lcd screen. So first issue was in Ippon com/usb - interface died and I replaced the UPS to another. However, issue with usbhid-ups based Ippon is still unsolved. I didn't tested it on 12 or 13 FreeBSD. I suspect that issue in the driver...

dmilith, 2021-06-30:
@kosfango In my case it's RPI2 on FreeBSD 12.2, I'm using "ETA UPS 1600 sinus", which is blazer_usb-compatible since it uses Megatec protocol.

The same blaze_usb driver works nicely on other RPI3 using the Debian system… so libusb-issue idea may also be a thing here.

xtronics, 2022-01-16:
I have had the same issue on two Debian boxs running bullseye now - 

`Jan 16 06:52:21 host-1 upscode2[1998]: Communications with UPS lost: Empty value from UPS for UPDS MOFFF!
Jan 16 06:52:21 host-1 upsd[2000]: Data for UPS [host-1] is stale - check driver
Jan 16 06:52:24 host-1 upsmon[2003]: Poll UPS [host-1@localhost] failed - Data stale
Jan 16 06:52:24 host-1 upsmon[2003]: Communications with UPS host-1@localhost lost
Jan 16 06:52:26 host-1 upscode2[1998]: Communications with UPS re-established
Jan 16 06:52:26 host-1 upsd[2000]: UPS [host-1] data is no longer stale
Jan 16 06:52:44 host-1 upsmon[2003]: Communications with UPS host-1@localhost established

Jan 15 23:52:12 host-2 upscode2[2363]: Communications with UPS lost: Empty string from UPS for UPDS!
Jan 15 23:52:20 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:52:40 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:52:52 host-2 upscode2[2363]: Communications with UPS lost: Empty string from UPS for UPDS!
Jan 15 23:53:00 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:53:00 host-2 upsmon[2368]: UPS host-2@localhost is unavailable
Jan 15 23:53:20 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:53:32 host-2 upscode2[2363]: Communications with UPS lost: Empty string from UPS for UPDS!
Jan 15 23:53:40 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:54:00 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:54:12 host-2 upscode2[2363]: Communications with UPS lost: Empty string from UPS for UPDS!
Jan 15 23:54:20 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:54:40 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:54:52 host-2 upscode2[2363]: Communications with UPS lost: Empty string from UPS for UPDS!
Jan 15 23:55:00 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:55:20 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:55:32 host-2 upscode2[2363]: Communications with UPS lost: Empty string from UPS for UPDS!
Jan 15 23:55:40 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:56:00 host-2 upsmon[2368]: Poll UPS [host-2@localhost] failed - Data stale
Jan 15 23:56:12 host-2 upscode2[2363]: Warning: excessive comm failures, limiting error reporting
Jan 15 23:56:12 host-2 upscode2[2363]: Communications with UPS lost: Empty string from UPS for UPDS!`

I'm wondering if it is related to the load on the machine? - a partial workaround might be?: 
(I did both - but the problem is not constant so YMMV)

Increased pollfreq - I was wondering if the UPS itself was having trouble keeping up - not likely.. 
POLLFREQ 25

I increased the time delay a bit -
NOCOMMWARNTIME 400

But it could be that the bug is actually in the UPSs themselves (Powerware 5119)..  I've seen this problem show up sporadically for some years now..  But with the newer version in bullseye - more frequently now?

This is using USB-2-serial converters based on: 

0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC

These FTDI chips pretty much set the standard - never had any problems with them in the many other applications I've used them in.  

Hard to figure-out when it mostly works..

jimklimov, 2022-01-17:
> but with the newer version in bullseye

Did this refer to packaged NUT? If so, distros usually package (and patch
over) official releases, so likely NUT 2.7.4 which is a few years old now.
Hard work is actively in progress to release a good 2.7.5 as soon as we're
confident in it.

If you could test a local build of NUT master branch (now including
libusb-1.0 support, which should be more stable as a library than
libusb-0.1), and let us know if it works (at all? ;] better?) for you,
would be great.

Message ID: ***@***.***>
>

jimklimov, 2024-10-07:
Just in case, did any of the original posters here have a chance to test their devices with recent NUT releases (2.8.x or custom builds from master) and libusb-1.0? Did anything behave differently?