networkupstools/nut #599: TrippLite SMX1500LCDT + FreeBSD 11.2 + libusb-compat-1.0 requires interruptonly and root

clepple, 2018-08-23:
Some combination of the Tripp-Lite SMX1500LCDT (a 3016 protocol USB UPS), FreeBSD 11.2, and libusb-compat-1.0 (mentioned in #300) using FreeBSD's libusb20 API requires `interruptonly` and `-u root` to get past the initial device detection phase.

Source: https://alioth-lists.debian.net/pipermail/nut-upsuser/2018-August/011210.html

zykh, 2018-08-25:
In my opinion:
- for the `-u root` thing, you already answered: see commit message of 3bbf837f3009b794d51c5985ec3df9c1c4d807a8 (since the kernel driver is active, auto-detach of kernel driver is activated, and libusb_claim_interface() fails when trying to detach the kernel driver, as fails the subsequent direct call to libsub_detach_kernel_driver()),
- the other issue (as well as #600) should not be related to the 'libusb-compat-1.0' branch or to the use of FreeBSD's libusb 1.0 instead of their libusb 0.1: the [first post](https://alioth-lists.debian.net/pipermail/nut-upsuser/2018-August/011192.html) of the thread manifests the same problems with master + FreeBSD's libusb 0.1.

zykh, 2018-08-25:
So, we just shouldn't set auto-detach on FreeBSD... kernel driver active or not (and we should not fail if we can't detach it ourselves): https://lists.freebsd.org/pipermail/freebsd-usb/2016-March/014161.html

clepple, 2018-08-26:
Hmm, I thought the LBYL patch was sufficient - I had been running non-root for a while on FreeBSD 11.1 for the past few months with `libusb-1.0+0.1` @ bac7d0f2 (which contains ca3298f832d).

I saw your update to `libusb-compat-1.0` (10d40386a05) so I will try that on 11.1.

clepple, 2018-08-26:
10d4038 works fine on FreeBSD 11.1 with the libusb-1.0 backend, on an MGE Pulsar Evolution 500. Tested both non-root and root access (the former was right after a reboot). It occurs to me that I can probably bring that FreeBSD box over to a Tripp-Lite UPS for testing later.