networkupstools/nut #601: Report about Liebert Corp. PowerSure PSA UPS model PSA500MT3-230U

grumbler66rus, 2018-08-23:
Hello!
Because i found very old information about PowerSure PSA 500 (https://networkupstools.org/ddl/Liebert/PowerSure_PSA_500.html), and I don't known about reporting, I writes report here.

```
# upsd -V
Network UPS Tools upsd 2.6.5
# lsusb | grep Liebert
Bus 001 Device 005: ID 10af:0001 Liebert Corp. PowerSure PSA UPS
# cat /etc/nut/ups.conf
[PSA]
vendorid = 10AF
productid = 0001
driver = usbhid-ups
port = auto
# upsc PSA
battery.charge: 80
battery.charge.low: 38
battery.charge.warning: 38
battery.type: PbAc
battery.voltage: 12.9
battery.voltage.nominal: 12.0
device.mfr: Emerson Network Power
device.model: LiebertPSA
device.serial:           
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0001
driver.parameter.vendorid: 10AF
driver.version: 2.6.5
driver.version.data: Belkin HID 0.15
driver.version.internal: 0.37
input.frequency: 50.0
input.voltage: 232.2
output.voltage: 230.4
ups.load: 29
ups.mfr: Emerson Network Power
ups.model: LiebertPSA
ups.productid: 0001
ups.serial:           
ups.status: OL CHRG
ups.vendorid: 10af
```

clepple, 2018-08-23:
Thanks. Do shutdowns work? What do “upsrw” and “upscmd -l” show?

marksumm, 2020-09-30:
The previously reported issue with the driver crashing is still present.  You can be forgiven for believing that it might be working because upsc is able to return some information for a while until the driver crashes.  After that, only restarting the driver will fix it...

`$ upsc ups@localhost

...hangs...`

`grep -i ups /var/log/syslog

...
Sep 30 09:38:58 microserver upsmon[2349]: UPS ups@localhost is unavailable
Sep 30 09:41:14 microserver upsmon[2349]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out
Sep 30 09:43:29 microserver upsmon[2349]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out`

`Broadcast message from nut@microserver (somewhere) (Wed Sep 30 09:45:44 2020): 
                                                                               
UPS ups@localhost is unavailable`

`$ sudo service nut-driver restart
                                                                               
Broadcast message from nut@microserver (somewhere) (Wed Sep 30 09:46:21 2020): 
                                                                               
Communications with UPS ups@localhost established `

`$ upsc ups@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 38
battery.runtime: 32
battery.type: PbAc
battery.voltage: 13.8
battery.voltage.nominal: 12.0
device.mfr: Emerson Network Power
device.model: LiebertPSA
device.serial:                     
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0001
driver.parameter.synchronous: no
driver.parameter.vendorid: 10af
driver.version: 2.7.4
driver.version.data: Belkin/Liebert HID 0.17
driver.version.internal: 0.41
input.frequency: 49.9
input.voltage: 233.9
output.voltage: 234.4
ups.load: 27
ups.mfr: Emerson Network Power
ups.model: LiebertPSA
ups.productid: 0001
ups.serial:                     
ups.status: OL
ups.vendorid: 10af`

closed 2022-04-25 by jimklimov