networkupstools/nut #603: New Device: CyberPower OR2200LCDRTXL2U

yaroz, 2018-08-29:
Hello.  I'd like to request support for the CyberPower OR2200LCDRTXL2U.  

When I initially power on the UPS, the following shows up in "dmesg -wH"
```
[Aug29 19:45] usb 1-1.4: new low-speed USB device number 6 using dwc_otg
[  +0.166866] usb 1-1.4: New USB device found, idVendor=0764, idProduct=0601
[  +0.000017] usb 1-1.4: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[  +0.000008] usb 1-1.4: Product: OR2200LCDRTXL2U
[  +0.000009] usb 1-1.4: Manufacturer: CPS
[  +0.235536] hid-generic 0003:0764:0601.0004: hiddev0,hidraw0: USB HID v1.10 Device [CPS OR2200LCDRTXL2U] on usb-3f980000.usb-1.4/input0
[  +6.458797] usb 1-1.4: USB disconnect, device number 6
```

I found a site stating that nothing is grabbing the device in a certain time, so it disconnects.  I was able to keep it connected by issuing "upscdrvctl start"

my /etc/nut/ups.conf is simple
```
[UPS]
driver = usbhid-ups
port = auto
desc = "test"
```

When I do a "upsc UPS" I get Error: Connection failure: Connection refused

In /lib/udev/rules.d/52-nut-usbups.rules there is no rule for this product, which has a productID of 0601.  I attempted to add one, but that did not fix the issue.

Here is the result of  /lib/nut/usbhid-ups -DD -a UPS -x vendorid=0764* 2>&1
```
Network UPS Tools - Generic HID driver 0.38 (2.7.2)
USB communication driver 0.32
   0.000000     debug level is '2'
   0.003070     upsdrv_initups...
   0.005054     Checking device (0764/0601) (001/008)
   0.026230     - VendorID: 0764
   0.026884     - ProductID: 0601
   0.027535     - Manufacturer: CPS
   0.028159     - Product: OR2200LCDRTXL2U
   0.028781     - Serial Number: unknown
   0.029413     - Bus: 001
   0.030028     Trying to match device
   0.030749     Device matches
   0.031409     failed to claim USB device: could not claim interface 0: Device or resource busy
   0.032097     detached kernel driver from USB device...
   0.040011     HID descriptor length 635
   0.154242     Report Descriptor size = 635
   0.155536     Using subdriver: CyberPower HID 0.3
   0.158637     Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.158781     Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 2
   0.158999     Path: UPS.PowerSummary.iManufacturer, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 3
   0.159184     Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 4
   0.159394     Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x01, Offset: 32, Size: 8, Value: 3
   0.162885     Path: UPS.PowerSummary.RemainingTimeLimit, Type: Feature, ReportID: 0x04, Offset: 0, Size: 16, Value: 300
   0.166004     Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 20
   0.169252     Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 10
   0.172383     Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 100
   0.172542     Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x07, Offset: 8, Size: 8, Value: 5
   0.172758     Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x07, Offset: 16, Size: 8, Value: 10
   0.173173     Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x07, Offset: 24, Size: 8, Value: 1
   0.173399     Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x07, Offset: 32, Size: 8, Value: 2
   0.173657     Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x07, Offset: 40, Size: 8, Value: 100
   0.177007     Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.177146     Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.177363     Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x08, Offset: 8, Size: 16, Value: 6000
   0.177777     Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x08, Offset: 8, Size: 16, Value: 6000
   0.181006     Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x09, Offset: 0, Size: 8, Value: 24
   0.184263     Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 16, Value: 27.4
   0.184407     Path: UPS.PowerSummary.Voltage, Type: Input, ReportID: 0x0b, Offset: 0, Size: 16, Value: 27.4
   0.184628     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x0b, Offset: 16, Size: 1, Value: 1
   0.184815     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x0b, Offset: 17, Size: 1, Value: 0
[... 28 lines cut ...]
   0.220632     Path: UPS.ff01001d.ff010019.ff010020, Type: Feature, ReportID: 0x25, Offset: 0, Size: 1, Value: 0
   0.223916     Path: UPS.ff01001d.ff010019.ff010021, Type: Feature, ReportID: 0x2c, Offset: 0, Size: 1, Value: 0
   0.224055     Path: UPS.ff01001d.ff010019.ff010021, Type: Input, ReportID: 0x2c, Offset: 0, Size: 1, Value: 0
   0.227256     Path: UPS.ff01001d.ff01001a.ff010001, Type: Feature, ReportID: 0x26, Offset: 0, Size: 1, Value: 0
   0.230381     Path: UPS.ff01001d.ff01001a.ff010002, Type: Feature, ReportID: 0x27, Offset: 0, Size: 8, Value: 1
   0.233447     Path: UPS.ff01001d.ff01001b.ff010016, Type: Feature, ReportID: 0x28, Offset: 0, Size: 8, Value: 0
   0.233523     Path: UPS.ff01001d.ff01001b.ff010016, Type: Input, ReportID: 0x28, Offset: 0, Size: 8, Value: 0
   0.233632     Path: UPS.ff01001d.ff01001b.ff010018, Type: Feature, ReportID: 0x28, Offset: 8, Size: 8, Value: 0
   0.233858     Path: UPS.ff01001d.ff01001b.ff010018, Type: Input, ReportID: 0x28, Offset: 8, Size: 8, Value: 0
   0.237018     Path: UPS.ff01001d.ff01001b.ff010015, Type: Feature, ReportID: 0x29, Offset: 0, Size: 8, Value: 1
   0.237170     Path: UPS.ff01001d.ff01001b.ff010015, Type: Output, ReportID: 0x29, Offset: 0, Size: 8, Value: 1
   0.237458     Path: UPS.ff01001d.ff01001b.ff010017, Type: Feature, ReportID: 0x29, Offset: 8, Size: 8, Value: 0
   0.237722     Path: UPS.ff01001d.ff01001b.ff010017, Type: Output, ReportID: 0x29, Offset: 8, Size: 8, Value: 0
   0.241007     Path: UPS.ff01001d.ff01001b.ff010010, Type: Feature, ReportID: 0x2d, Offset: 0, Size: 1, Value: 0
   0.241148     Path: UPS.ff01001d.ff01001b.ff01001e, Type: Feature, ReportID: 0x2d, Offset: 1, Size: 1, Value: 0
   0.241366     Path: UPS.ff01001d.ff01001b.ff01001f, Type: Feature, ReportID: 0x2d, Offset: 2, Size: 1, Value: 0
   0.241552     Path: UPS.ff01001d.ff01001b.ff010010, Type: Input, ReportID: 0x2d, Offset: 0, Size: 1, Value: 0
   0.241689     Path: UPS.ff01001d.ff01001b.ff01001e, Type: Input, ReportID: 0x2d, Offset: 1, Size: 1, Value: 0
   0.241776     Path: UPS.ff01001d.ff01001b.ff01001f, Type: Input, ReportID: 0x2d, Offset: 2, Size: 1, Value: 0
   0.245010     Path: UPS.ff01001d.ff01001b.ff010013, Type: Feature, ReportID: 0x2a, Offset: 0, Size: 1, Value: 0
   0.248255     Path: UPS.ff01001d.ff01001b.ff010014, Type: Feature, ReportID: 0x2b, Offset: 0, Size: 1, Value: 0
   0.248394     Report descriptor retrieved (Reportlen = 635)
```

If I can provide anything else to assist in getting this working, please let me know.

yaroz, 2018-08-29:
This seems to be the same issue as #510 but a different model number.

jimklimov, 2021-11-07:
Should be supported after the merge of #538 (so with builds of current NUT master branch, or eventually with release 2.7.5 and newer).

closed 2021-11-07 by jimklimov