networkupstools/nut #614: Build warnings in current codebase

jimklimov, 2018-09-08:
I guess someone should take a look into analyzing those. To me most of them seem like legit problems.
Also I'd take a look at adding a `configure` option to make warnings fatal, and a Travis sub-job to test for that.

* Printing into badly sized buffers:
````
al175.c: In function ‘al175_do.constprop’:
al175.c:400:28: warning: ‘%2X’ directive output may be truncated writing between 2 and 4 bytes into a region of size between 3 and 5 [-Wformat-truncation=]
  snprintf(data+2, 6+1, "%2X%2X%2X", pr1, pr2, pr3);
                            ^~~
al175.c:400:24: note: directive argument in the range [0, 65535]
  snprintf(data+2, 6+1, "%2X%2X%2X", pr1, pr2, pr3);
                        ^~~~~~~~~~~
al175.c:400:2: note: ‘snprintf’ output between 7 and 11 bytes into a destination of size 7
  snprintf(data+2, 6+1, "%2X%2X%2X", pr1, pr2, pr3);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
````

````
bcmxcp.c: In function ‘setvar’:
bcmxcp.c:2253:114: warning: ‘snprintf’ output truncated before the last format character [-Wformat-truncation=]
                 snprintf(success_msg, sizeof(success_msg)-1, " Mains return minimum battery capacity set to %d %%", tmp);
                                                                                                                  ^
bcmxcp.c:2253:17: note: ‘snprintf’ output between 50 and 60 bytes into a destination of size 49
                 snprintf(success_msg, sizeof(success_msg)-1, " Mains return minimum battery capacity set to %d %%", tmp);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
bcmxcp.c: In function ‘init_outlet’:
bcmxcp.c:1019:73: warning: ‘snprintf’ output may be truncated before the last format character [-Wformat-truncation=]
   snprintf(outlet_name, sizeof(outlet_name)-1, "outlet.%d.delay.shutdown", num);
                                                                         ^
bcmxcp.c:1019:3: note: ‘snprintf’ output between 24 and 25 bytes into a destination of size 24
   snprintf(outlet_name, sizeof(outlet_name)-1, "outlet.%d.delay.shutdown", num);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
bcmxcp.c: In function ‘upsdrv_initinfo’:
bcmxcp.c:1483:59: warning: ‘.shutdown.return’ directive output may be truncated writing 16 bytes into a region of size between 9 and 18 [-Wformat-truncation=]
    snprintf(outlet_name, sizeof(outlet_name)-1, "outlet.%d.shutdown.return", res);
                                                           ^~~~~~~~~~~~~~~~
bcmxcp.c:1483:4: note: ‘snprintf’ output between 25 and 34 bytes into a destination of size 26
    snprintf(outlet_name, sizeof(outlet_name)-1, "outlet.%d.shutdown.return", res);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
bcmxcp.c:1487:68: warning: ‘snprintf’ output may be truncated before the last format character [-Wformat-truncation=]
    snprintf(outlet_name, sizeof(outlet_name)-1, "outlet.%d.load.off", res);
                                                                    ^
bcmxcp.c:1487:4: note: ‘snprintf’ output between 18 and 27 bytes into a destination of size 26
    snprintf(outlet_name, sizeof(outlet_name)-1, "outlet.%d.load.off", res);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
````

````
nutdrv_qx_voltronic.c: In function ‘voltronic_fault’:
nutdrv_qx_voltronic.c:2840:50: warning: ‘%s’ directive output may be truncated writing up to 511 bytes into a region of size 497 [-Wformat-truncation=]
   snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
                                                  ^~
nutdrv_qx_voltronic.c:2840:3: note: ‘snprintf’ output between 17 and 528 bytes into a destination of size 512
   snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nutdrv_qx_voltronic.c:2847:51: warning: ‘%s’ directive output may be truncated writing up to 511 bytes into a region of size 497 [-Wformat-truncation=]
    snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
                                                   ^~
nutdrv_qx_voltronic.c:2847:4: note: ‘snprintf’ output between 17 and 528 bytes into a destination of size 512
    snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nutdrv_qx_voltronic.c:2951:52: warning: ‘%s’ directive output may be truncated writing up to 511 bytes into a region of size 497 [-Wformat-truncation=]
     snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
                                                    ^~
nutdrv_qx_voltronic.c:2951:5: note: ‘snprintf’ output between 17 and 528 bytes into a destination of size 512
     snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nutdrv_qx_voltronic.c:3174:51: warning: ‘%s’ directive output may be truncated writing up to 511 bytes into a region of size 497 [-Wformat-truncation=]
    snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
                                                   ^~
nutdrv_qx_voltronic.c:3174:4: note: ‘snprintf’ output between 17 and 528 bytes into a destination of size 512
    snprintf(alarm, sizeof(alarm), "Unknown fault [%s]", item->value);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
````

* Somewhat similar: validly we expect a short number here, but with some input/data/memory/... errors can end up overflowing the print:
````
eaton-pdu-marlin-mib.c: In function ‘marlin_outlet_group_phase_fun’:
eaton-pdu-marlin-mib.c:193:39: warning: ‘%i’ directive output may be truncated writing between 1 and 10 bytes into a region of size 2 [-Wformat-truncation=]
     snprintf(marlin_scratch_buf, 3, "L%i", (outlet_group_nb -3));
                                       ^~
eaton-pdu-marlin-mib.c:193:37: note: directive argument in the range [1, 2147483644]
     snprintf(marlin_scratch_buf, 3, "L%i", (outlet_group_nb -3));
                                     ^~~~~
In file included from /usr/include/stdio.h:862:0,
                 from ../include/common.h:32,
                 from main.h:4,
                 from eaton-pdu-marlin-mib.h:27,
                 from eaton-pdu-marlin-mib.c:31:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 3 and 12 bytes into a destination of size 3
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
````

* And on a related subject:

````
snmp-ups.c: In function ‘match_sysoid’:
snmp-ups.c:1332:4: warning: ‘memset’ used with length equal to number of elements without multiplication by element size [-Wmemset-elt-size]
    memset(mib2nut_sysOID, 0, MAX_OID_LEN);
    ^~~~~~
````

* Unused var... tag as such for compiler to be quiet, or use it actually?
````
liebert-esp2.c:102:2: warning: ‘cmd_selftestres’ defined but not used [-Wunused-const-variable=]
  cmd_selftestres[] = { 1,148,2,1,13,165 }, /* SELF_TEST_RESULT */
  ^~~~~~~~~~~~~~~
liebert-esp2.c:101:2: warning: ‘cmd_battestres’ defined but not used [-Wunused-const-variable=]
  cmd_battestres[] = { 1,148,2,1,12,164 }, /* BATTERY_TEST_RESULT */
  ^~~~~~~~~~~~~~
liebert-esp2.c:100:2: warning: ‘cmd_bitfield7’ defined but not used [-Wunused-const-variable=]
  cmd_bitfield7[]  = { 1,148,2,1,7,159 }, /* AMBIENT_OVERTEMP (2) */
  ^~~~~~~~~~~~~
````

* Are these logical errors (value equals something), or legitimate comparisons (essentially if a flag is set)?
````
rhino.c: In function ‘ScanReceivePack’:
rhino.c:308:23: warning: logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses]
   if( !( SourceFail ) == RedeAnterior )
                       ^~
rhino.c:308:7: note: add parentheses around left hand side expression to silence this warning
   if( !( SourceFail ) == RedeAnterior )
       ^~~~~~~~~~~~~~~
       (              )
````

````
solis.c: In function ‘ScanReceivePack’:
solis.c:682:22: warning: logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses]
  if( !( SourceFail ) == SourceLast ) {
                      ^~
solis.c:682:6: note: add parentheses around left hand side expression to silence this warning
  if( !( SourceFail ) == SourceLast ) {
      ^~~~~~~~~~~~~~~
      (              )
````

zykh, 2018-09-08:
- nutdrv_qx_voltronic.c:
  If I recall correctly, these things should already be fixed (since circa jan 2017) in a nutdrv_qx development branch, which is waiting (it's been on hold for a year, by now) that 2.7.5 is out, to be rebased and to get the final touches.
  Anyway, the changes in that branch are quite big, and I'm already crying now, just as I think of all the conflicts I'll have to address because of the libusb changes... So, if possible, please don't try to solve these nutdrv_qx issues now.

Also, actually, I get a few more of these warnings...

jimklimov, 2021-11-14:
See also fightwarn effort and many tickets and PRs linked from #823

jimklimov, 2022-01-18:
As a short summary, over the past few years I built up the CI farms (at least 3) to check NUT codebase regularly, and scripted rules and recipes to help detect issues, and iteratively based on that data - refactored the codebase to avoid glaring mismatches. The low-hanging-fruit amounted to about 3k commits by now, to clear the self-proclaimed "medium" level of the warnings and gauge feasibility of the rest:
* Regarding C standards, "Strict C" is a much harder target to get right (in the view of both CLANG and GCC at least) than the "GNU C" dialect; by now the NUT CI farm passes some "Strict C" builds with `gcc` at least;
* Also regarding C standards, the C89/C90/ANSI lays some constraints incompatible with current NUT codebase. If someone does need a build on a(ntique) system only capable of that version, they are in for an adventure (PRs welcome then). Currently the NUT CI farm passes some builds in C89 or GNU89 mode, using modern toolkits capable of more than that and so spewing some condescending warnings (some of which we must ignore in those scenarios);
* CppCheck still has several hundred complaints remaining, and at least some of those seem legitimate;
* The "hard" level of warnings was not tackled much, so still has hundreds if not thousands of complaints, mostly in esoteric areas like spelled-out-numbers being implicitly `double` but assigned into `float` fields or variables.

closed 2022-03-12 by jimklimov