networkupstools/nut #616: Support some more Legrand devices

zykh, 2018-09-10:
This is to track the progress on the topic.
Reference: https://alioth-lists.debian.net/pipermail/nut-upsdev/2018-July/007392.html
 
- [x] `usbhid-ups`: add new dedicated subdriver
- [ ] `nutdrv_qx`: support devices which don't terminate their replies with the expected CR (#441)
- [x] `nutdrv_qx`: add VID:PID to `krauler` USB subdriver
- [x] `metasys`: add new devices + fix a few glitches
- [x] update HCL
- [ ] update DDL (defer to after release?)
- [x] get/publish protocol used by the new devices ('id code' >= 11) supported by the `metasys` driver: https://github.com/networkupstools/nut-website/commit/570c1e5e9f1fb4f82abef9359897b5159e88fc50
- [ ] (possibly) get/publish protocol used by the devices ('id code' < 11) already supported by the `metasys` driver

Anything else?

Current branch: https://github.com/zykh/nut/tree/issue-441+legrand_updated
(this is likely to cause some conflicts with the libusb branches, so I'd prefer to wait for #300 to be solved before rebasing and merging this one)

aquette, 2019-02-28:
@zykh do you think this one can still go into 2.7.5 if we want to release now?

zykh, 2019-03-02:
@aquette: sure, it's ready, but, as mentioned above, if 2.7.5 will contain one of the libusb branches, I'd prefer to (rebase and) merge this one after the libusb branch, to avoid possible conflicts there.

greyslater, 2021-07-24:
Hello together,
just bought a new Legrand KEOR PDU 800 and can't get it recognized by the drivers under version 2.7.4
I found compatibility (from 2016) with the KEOR Multiplug but I guess this was the previous Hardware version.
the new products seem to be different

Cheerz, Chris

jimklimov, 2021-07-28:
Seems this support was added after the 2.7.4 release, and a 2.7.5 is sadly
still pending.
If you can try building from sources available on Github, you can confirm
if that would support your device or some further work is needed.

nseries100, 2021-08-10:
Hi there, @greyslater I'm waiting for Keor 800 support too :)

greyslater, 2021-08-15:
@jimklimov 
unfortunately i can't but do we have any progress in general here on 2.7.5
as far as i understand @zykh already implemented it in the forked repository

greyslater, 2021-08-15:
hi @nseries100 - i came upon all this because i connected the Keor 800 to a QNAP NAS and it's not recognized
found out that QTS (QNAP OS) uses NUT under the hood

blecher-at, 2021-08-19:
Got this branch running on current master by resolving some of the conflicts.
removed metasys changes for now as there were too many conflicts, anything else built fine.
see https://github.com/networkupstools/nut/pull/1037

after compiling (on my raspian) using the guide here https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

`./configure --includedir=/usr/include --mandir=/usr/share/man --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode --disable-silent-rules --libdir=/usr/lib/$(gcc -print-multiarch) --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/$(gcc -print-multiarch)/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-usb --disable-dependency-tracking
`
and then overwriting the binaries 
`cp drivers/usbhid-ups /usr/lib/nut/usbhid-ups` 

i was able to use my Legrand KEOR PDU 800
still need to figure out the metasys (don't have such a device), or split the issue, and check if anything else needs to be done on this PR.

reitermarkus, 2021-08-28:
My Legrand KEOR PDU 800 is also working (reporting values) with @blecher-at's updated branch.

deepbluese, 2021-11-03:
Next Problem:

Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.43 (2.7.4-3009-gd831a9ae)
USB communication driver 0.33
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
Can't claim USB device [1cb0:0038]: could not detach kernel driver from interface 0: Operation not permitted
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.43 (2.7.4-3009-gd831a9ae)
USB communication driver 0.33
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
Can't claim USB device [1cb0:0038]: could not detach kernel driver from interface 0: Operation not permitted
Driver failed to start (exit status=1)
pi@raspberrypi:~ $ usbreset
Usage:
  usbreset PPPP:VVVV - reset by product and vendor id
  usbreset BBB/DDD   - reset by bus and device number
  usbreset "Product" - reset by product name

Devices:
  Number 002/002  ID 090c:1000  Flash Drive FIT
  Number 001/003  ID 24ae:2013  Rapoo 2.4G Wireless Device
  Number 001/002  ID 2109:3431  USB2.0 Hub
  Number 001/004  ID 1cb0:0038  KEOR PDU 800
pi@raspberrypi:~ $  /usr/lib/nut/usbhid-ups -DDD -a ups
Network UPS Tools - Generic HID driver 0.43 (2.7.4-3009-gd831a9ae)
USB communication driver 0.33
   0.000000	Can't open /etc/nut/ups.conf: Can't open /etc/nut/ups.conf: Permission denied
pi@raspberrypi:~ $ sudo  /usr/lib/nut/usbhid-ups -DDD -a ups
Network UPS Tools - Generic HID driver 0.43 (2.7.4-3009-gd831a9ae)
USB communication driver 0.33
   0.000000	[D1] debug level is '3'
   0.002137	[D2] Initializing an USB-connected UPS with library (null) (NUT subdriver name='USB communication driver' ver='0.33')
   0.002229	[D1] upsdrv_initups (non-SHUT)...
   0.004143	[D3] usb_busses=0x1c434b8
   0.004246	[D2] Checking device (090C/1000) (002/002)
   0.004437	[D2] - VendorID: 090c
   0.004505	[D2] - ProductID: 1000
   0.004568	[D2] - Manufacturer: unknown
   0.004627	[D2] - Product: unknown
   0.004686	[D2] - Serial Number: unknown
   0.004747	[D2] - Bus: 002
   0.004805	[D2] - Device: 002
   0.004863	[D2] - Device release number: 1100
   0.004923	[D2] Trying to match device
   0.004983	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.005059	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.005122	[D2] Device does not match - skipping
   0.005211	[D2] Checking device (1D6B/0003) (002/001)
   0.005343	[D2] - VendorID: 1d6b
   0.005408	[D2] - ProductID: 0003
   0.005467	[D2] - Manufacturer: unknown
   0.005525	[D2] - Product: unknown
   0.005584	[D2] - Serial Number: unknown
   0.005645	[D2] - Bus: 002
   0.005703	[D2] - Device: 001
   0.005761	[D2] - Device release number: 0510
   0.005820	[D2] Trying to match device
   0.005879	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.005947	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.006007	[D2] Device does not match - skipping
   0.006094	[D2] Checking device (1CB0/0038) (001/004)
   0.006245	[D2] - VendorID: 1cb0
   0.006308	[D2] - ProductID: 0038
   0.006366	[D2] - Manufacturer: unknown
   0.006427	[D2] - Product: unknown
   0.006484	[D2] - Serial Number: unknown
   0.006543	[D2] - Bus: 001
   0.006602	[D2] - Device: 004
   0.006662	[D2] - Device release number: 0200
   0.006719	[D2] Trying to match device
   0.006780	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.006844	interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
   0.006907	[D3] match_function_regex: matching a device...
   0.007057	[D2] Device matches
   0.007131	[D2] failed to claim USB device: could not claim interface 0: Operation not permitted
   0.007202	[D2] failed to detach kernel driver from USB device: could not detach kernel driver from interface 0: Operation not permitted
   0.007270	[D2] failed to claim USB device: could not claim interface 0: Operation not permitted
   0.007333	[D2] failed to detach kernel driver from USB device: could not detach kernel driver from interface 0: Operation not permitted
   0.007395	[D2] failed to claim USB device: could not claim interface 0: Operation not permitted
   0.007456	[D2] failed to detach kernel driver from USB device: could not detach kernel driver from interface 0: Operation not permitted
   0.007523	[D2] failed to claim USB device: could not claim interface 0: Operation not permitted
   0.007588	[D2] failed to detach kernel driver from USB device: could not detach kernel driver from interface 0: Operation not permitted
   0.007652	Can't claim USB device [1cb0:0038]: could not detach kernel driver from interface 0: Operation not permitted

jimklimov, 2021-11-07:
Generally, the `Operation not permitted` probably mean that your NUT run-time user account may not write (or even read?) the device filesystem nodes for the USB device. It should be permitted to access (or even "own") the device for the UPS, and apply that across reboots and possibly device re-plugs, which may be a very different adventure depending on the OS used (e.g. USB ports may get re-enumerated based on their population with connected devices or hubs).

One alternative, not perfect but usable, would be to run the NUT driver as `root`.

UPDATE: Re-reading your log, I see you've also hit the issue with `ups.conf` not accessible to your test user so you re-ran the driver via `sudo`; in that case check if some OS/kernel subsystem (such as udev) may be grabbing and holding your UPS as a HID device. A build of NUT should generate config snippets for several device-management frameworks to tell them that a NUT driver is responsible for a matching device.

blecher-at, 2021-12-02:
@jimklimov can we merge this? or what's missing. It runs on my device for over a month now without issues.

jimklimov, 2021-12-03:
Thanks for confirmation, hope to take another look soon, then.

muetdhiver, 2021-12-13:
I got 2x of those Keor 800 and the patch is working well. Thank you for it.

Mr-sulu, 2021-12-16:
Hi, I've a KEOR 800 too attached to a synology 220+. Is not clear to me how to apply the patch to NUT insite DSM to have the legrand recognized. Someone can help me please?
Many thanks!

johndekroon, 2021-12-16:
> Hi, I've a KEOR 800 too attached to a synology 220+. Is not clear to me how to apply the patch to NUT insite DSM to have the legrand recognized. Someone can help me please?
> Many thanks!

Unless you have root access and able to compile it yourself, you have to wait until this is merged, they release a new version and Synology updates NUT.

jimklimov, 2022-02-10:
The merge to master was done last month; release of NUT and eventual updated packaging in different distros is still on the menu - got some loose ends to tie up first, and late-coming fixes.

Regarding this issue, so far not closing it as the early discussion mentioned some follow-up work, though it has its own tickets (e.g. #619)

jimklimov, 2022-04-24:
To review what's left to do - if there are any more issues to formulate - as part of 2.8.1

nseries100, 2022-06-21:
I'm trying @blecher-at setup:

- cloned https://github.com/blecher-at/nut.git  
- run ./autogen.sh
- run his ./configure command

but after make command I'm getting this error:
`make: *** No rule to make target '../common/libcommon.la', needed by 'al175'.  Stop.
`
Any hint?

jimklimov, 2022-06-21:
As the recent comment says, changes from @blecher-at were in PR #1075 and merged in the winter. So should be just part of current master and NUT 2.8.0 release.

In your sample, if that is verbatim what you did, you check out *some* default branch, probably master that is currently some 3k commits behind the "real" current NUT codebase. Note that source branch of the PR was deleted by author after the merge.

nseries100, 2022-06-22:
I cloned master branch and its working fine now with my KEOR 800. Thx

vyruz1986, 2022-06-26:
> I cloned master branch and its working fine now with my KEOR 800. Thx

Would you be so kind and share the output you're getting from `upsc` with us? I'm on the fence for a KEOR 800 PDU but would like to know whether it reports output current total and per output plug. Thanks in advance!

blecher-at, 2022-06-27:
It outputs the current of all devices, the PDU800 doesnt support metrics per output plug.

`#upsc lups
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.runtime: 2520
battery.voltage: 1.4
battery.voltage.nominal: 1.2
device.mfr: Legrand
device.model: KEOR PDU 800
device.serial:
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0038
driver.parameter.synchronous: no
driver.parameter.vendorid: 1cb0
driver.version: 2.7.4-4539-gefb9ee45
driver.version.data: Legrand HID 0.2
driver.version.internal: 0.45
driver.version.usb: libusb-1.0.24 (API: 0x1000108)
input.transfer.high: 270
input.transfer.low: 180
input.voltage: 236
input.voltage.nominal: 230
output.frequency: 50
output.voltage: 6
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 20
ups.mfr: Legrand
ups.model: KEOR PDU 800
ups.productid: 0038
ups.realpower.nominal: 480
ups.serial:
ups.status: OL CHRG
ups.vendorid: 1cb0
`

MonsieurPoutounours, 2022-07-19:
Hi,

Thanks to everyone implied in Keor PDU support. 

With nut version 2.8.0, compiled from source with root user, i am still facing issues.

Here is the output (loop) of /usr/local/ups/sbin/upsd -DDD

```
Network UPS Tools upsd 2.8.0
   0.000000     fopen /var/state/ups/upsd.pid: No such file or directory
   0.000012     Could not find PID file '/var/state/ups/upsd.pid' to see if previous upsd instance is already running!

   0.000197     [D3] listen_add: added 192.168.10.2:3493
   0.000264     [D1] debug level is '3'
   0.000274     [D3] setuptcp: try to bind to 192.168.10.2 port 3493
   0.000302     listening on 192.168.10.2 port 3493
   0.000477     /var/state/ups is world readable
   0.000570     Can't connect to UPS [keor] (usbhid-ups-keor): Connection refused
   0.000864     [D2] user_add_action: adding 'SET' for admin
   0.000875     [D2] user_add_instcmd: adding 'ALL' for admin
   0.000886     Running as foreground process, not saving a PID file
   0.000894     [D1] mainloop: UPS [keor] is not currently connected
   0.000905     [D1] mainloop: UPS [keor] is now connected as FD -1
   0.000909     [D2] mainloop: polling 1 filedescriptors
   2.002955     [D2] mainloop: no data available
   2.003004     [D1] mainloop: UPS [keor] is not currently connected
   2.003042     [D1] mainloop: UPS [keor] is now connected as FD -1
   2.003050     [D2] mainloop: polling 1 filedescriptors
   4.005320     [D2] mainloop: no data available
   4.005370     [D1] mainloop: UPS [keor] is not currently connected
   4.005410     [D1] mainloop: UPS [keor] is now connected as FD -1
   4.005416     [D2] mainloop: polling 1 filedescriptors
   6.007459     [D2] mainloop: no data available
   6.007508     [D1] mainloop: UPS [keor] is not currently connected
   6.007546     [D1] mainloop: UPS [keor] is now connected as FD -1
   6.007552     [D2] mainloop: polling 1 filedescriptors
```

jimklimov, 2022-07-20:
This is a log of the data server that says it failed to connect to a driver
socket. What does the driver log say? Did you start it? :)

MonsieurPoutounours, 2022-07-21:
You're right. Driver was not starting correctly and i was fooled by "UPS [keor] is now connected as FD -1".
It is now working on a Keor PDU 800 with 2.8.0 package from Debian experimental branch (has dependencies from the unstable branch too). 

Thanks @jimklimov

mrMiimo, 2022-09-20:
I think I have the same problem described in this thread...
My setup:
```
[UPS]
driver=usbhid-ups
port = auto
providerid = 1cb0
```

And I always have this error:

```
No matching HID UPS found
Controller failed to start (exit status = 1)
```

Does anyone know when this PR will be merged? Tnks

matushorvath, 2022-09-20:
@mrMiimo You could try this:
```
[UPS]
        driver = nutdrv_qx
        port = auto
```

It seems `1cb0` is mentioned in nutdrv_qx, so there is a good chance it will work. It worked for me with a different Legrand device (mine is provider `0665`).
https://github.com/networkupstools/nut/blob/master/drivers/nutdrv_qx.c#L1972

jimklimov, 2022-09-20:
Also, not provider(id) but vendor(id). Terms matter.

"This PR" as reworked by @blecher-at was merged almost a year ago. Are you running NUT v2.8.0 or newer (current master from github)?

mrMiimo, 2022-09-23:
> Also, not provider(id) but vendor(id). Terms matter.
> 
> "This PR" as reworked by @blecher-at was merged almost a year ago. Are you running NUT v2.8.0 or newer (current master from github)?

ok, i realized i am not running "nut 2.8" or new, but nut 2.7.4, this is the version i see when i run "`upsdrvctl  start`":
```
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic Q * USB / Serial driver 0.28 (2.7.4)
```

Sorry but I'm new to NUT...
I am trying to install 2.8 but it seems to me that there is no package for pi 🫤

jimklimov, 2022-09-23:
Packages are not for "Pi" but for PS distro, like Debian (here usually Raspbian, Armbian and other derivatives). Package update by Debian team maintainer was AFAIK proposed some time ago and stalled in "experimental" or "unstable" layer, I think - something to track in the distro. NUT as a project has little leverage in others' sandboxes.

mrMiimo, 2022-09-23:
> Packages are not for "Pi" but for PS distro, like Debian (here usually Raspbian, Armbian and other derivatives). Package update by Debian team maintainer was AFAIK proposed some time ago and stalled [...]

well, after a couple of initial problems .. I installed and compiled NUT 2.8 ...
but now I get this error:

```
Network UPS Tools - UPS driver controller 2.8.0-Windows-30-g692bad99b
Network UPS Tools - Generic HID driver 0.49 (2.8.0-Windows-30-g692bad99b)
USB communication driver (libusb 0.1) 0.43
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 65535 in ReportID: 0x0f
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x12
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x13
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 65535 in ReportID: 0x32
Using subdriver: Legrand HID 0.2
```

with this ups.conf

```
[ups]
        driver = "usbhid-ups"
        port = "auto"
	pollonly
        desc = "Legrand"
	vendorid = "1cb0"
	productid = "0038"
```

I think maybe I'll have to look for other ways 🫤

jimklimov, 2022-09-24:
These are not errors but info messages - many vendors do not really follow the protocol correctly so we tried to guess what they meant (since 2.8.0); maybe some readings will yield trash that we can't vouch for.

Did the driver start however? With a running `upsd` server, does it report info (e.g. to `upsc` CLI client)?

mrMiimo, 2022-09-24:
> These are not errors but info messages - many vendors do not really follow the protocol correctly so we tried to guess what they meant (since 2.8.0); maybe some readings will yield trash that we [...]

i don't think it started, i think i'm missing something ... when i run upsd i get this:

```
root@raspberrypi:/home/pi# /usr/local/ups/sbin/upsd -F ups@localhost
Network UPS Tools upsd 2.8.0-Windows-30-g692bad99b
fopen /var/state/ups/upsd.pid: No such file or directory
Could not find PID file '/var/state/ups/upsd.pid' to see if previous upsd instance is already running!

Network server for UPS data.
```
i am trying to understand, maybe i miss starting a service

jimklimov, 2022-09-24:
Can't hep but ask: did you read documentation? Command-line help? NUT design overview (the 3 layers of drivers-dataserver-clients)?

At best `upsd` would ignore the `ups@localhost` at least if it does not abort outright. (The message about PID file is just that - it can't determine if an older copy of `upsd` is running and so would conflict)

When you have the driver(s) and a data server running on same system, you can use clients like `upsc` to query for info from the same or different system, like `upsc upsname(@localhost)`.

mrMiimo, 2022-09-24:
> Can't hep but ask: did you read documentation? Command-line help? NUT design overview (the 3 layers of drivers-dataserver-clients)?

You are right to ask (I would have asked at the first attempt 😅) 
I have read everything (Im surprised by the complete documentation), some things I do not fully understand .. but rereading and practicing I understand them .. 

> At best `upsd` would ignore the `ups@localhost` at least if it does not abort outright. (The message about PID file is just that - it can't determine if an older copy of `upsd` is running and so [...]

oh... ok.
A little while ago I found a docker image of an old version of NUT I suppose:

```
/etc/nut # upsdrvctl -V
Network UPS Tools - UPS driver controller 20220715-1526-gded7a4cdc7
```
I just had to configure the container ports and other stuff, but after configuring ups.conf and nut.conf, everything worked on the first try. I get a reply from ups:

```
pi@raspberrypi:~ $ upsc ups@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.runtime: 1620
battery.voltage: 1.3
battery.voltage.nominal: 1.2
device.mfr: Legrand 
device.model: KEOR PDU 800
device.serial:                 
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/bus/usb/001/004
driver.parameter.synchronous: auto
driver.version: 20220715-1526-gded7a4cdc7
driver.version.data: Legrand HID 0.2
.................
```

now my goal is:
- configure upsmon and upssched
- make sure it turns off all devices on the network when it reaches a critical battery level
- and configure the sending of e-mails and SMS at each event

mrMiimo, 2022-09-24:
> now my goal is:
> 
> * configure upsmon and upssched
> * make sure it turns off all devices (switch, nvr, udm etc) on the network when it reaches a critical battery level
> * and configure the sending of e-mails and SMS at each event

done everything, including sms alerts! Thanks jimklimov!!!

[image]

mmhorda, 2022-11-25:
battery.voltage: 1.3
battery.voltage.nominal: 1.2

is this a bug? I thoght that volatege should be a bit higher.

VozDeOuro, 2022-12-10:
hello i saw that NUT support some SMS Legrand devices 
i want to know if you has plans to implement others SMS Legrand drives, like SMS Power View modules.

if you want i can send to you the program that they made available for us

jimklimov, 2023-01-06:
> battery.voltage: 1.3 battery.voltage.nominal: 1.2
> 
> is this a bug? I thoght that volatege should be a bit higher.

If that is what they report... an UPS running on AA batteries? :)
However, a typical six-pack of cells in one accumulator battery might mean `1.2V * 6 = 7.2V ` which rings a bell, for example...

There were some changes about detection/guesswork of battery cell count and so overall voltage. proposed for a `nutdrv_qx` driver type in #1279 / #1652 which might get merged soon; but they won't directly help for `usbhid-ups` as a separate driver.

jimklimov, 2023-01-06:
> hello i saw that NUT support some SMS Legrand devices i want to know if you has plans to implement others SMS Legrand drives, like SMS Power View modules.
> 
> if you want i can send to you the program that they made available for us

Is that a binary program, or sources (licensed so) that someone might read and port that to NUT?

Generally, this would need a developer (random but enthusiastic volunteer from the community, ideally with access to HW and specs) to implement or sniff and reverse-engineer the protocol needed. I do hope and encourage that you find and entice someone to do this. Other than that, no "plans" exist, unfortunately.

MatteoScg, 2023-01-13:
> @greyslater
hi nseries100 - i came upon all this because i connected the Keor 800 to a QNAP NAS and it's not recognized
found out that QTS (QNAP OS) uses NUT under the hood

Hi @greyslater,
have you found the way to let your QNAP NAS recognize the Legrand UPS?
I am encountering the same issue by connecting my Legrand Keor SP 1000 to my QNAP TS-251A.

Thank you

gyulaasztalos, 2023-07-18:
> If that is what they report... an UPS running on AA batteries? :) However, a typical six-pack of cells in one accumulator battery might mean `1.2V * 6 = 7.2V ` which rings a bell, for example... [...]

Hi @jimklimov 

I have the Legrand KEOR PDU 800 (1cb0:0038) and I also got the incorrect voltage values.
I could fix it with this:
```
diff --git a/drivers/hidparser.c b/drivers/hidparser.c
index 0d7b53f80..6dd448ed9 100644
--- a/drivers/hidparser.c
+++ b/drivers/hidparser.c
@@ -332,7 +332,7 @@ static int HIDParse(HIDParser_t *pParser, HIDData_t *pData)
                                        pParser->Data.LogMax,
                                        pParser->Value,
                                        pParser->Data.ReportID);
-                               pParser->Data.LogMax = (long) pParser->Value;
+                               pParser->Data.LogMax = 65535;
                        }
                        break;
 
diff --git a/drivers/legrand-hid.c b/drivers/legrand-hid.c
index 371555328..5d7dc31cd 100644
--- a/drivers/legrand-hid.c
+++ b/drivers/legrand-hid.c
@@ -134,9 +134,9 @@ static hid_info_t legrand_hid2nut[] = {
        { "input.voltage.nominal", 0, 0, "UPS.Flow.ConfigVoltage", NULL, "%.0f", HU_FLAG_STATIC, NULL },
 
        /* Battery Data */
-       { "battery.voltage.nominal", 0, 0, "UPS.PowerSummary.ConfigVoltage", NULL, "%.0f", HU_FLAG_STATIC, divide_by_10_conversion },
+       { "battery.voltage.nominal", 0, 0, "UPS.PowerSummary.ConfigVoltage", NULL, "%.0f", HU_FLAG_STATIC, NULL },
        { "battery.voltage.nominal", 0, 0, "UPS.BatterySystem.Battery.ConfigVoltage", NULL, "%.0f", HU_FLAG_STATIC, NULL },
-       { "battery.voltage", 0, 0, "UPS.PowerSummary.Voltage", NULL, "%.0f", 0, divide_by_10_conversion },
+       { "battery.voltage", 0, 0, "UPS.PowerSummary.Voltage", NULL, "%.0f", 0, NULL },

@@ -144,8 +144,8 @@ static hid_info_t legrand_hid2nut[] = {
        { "battery.charge.low", 0, 0, "UPS.PowerSummary.RemainingCapacityLimit", NULL, "%.0f", HU_FLAG_STATIC, NULL },
 
        /* Output Data */
-       { "output.voltage", 0, 0, "UPS.Output.Voltage", NULL, "%.0f", 0, NULL },
-       { "output.voltage", 0, 0, "UPS.PowerConverter.Output.Voltage", NULL, "%.0f", 0, legrand_times10M_info },
+       { "output.voltage", 0, 0, "UPS.Output.Voltage", NULL, "%03.1f", 0, NULL },
+       { "output.voltage", 0, 0, "UPS.PowerConverter.Output.Voltage", NULL, "%.0f", 0, NULL },
```
And the result:
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.runtime: 1620
battery.voltage: 14
battery.voltage.nominal: 12
device.mfr: Legrand 
device.model: KEOR PDU 800
device.serial:                 
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.productid: 0038
driver.parameter.synchronous: auto
driver.parameter.vendorid: 1cb0
driver.state: quiet
driver.version: 2.8.0-Windows-726-g73ee107fa
driver.version.data: Legrand HID 0.2
driver.version.internal: 0.49
driver.version.usb: libusb-0.1 (or compat)
input.transfer.high: 270
input.transfer.low: 180
input.voltage: 236
input.voltage.nominal: 230
output.frequency: 50
output.voltage: 235.6
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 32
ups.mfr: Legrand 
ups.model: KEOR PDU 800
ups.productid: 0038
ups.realpower.nominal: 480
ups.serial:                 
ups.status: OL CHRG
ups.vendorid: 1cb0
```
I’m not a programmer, I don’t know how such fix should be done properly or how should I report this. I just wanted to fix my PDU800 and I'm trying to help, but I'm afraid this could break voltage display on other Legrand devices or might cause other problems.

jimklimov, 2023-07-18:
Thanks, so it seems some devices have it fixed and others not... Maybe the conversion function should stay, and detect somehow (device id? value range? config option?) whether it should act or keep the value as is.

pczolee, 2024-01-09:
Hello.

I have an Legrand Keor SP UPS. I built the latest nut, so the UPS is recognized and it look like working, but there is a little problem, the reported values (input/output voltage) are bad (but sometimes are good). For example there is two output:
# upsc lups
battery.charge: 80
battery.charge.warning: 30
battery.runtime: 2811
battery.voltage: 13.4
battery.voltage.nominal: 12
device.mfr: Legrand  
device.model: Legrand UPS  
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0032
driver.parameter.synchronous: auto
driver.parameter.vendorid: 1cb0
driver.state: quiet
driver.version: 2.8.1.1
driver.version.data: Legrand HID 0.2
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.21 (API: 0x1000105)
input.transfer.high: 240.0
input.transfer.low: 170
input.voltage: 25.0
input.voltage.nominal: 230
output.voltage: 840.0
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: Legrand  
ups.model: Legrand UPS  
ups.productid: 0032
ups.realpower.nominal: 800
ups.status: OL CHRG
ups.vendorid: 1cb0

And after a few minutes:
# upsc lups
battery.charge: 100
battery.charge.warning: 30
battery.runtime: 2811
battery.voltage: 13.4
battery.voltage.nominal: 12
device.mfr: Legrand  
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0032
driver.parameter.synchronous: auto
driver.parameter.vendorid: 1cb0
driver.state: quiet
driver.version: 2.8.1.1
driver.version.data: Legrand HID 0.2
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.21 (API: 0x1000105)
input.transfer.high: 240.0
input.transfer.low: 170
input.voltage: 217.0
input.voltage.nominal: 230
output.voltage: 10.0
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: Legrand  
ups.productid: 0032
ups.realpower.nominal: 800
ups.status: OL CHRG
ups.vendorid: 1cb0

How can I help to debug and solve this issue?

Thanks.

Ariessanea1, 2024-01-09:
It fix

pczolee, 2024-01-09:
> It fix
> […](#)

Can you explain what do you mean? Thanks

muetdhiver, 2024-01-10:
Hello, 
I think my Grafana's monitoring illustrates what [pczolee](https://github.com/pczolee) is pointing. (Yes I have 2 Keor on 2 different monophased incoming power  lines)
my VOut is not reporting correct values.
[image]

Regards

pczolee, 2024-01-12:
The situation is much worse for me (it should be around 230V)! :(

Vin:
[image]

Vout:
[image]

pczolee, 2024-01-30:
No idea? :(

jimklimov, 2024-01-30:
@pczolee : sorry, travelling these weeks. OTOH, you might want to bump debug verbosity, and since you are using now a custom build of NUT - tinker with code (add more debugs where you think visibility is missing) to see where and based on what inputs the value jumps from normal range (Vin seems to be there sometimes) to obscenely wrong.

Some further ideas:
* NUT v2.8.0 introduced a Log Min/Max fixup for bad USB report descriptors (see #1024 and issue trail from there). 2.8.1 added a run-time config toggle to disable the fixup, mostly to check if we misunderstood something about USB and the fix adds more problems than it solves on certain devices. I don't remember the name while commuting, but give it a shot.
* Check USB cabling and its surroundings! Maybe some EMI noise (motors, fridges, neon lights...) just cause random bits to be seen? Try a cable with grounding mesh in the skin, etc.