networkupstools/nut #617: Powercom IMP-3000AP UPS

Limych, 2018-09-11:
Please, add support for Powercom IMP-3000AP UPS

```
ugen5.2: <POWERCOM Co.,LTD HID UPS Battery> at usbus5, cfg=0 md=HOST spd=LOW (1.5Mbps) pwr=ON (100mA)

  bLength = 0x0012
  bDescriptorType = 0x0001
  bcdUSB = 0x0110
  bDeviceClass = 0x0000  <Probed by interface class>
  bDeviceSubClass = 0x0000
  bDeviceProtocol = 0x0000
  bMaxPacketSize0 = 0x0008
  idVendor = 0x0d9f
  idProduct = 0x0004
  bcdDevice = 0x0002
  iManufacturer = 0x0003  <retrieving string failed>
  iProduct = 0x0001  <retrieving string failed>
  iSerialNumber = 0x0002  <retrieving string failed>
  bNumConfigurations = 0x0001
```

Limych, 2018-09-11:
```
root@NAS:~ # cat /etc/local/nut/ups.conf
[ups]
    driver = usbhid-ups
    port = /dev/ugen5.2
    desc = "Powercom IMP-3000AP"
vendorid=0d9f
productid=0004

```

kytster, 2018-11-25:
Ubuntu server !8.04 and Powercom Imperial IMP-825AP

```
   0.032164	Checking device (0D9F/0004) (001/005)
   0.033710	- VendorID: 0d9f
   0.033739	- ProductID: 0004
   0.033748	- Manufacturer: unknown
   0.033756	- Product: unknown
   0.033765	- Serial Number: unknown
   0.033777	- Bus: 001
   0.033787	- Device release number: 0002
   0.033798	Trying to match device
   0.033865	Device matches
   0.033948	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.034416	Unable to get HID descriptor (error sending control message: Protocol error)
   0.034446	HID descriptor length (method 1) -1
   0.034462	HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 e4 03
   0.034471	HID descriptor length (method 2) 996
   0.034480	HID descriptor length 996
   0.034941	Unable to get Report descriptor: Protocol error
```
```
[powercom]
driver=usbhid-ups
port=auto
desc="Powercom Imperial"

```

If it makes sense:
 Once, I had things worked after I plugged UPS to USB2 port instead of USB 3. But then accidentally I switched off server without normal shutdown, and no luck since then

Yellow-Horror, 2018-12-22:
Similar trouble with Powercom WAR-1000AP:
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.003678     upsdrv_initups...
   0.005565     Checking device (0D9F/0004) (/dev/usb//dev/ugen0.2)
   0.212468     - VendorID: 0d9f
   0.212509     - ProductID: 0004
   0.212529     - Manufacturer: unknown
   0.212545     - Product: unknown
   0.212560     - Serial Number: unknown
   0.212576     - Bus: /dev/usb
   0.212591     - Device release number: 0001
   0.212606     Trying to match device
   0.212634     Device matches
   0.212663     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.282189     Unable to get HID descriptor (Unknown error)
   0.282240     HID descriptor length (method 1) -1
   0.282269     HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 c6 03
   0.282286     HID descriptor length (method 2) 966
   0.282302     HID descriptor length 966
   0.352360     Unable to get Report descriptor: Input/output error

Yellow-Horror, 2018-12-23:
The my issue above persists on XigmaNAS system only. When i try the Linux Mint on the same hardware, the NUT works like a charm. It shows Generic HID driver 0.38 (2.7.2) and USB communication driver 0.32. May be this matters.

Yellow-Horror, 2018-12-23:
Seems that Limych and reida above also have FreeBSD-based systems.

einhander, 2019-11-09:
Hi, I have same Issue with Powercom IMP 525AP on Debian buster.
Here is a relevant part of /lib/nut/usbhid-ups -DDDD -a ups:
````
0.113873	Checking device (0D9F/0004) (005/003)
   0.125578	- VendorID: 0d9f
   0.125590	- ProductID: 0004
   0.125597	- Manufacturer: unknown
   0.125621	- Product: unknown
   0.125638	- Serial Number: unknown
   0.125646	- Bus: 005
   0.125653	- Device release number: 0002
   0.125660	Trying to match device
   0.125728	Device matches
   0.125786	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.129575	Unable to get HID descriptor (error sending control message: Protocol error)
   0.129586	HID descriptor length (method 1) -1
   0.129597	i=0, extra[i]=09, extra[i+1]=21
   0.129622	HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 e4 03
   0.129633	HID descriptor length (method 2) 996
   0.129643	HID descriptor length 996
   0.133587	Unable to get Report descriptor: Protocol error
````

lsusb -d 0d9f: -vvvv

````
Bus 005 Device 003: ID 0d9f:0004 Powercom Co., Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0d9f Powercom Co., Ltd
  idProduct          0x0004 
  bcdDevice            0.02
  iManufacturer           3 (error)
  iProduct                1 (error)
  iSerial                 2 (error)
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     996
          Report Descriptor: (length is -1)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             100
can't get debug descriptor: Resource temporarily unavailable
cannot read device status, Resource temporarily unavailable (11)
````

And it's occures after shutdown and cleaning the PC.
I tried Windows tool to check out UPS, and it's working. After disconection from windows (I used VM, so phisucal hardware is the same)  iManufacturer  field rerorted correctly, but   usbhid-ups reports the same error.

maurizio-emmex, 2023-10-20:
The same issue with Powercom BNT 1000 AP in FreeBSD.
```
uname -a
FreeBSD clover-nas4.local 13.2-RELEASE-p4 FreeBSD 13.2-RELEASE-p4 GENERIC amd64

grep -A 3 powercom /usr/local/etc/nut/ups.conf
[powercom]
        driver = usbhid-ups
        port = auto
        vendorid = 0d9f

/usr/local/libexec/nut/usbhid-ups -a powercom -DDDD -u root
                [D2] Checking device 10 of 10 (0D9F/0004)
   0.034472     [D1] nut_libusb_open: invalid libusb bus number 0
   0.171216     [D1] nut_libusb_open get iManufacturer failed, retrying...
   0.310911     [D1] nut_libusb_open get iManufacturer failed, retrying...
   0.446974     [D1] nut_libusb_open get iManufacturer failed, retrying...
   0.579042     [D1] nut_libusb_open get iProduct failed, retrying...
   0.714915     [D1] nut_libusb_open get iProduct failed, retrying...
   0.854908     [D1] nut_libusb_open get iProduct failed, retrying...
   0.994912     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   1.130976     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   1.266060     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   1.266087     [D2] - VendorID: 0d9f
   1.266092     [D2] - ProductID: 0004
   1.266096     [D2] - Manufacturer: unknown
   1.266100     [D2] - Product: unknown
   1.266103     [D2] - Serial Number: unknown
   1.266107     [D2] - Bus: 002
   1.266110     [D2] - Bus Port: 001
   1.266114     [D2] - Device: 002
   1.266121     [D2] - Device release number: 0002
   1.266124     [D2] Trying to match device
   1.266130     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   1.266138     [D3] match_function_regex: matching a device...
   1.266159     [D2] Device matches
   1.266162     [D2] Reading first configuration descriptor
   1.266214     [D3] libusb_kernel_driver_active() returned 0: Success
   1.266220     [D2] Claimed interface 0 successfully
   1.266224     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   1.335906     [D2] Unable to get HID descriptor (No device)
   1.335922     [D3] HID descriptor length (method 1) -1
   1.335928     [D4] i=0, extra[i]=09, extra[i+1]=21
   1.335938     [D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 e4 03
   1.335948     [D3] HID descriptor length (method 2) 996
   1.335952     [D2] HID descriptor length 996
   1.402087     [D2] Unable to get Report descriptor: Input/output error
   1.402136     [D2] libusb1: No appropriate HID device found
   1.402142     libusb1: Could not open any HID devices: insufficient permissions on everything
   1.402146     No matching HID UPS found
   1.402162     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```
   It gives me an error:
   Unable to get Report descriptor: Input/output error

```
lsusb -d 0d9f: -vvvv
   Bus /dev/usb Device /dev/ugen0.2: ID 0d9f:0004 Powercom Co., Ltd 
Device Descriptor:                 
  bLength                18        
  bDescriptorType         1                                                    
  bcdUSB               1.10        
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0        
  bDeviceProtocol         0                                                    
  bMaxPacketSize0         8
  idVendor           0x0d9f Powercom Co., Ltd
  idProduct          0x0004      
  bcdDevice            0.02                                                    
  iManufacturer           3      
  iProduct                1                                                    
  iSerial                 2                                                    
  bNumConfigurations      1                                                    
  Configuration Descriptor:                                                    
    bLength                 9    
    bDescriptorType         2                                                  
    wTotalLength           34
    wTotalLength           34                                                                                                                                  
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     996
          Report Descriptor: (length is -1)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None 
          Usage Type               Data 
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             100
cannot read device status, Input/output error (5)
```
Regards,
Maurizio