networkupstools/nut #618: Sometimes Bcmxcp_usb (Eaton) driver gone

JulyIGHOR, 2018-09-13:
I'm using NUT with some Eaton backup power device.
Using driver Bcmxcp_usb it works fine, and I can see status "ALARM Online"
It works just fine and if power offline, I got notifications.
But few days later, 5-10 it just stops working, and all info not visible just looks like USB cable unplugged.
Status line where was "ALARM Online" become empty.
When I detect it I go to NUT settings and press "Reload" button.
Each time it helps and status starts to work again to "ALARM Online".

I hope to get this bug fixed, or get help with any advice how to make it reloaded each time it gone automatically.

zykh, 2018-09-13:
hmm... without seeing the logs I can't say for sure... and your description seems from something other than straight NUT (e.g. the 'reload' button), so maybe there's something else (out of our control) going on...

The first thing that comes to my mind as a possible interpretation (and that could easily be fixed) is that, maybe, the device briefly disconnects or becomes 'unpollable' for a short amount of time and, if the `bcmxcp_usb` driver, once declared the device 'disconnected', can't reestablish the communication in the given (5) number of tries (as opposed to other drivers, which keep trying forever), it exit()s.
But this is just something in the air...

So:
- any chance to get some logs?
- if I throw some code at you, will you be able/have enough time to rebuild the driver and test it?

JulyIGHOR, 2018-09-13:
Sure I'll test any updates you give.
Here is log screenshot:
[image]

zykh, 2018-09-16:
It seems to be consistent with my previous interpretation...
Try `bcmxcp_usb` compiled from this branch: https://github.com/zykh/nut/tree/issue-618

JulyIGHOR, 2018-09-17:
Can you please send bin file to replace? I don't have toolchain to compile it.

zykh, 2018-09-17:
What system is that?

zykh, 2018-09-30:
@JulyIGHOR: I looked at that very quickly... since it mentions Slackware, if the executables are built using [plain vanilla Slackware configuration from SlackBuilds](https://slackbuilds.org/repository/14.2/system/nut/), it's worth trying the `bcmxcp_usb` driver as built on Slackware (14.2 x64).
So, here it is, in two flavours:
1. [using libusb-0.1](https://github.com/networkupstools/nut/files/2431104/bcmxcp_usb-libusb-0.1.tar.gz),
2. [using libusb-1.0](https://github.com/networkupstools/nut/files/2431105/bcmxcp_usb-libusb-1.0.tar.gz).

Let me know if at least one of them works...

JulyIGHOR, 2018-09-30:
Thanks.
In some reason it many days already not breaks.
I think it only happens when we have lots of electricity jumps during short period.
So I'll try to disconnect/connect USB than, maybe it will make this issue happens again.
I'll answer here when tested.

JulyIGHOR, 2018-11-03:
I have found out reason of problem. My Eaton has hardware issue, it just freezing sometimes.

closed 2018-11-03 by JulyIGHOR

zykh, 2018-11-05:
@JulyIGHOR, just one question: have you had a chance to test `bcmxcp_usb` compiled starting from [the 'issue-618' branch](https://github.com/zykh/nut/tree/issue-618) or one of the drivers from [my last post](https://github.com/networkupstools/nut/issues/618#issuecomment-425683514)?
I ask because those changes add a good amount of simplification to the `bcmxcp_usb` driver, also easing the maintenance of it, and so I'd like to push them to our libusb branch... and I would do that with a lighter heart, if you didn't experience any regression with (at least one of) them.