networkupstools/nut #622: CyberPower Systems EC850LCD supported by usbhid-ups

amanjpro, 2018-09-26:
CyberPower Systems EC850LCD:

```
$ upsc ups@localhost
battery.charge: 89
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4316
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 14.1
battery.voltage.nominal: 12
device.mfr: CPS
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ugen0.3
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 140
input.transfer.low: 96
input.voltage: 120.0
input.voltage.nominal: 120
output.voltage: 120.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 8
ups.mfr: CPS
ups.productid: 0501
ups.realpower.nominal: 450
ups.status: OL CHRG
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: 0
ups.vendorid: 0764
```

```
$ upsrw ups@localhost
[battery.charge.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 300

[input.transfer.high]
Description unavailable
Type: STRING
Maximum length: 10
Value: 140

[input.transfer.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 96

[ups.delay.shutdown]
Description unavailable
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Description unavailable
Type: STRING
Maximum length: 10
Value: 30
```
I have already run the `shutdown sequence` and everything is passing.

The link to the manufacturer’s online description of the device:
https://www.cyberpowersystems.com/product/ups/ec850lcd/

ioogithub, 2022-04-23:
Hi @amanjpro I purchased an EC850LCD based on seeing this issue with the intention of using it exclusively with NUT.

When I connect the UPS to my computer with the USB cable I get the following message every 4 seconds flooding journalctl:

```
Apr 22 22:30:04 ubuntu kernel: usb 1-1.3: USB disconnect, device number 9
Apr 22 22:30:10 ubuntu kernel: usb 1-1.3: new low-speed USB device number 10 using xhci_hcd
Apr 22 22:30:10 ubuntu kernel: usb 1-1.3: New USB device found, idVendor=0764, idProduct=0501, bcdDevice= 0.01
Apr 22 22:30:10 ubuntu kernel: usb 1-1.3: New USB device strings: Mfr=3, Product=1, SerialNumber=0
Apr 22 22:30:10 ubuntu kernel: usb 1-1.3: Product: EC850LCD
Apr 22 22:30:10 ubuntu kernel: usb 1-1.3: Manufacturer: CPS
Apr 22 22:30:10 ubuntu kernel: hid-generic 0001:15342:1050.000B: hiddev1,hidraw2: USB HID v1.10 Device [CPS EC850LCD] on usb-0000:01:00.0-1.3/input0
```
The lines repeat every 4 seconds and the first line 'device number' increments every time.

After I install and configure NUT, the message is replaced with:

```
ubuntu upsmon[997]: Poll UPS [<system>] failed - [<system>] does not exist on server localhost
ubuntu upsmon[997]: Poll UPS [<system>] failed - [<system>] does not exist on server localhost
ubuntu upsmon[997]: Poll UPS [<system>] failed - [<system>] does not exist on server localhost
```
...every 4 seconds as long as the UPS is connected to the computer.

Do you know if this is normal behavior, a configuration issue or perhaps a hardware error?

I have never seen a device continuously connect and disconnect from a USB port every few seconds but I am also new to UPS devices. I see some users are reporting a stale data issue but I can't determine if this is what I am seeing or not.

I apologize if this is not the correct forum to ask this question. I could not really find any info when searching to find out what I am dealing with here. I thought you might have some experience using NUT with this device.

jimklimov, 2022-04-24:
Sounds like either you have several frameworks like udev/upower/... trying
to take over the usb device (but then maybe you'd see same behavior with
other usb things), or some flaky cable, connector, port, maybe port's power
management or usb hub on the path.

OTOH, try plugging into a different physical port of the comouter and/or
try another cable?

ioogithub, 2022-04-24:
Thanks for the suggestions [@jimklimov](https://github.com/jimklimov), after a few hours of researching I found the solution in the always excellent archlinux wiki: https://wiki.archlinux.org/title/Network_UPS_Tools#CyberPower_UPS_keeps_disconnecting/reconnecting

Apparently this behavior is common with cyberpower devices. The solution in the above link fixed the problem.

commit 0ea1f1e3 (2022-04-25): driver.list.in: Add HCL info for CyberPower Systems EC850LCD (USB)

Closes: #622
closed 2022-04-25 by jimklimov