networkupstools/nut #625: CPS: for certain models, "ondelay" seems to trigger a reboot regardless of power state

clepple, 2018-10-04:
Partial list of affected models:
* OR500LCDRM1U
* Value600ELCD

Reference: 
* https://github.com/networkupstools/nut/issues/432#issuecomment-405371395
* https://alioth-lists.debian.net/pipermail/nut-upsuser/2018-October/011249.html

oviano, 2023-10-22:
I have just discovered this issue with a new CyberPower CP1500EPFLCD UPS.

I had followed the various tutorials and was testing it all out. I had an offdelay of 60s and an ondelay of 120s.

Sure enough, after 120s the UPS returned back on even though there was no mains power! I understand my only workaround is to set the ondelay to 0s, and hope that when mains power returns it returns long enough to charge the battery.

Is this something that has to be fixed by CyberPower or by NUT?

jimklimov, 2023-10-22:
I believe, by CyberPower.

One caveat might be, generally, that different UPS devices implement various commands regarding shutdown and return-or-not under various external conditions. It may be possible that different models (or firmware revisions) of the same vendor implement stuff inconsistently (not always a bad thing, if they converge from some ad-hoc solution to some de-facto standard and sane approach, for example).

So the caveat for me might be if (some) CPS models do implement the command we want, but the NUT drivers call some other command as the default shutdown handler.

For `usbhid-ups` CPS subdriver we map the following options at the moment: https://github.com/networkupstools/nut/blob/7ac1da177245b836c21ff8422dc4a58df65fda79/drivers/cps-hid.c#L225-L228

And the variant implementations of power commands for the driver are tried at https://github.com/networkupstools/nut/blob/7ac1da177245b836c21ff8422dc4a58df65fda79/drivers/usbhid-ups.c#L733-L782

oviano, 2023-10-23:
I'll investigate upscmd, I haven't looked at that.

In the meantime I have logged a support ticket with CyberPower and will post back here with any information they provide.

oviano, 2023-11-02:
So I went to the trouble of raising a ticket with CyberPower, and also sent them my NUT config files and description of the issue, including a link to this issue.

This was there response:

_I would like to inform you that our UPS did not test using any applications or resources from other parties. Furthermore, our UPS systems are unable to set up power on delay._

So, not interested in anything other than their official software basically.

jimklimov, 2023-11-02:
Thank you for that! :)

And to think, how other vendors go to the trouble of providing NUT drivers or even shipping NUT builds as their companion software!

oviano, 2023-11-02:
Yep, wish I'd researched it a bit better tbh and found one who better support NUT.

Shame, as I like the hardware.

PR #2722 (2024-12-18): Fix default CPS HID timings
CyberPower Systems (CPS) devices are often mentioned in the contexts of connection stability issues, and of mis-behaving with shutdown delays (e.g. immediately turning off where the user expected a 10 or 30 sec delay as configured).

Many of those cases fit known solutions, encoded here:
* more intensive `pollinterval` and `pollfreq` may help the UPS controller not go to sleep; default values for the latter on CPS HID is bumped here. In practice `pollonly` may play a role, but since it is a flag - we "only" loudly suggest it here for CPS devices.
* on/off delays ten [... cut]
closed 2024-12-18 by jimklimov