networkupstools/nut #626: usbhid-ups signal 15 exiting on startup

ragd011, 2018-10-04:
I'm not sure if this is the right place to report this issue so please kindly redirect me if necessary.  

I've been using nut for a few years on Ubuntu 14 and 16.  I recently did a fresh install of 18.04.  I installed and configured nut and noticed after reboot it was not working.

Syslog shows the driver loading and everything looks good but before upsmon can start, usbhid-ups exits with a signal 15 and communication fails.

To get it working after a reboot, you have to manually call upsdrvctl and the upsd.  Once that's done all works fine.  I couldn't figure out what was causing the issue so I tried updating the drivers using the wiki but that didn't help.

Finally I decided to move to using a newly built raspberry Pi running raspbian stretch.  once I got it all setup and rebooted, it did the exact same thing.

let me know if there is any pertinent info that needs to be provided.  Thanks, Rob.

val-kulkov, 2018-11-08:
It looks like I have the same problem. Ubuntu 18.04.1 LTS. The driver is usbhid-ups.

ragd011, 2018-11-09:
> I haven't had a chance to upgrade to 18.04 - how does it interact with the
> init system? Is there a native systemd unit, or is it still using a SysV
> init script?
>
> I think it is worth filing the bug on Ubuntu's bug tracker, since the
> signal 15 is coming from outside the NUT driver, and that is likely from
> the init system. https://bugs.launchpad.net/ubuntu/+source/nut
>
> One more thing to look for: are there duplicate entries for the usbhid-ups
> driver in syslog, with two different process IDs?
>
> Oct 25 22:50:05 hippo usbhid-ups[1854]: Startup successful
> Oct 29 08:47:59 hippo usbhid-ups[22819]: Startup successful
> Oct 29 10:24:07 hippo usbhid-ups[22819]: Signal 15: exiting
>
> In this case, PID 1854 started when the computer booted, and 22819 was
> started and manually stopped later (testing a second UPS after replacing
> its battery). If you have more than one driver running, with only one UPS
> attached and configured in ups.conf, then the second driver will try to
> send a signal 15 to the other driver to avoid contention. If this is
> happening on its own, the init system is probably to blame.
>
> (@ragd011 <https://github.com/ragd011> sorry I missed this when you first
> posted - still unpacking from moving)

No prob. Understand. I'll take a look and try to get some more info. Also
I'll seeing maybe 18.04 had some built-in ups monitoring that's calling the
hangup. Thanks.

ragd011, 2018-11-10:
OK I ran a pstree and systemd is the no 1 process.  Here's the output of my syslog right after a restart.  let me know what pertinent info to put in the ubuntu bug and I'll go ahead and fill it out.  Thanks.
BTW any chance it has something to do with the ipv6 stuff happening in between?

Nov 10 11:31:11 vidcap-1804 upsdrvctl[868]: Using subdriver: CyberPower HID 0.5
Nov 10 11:31:11 vidcap-1804 upsdrvctl[868]: Network UPS Tools - Generic HID driver 0.42 (2.7.4-505-g804e4e45)
Nov 10 11:31:11 vidcap-1804 upsdrvctl[868]: USB communication driver 0.33
Nov 10 11:31:11 vidcap-1804 upsdrvctl[868]: cps_adjust_battery_scale: battery readings will be scaled by 2/3
Nov 10 11:31:11 vidcap-1804 upsdrvctl[868]: Network UPS Tools - UPS driver controller 2.7.4
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.5653] Loaded device plugin: NMTeamFactory (/usr/lib/x86_64-linux-gnu/NetworkManager/libnm-device-plugin-team.so)
Nov 10 11:31:11 vidcap-1804 usbhid-ups[934]: Startup successful
Nov 10 11:31:11 vidcap-1804 systemd[1]: Started Network UPS Tools - power device driver controller.
Nov 10 11:31:11 vidcap-1804 systemd[1]: Starting Network UPS Tools - power devices information server...
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.5878] Loaded device plugin: NMAtmManager (/usr/lib/x86_64-linux-gnu/NetworkManager/libnm-device-plugin-adsl.so)
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.5889] Loaded device plugin: NMWwanFactory (/usr/lib/x86_64-linux-gnu/NetworkManager/libnm-device-plugin-wwan.so)
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.6142] device (lo): carrier: link connected
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.6148] manager: (lo): new Generic device (/org/freedesktop/NetworkManager/Devices/1)
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.6158] manager: (enp3s0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.6167] device (enp3s0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Nov 10 11:31:11 vidcap-1804 kernel: [   32.980879] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
Nov 10 11:31:11 vidcap-1804 kernel: [   33.194138] r8169 0000:03:00.0 enp3s0: link down
Nov 10 11:31:11 vidcap-1804 kernel: [   33.194239] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
Nov 10 11:31:11 vidcap-1804 NetworkManager[793]: <info>  [1541867471.9476] modem-manager: ModemManager available
Nov 10 11:31:12 vidcap-1804 upsd[935]: fopen /var/run/nut/upsd.pid: No such file or directory
Nov 10 11:31:12 vidcap-1804 upsd[935]: not listening on 192.168.1.90 port 3493
Nov 10 11:31:12 vidcap-1804 upsd[935]: listening on 127.0.0.1 port 3493
Nov 10 11:31:12 vidcap-1804 upsd[935]: no listening interface available
Nov 10 11:31:12 vidcap-1804 upsd[935]: not listening on 192.168.1.90 port 3493
Nov 10 11:31:12 vidcap-1804 upsd[935]: listening on 127.0.0.1 port 3493
Nov 10 11:31:12 vidcap-1804 upsd[935]: no listening interface available
Nov 10 11:31:12 vidcap-1804 upsd[935]: Network UPS Tools upsd 2.7.4
Nov 10 11:31:12 vidcap-1804 systemd[1]: nut-server.service: Control process exited, code=exited status=1
Nov 10 11:31:12 vidcap-1804 systemd[1]: nut-server.service: Failed with result 'exit-code'.
Nov 10 11:31:12 vidcap-1804 systemd[1]: Failed to start Network UPS Tools - power devices information server.
Nov 10 11:31:12 vidcap-1804 systemd[1]: nut-driver.service: Unit not needed anymore. Stopping.
Nov 10 11:31:12 vidcap-1804 systemd[1]: Stopping Network UPS Tools - power device driver controller...
Nov 10 11:31:12 vidcap-1804 systemd[1]: Starting Network UPS Tools - power device monitor and shutdown controller...
Nov 10 11:31:12 vidcap-1804 upsdrvctl[944]: Network UPS Tools - UPS driver controller 2.7.4
Nov 10 11:31:12 vidcap-1804 usbhid-ups[934]: Signal 15: exiting
Nov 10 11:31:12 vidcap-1804 systemd[1]: Stopped Network UPS Tools - power device driver controller.

ragd011, 2018-11-10:
Also, does this sounds like the same issue?  https://bugs.launchpad.net/ubuntu/+source/nut/+bug/1253861

I tried the fix but it didn't resolve my issue.

clepple, 2018-11-10:
> BTW any chance it has something to do with the ipv6 stuff happening in between?

Sort of related - it looks like NUT may be starting up before the network interface is completely ready.

What if you change `upsd.conf` to `LISTEN 0.0.0.0` instead of specifying the IP address and 127.0.0.1? If you have another network interface that you are trying to keep from seeing the NUT port, I would recommend using firewall rules for that, with interface names (be sure to include the loopback interface).

clepple, 2018-11-11:
> Also, does this sounds like the same issue?
> https://bugs.launchpad.net/ubuntu/+source/nut/+bug/1253861

While you might be able to delay things long enough for the network to come up (as was done in that bug to wait for the USB connection), you'll notice at the top of your log that the UPS was detected in your case.

ragd011, 2018-11-11:
Ha, I changed upsd.conf to 0.0.0.0 like you said and it works first try.  I only have one nic.  Pretty simple setup.  Thanks!

jimklimov, 2018-11-11:
As a side note, I gather you've used packaged NUT from your distros made
from an older-dated stable release? Not a recent Git HEAD?

I did some features related to systemd and SMF integration that were merged
recently, which first I feared could be your problem (hey, young new
fearure) but in the end I think can be your ojt-of-the-box solution as it
rectified dependencies for the service unit startup.

I would appreciate if you can try a build from current github upstream and
post back any good or bad news :)

Also, for usbhid drivers you may be interested to merge a libusb-1.0 (or
0.1+1.0-compat) branch on top of current master, these are currently in PR
state: the newer libusb code should be more stable over long runs, e.g.
overcome usb hub intermittent timeouts and disconnects, I believe.

Hope this helps,
Jim Klimov

ragd011, 2018-11-12:
Sure, I'll give it a shot.  So Basically apt remove what I have, delete configs, then clone the master and build/install it?  I think the wiki had how to do the drivers?

jimklimov, 2018-11-13:
Yes, basically so :)

To make as much as possible a drop-in replacement of NUT for your distro,
you can start from the configure options in its source package recipe of
your distro (same paths and all); then maybe take care of new options if
there are any, and register the systemd units after a make install.

For reference our product fork has some debian packaging recipes at
https://github.org/42ity/nut/tree/FTY/obs (beware that spec variant there
is an old placeholder and may be not in good shape).

Jim

jimklimov, 2018-11-13:
Oh, and you do not have to remove configs, especially if same paths would
be used. Normally an upgrade should be transparent to end users. A switch
from package to BYOD may be not so, but i wouldn't start by removing
configs (but might take another look at users allowed to read them) :)