networkupstools/nut #634: POWERCOM RPT-600AP unable to set battery.charge.*

bikrus, 2018-10-28:
Hi!
### Issue
Either of `battery.charge.*` in `/etc/nut/ups.conf` doesn't work even after reboot.
```ini
maxretry = 3
[RPT-600AP]
        driver = usbhid-ups
        port = auto
        vendorid = 0d9f
        battery.charge.low: 70
        battery.charge.warning: 80
```
```ini
maxretry = 3
[RPT-600AP]
        driver = usbhid-ups
        port = auto
        vendorid = 0d9f
        ignorelb
        override.battery.charge.low: 70
        override.battery.charge.warning: 80
```
The output of `upsc RPT-600AP` still show old values
```console
$ upsc RPT-600AP
...
battery.charge.low: 10
battery.charge.warning: 30
...
```

> In case of missing some valuable info - please, point me.

### Product info
**Related product report**: [[HCL] POWERCOM RPT-600AP supported by usbhid-ups](https://github.com/networkupstools/nut/issues/633)

bikrus, 2018-10-28:
Sorry. My mistake. Instead of ` = ` I have misstyped ` : ` in param set. Next config snippet does work.
```ini
[RPT-600AP]
        driver = usbhid-ups
        port = auto
        vendorid = 0d9f
        ignorelb
        override.battery.charge.low = 70
        override.battery.charge.warning = 80
```
```console
$ upsc RPT-600AP
...
battery.charge.low: 70
battery.charge.warning: 80
...
```

clepple, 2018-10-28:
Glad that worked!

closed 2018-10-28 by clepple