networkupstools/nut #637: Tripp-Lite SU3000LCD2UHV support issue.

Dragon201701, 2018-11-30:
Do nut support SU3000LCD2UHV?

clepple, 2018-12-01:
Probably via either the [`usbhid-ups`](https://networkupstools.org/docs/man/usbhid-ups.html) driver (or possibly `tripplite-usb`), since the [datasheet](https://assets.tripplite.com/product-pdfs/en/su3000lcd2uhv.pdf) says "USB (HID enabled)". The serial drivers have not been updated for many of the Tripp Lite models.

If you have one connected to a Linux system, what does `lsusb -v -d 09ae:` return?

Dragon201701, 2018-12-01:
[image]

Dragon201701, 2018-12-01:
[image]

Dragon201701, 2018-12-01:
I have run these commend and try all three drivers early today and found these stuff

clepple, 2018-12-01:
What if you add `productid = 1330` to `ups.conf` and retry with `usbhid-ups`?

clepple, 2018-12-01:
I thought that name sounded familiar. There may be an issue with just using `productid=1330` on NUT 2.7.4, but we did merge some changes for this model into the master branch: https://github.com/networkupstools/nut/pull/297

Dragon201701, 2018-12-06:
I got 
`
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Can't claim USB device [09ae:1330]: could not detach kernel driver from interface 0: Operation not permitted
Driver failed to start (exit status=1)
`
The official driver should support serial but I keep getting an RI at the stop bit if I do `sudo cat /proc/tty/driver/serial' and I have set port to 115200bps, 8 data bits, no parity, one stop bit with command:`115200bps, 8 data bits, no parity, one stop bit`

clepple, 2018-12-07:
> Can't claim USB device [09ae:1330]: could not detach kernel driver from interface 0: Operation not permitted Driver failed to start (exit status=1)

Running as root?

clepple, 2018-12-07:
I got this issue mixed up with another one. Since you had to add `productid=1330`, you would also need to either adjust the udev rules (which distro is this?), or also add `user=root` to the config file (which is only good for testing the driver, otherwise you have to also adapt the upsd configuration).

clepple, 2018-12-07:
Not sure what you mean about the serial port. Do you mean Tripp Lite's official driver?

The NUT serial `tripplite` driver has not been updated for any of the newer protocols. As far as baud rate, it is probably going to use 2400bps. Also, you may need to completely shut down the UPS when changing between the serial and USB ports.

Dragon201701, 2018-12-07:
Well, this product is a SmartOnline product so should I use tripplitesu instead? I have been trying that for this entire week but I found I can't even communicate to UPS if I do echo "hello" > /dev/ttyS0 while cat them in a separate terminal.

Dragon201701, 2018-12-07:
I do `sudo -i` then `/sbin/upsdrvctl start' and got 'Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
This TrippLite device (09ae:1330) is not (or perhaps not yet) supported
by usbhid-ups. Please make sure you have an up-to-date version of NUT. If
this does not fix the problem, try running the driver with the
'-x productid=1330' option. Please report your results to the NUT user's
mailing list <nut-upsuser@lists.alioth.debian.org>.
'

Dragon201701, 2018-12-07:
So how do I rebuild a driver?

Dragon201701, 2018-12-07:
or perhaps serial working better than USB?

clepple, 2018-12-09:
> So how do I rebuild a driver?

As mentioned before, instructions are on the wiki: 
https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

> or perhaps serial working better than USB?

I'm not sure how else to explain this - someone got a 09ae:1330 UPS working with USB (just not with plain NUT 2.7.4). That's what I would recommend trying.

The serial drivers haven't been updated for many years. The Tripp-Lite serial protocol is not well documented, so you would have to do a lot more research to make that work.