networkupstools/nut #643: Is the ups.status flag CAL enough to support reporting tests in progress?

aquette, 2018-12-10:
the ups.status ['CAL' flag](https://github.com/networkupstools/nut/blob/master/docs/new-drivers.txt#L225) is documented as: "UPS is currently performing runtime calibration (on battery)"

On the other side, we have the ['ups.test.result]'(https://github.com/networkupstools/nut/blob/master/docs/nut-names.txt#L94) variable and the related commands](https://github.com/networkupstools/nut/blob/master/docs/nut-names.txt#L638), among which some may induce a switch to battery.

Is it enough to support reporting tests in progress? Or is another status needed ('TEST')

An audit is needed to see:
* how 'CAL' flag is used throughout the code-base
* how 'ups.test.result' variable is used throughout the code-base
* how the 'test.*' commands are used throughout the code-base
* depending on the result above
** possibly update the code-base to use the 'CAL' or 'TEST' flag
** update doc

clepple, 2018-12-11:
Given that we frequently don't know whether vendors are actually performing runtime calibration, or just checking that the voltage doesn't drop below a threshold when on battery, I don't see much value in another status value. It seems more useful me to document the discrepancy between the flag and the result variable, and move on. When information is known about the calibration method, that can be added to a model-specific or vendor-specific page in the DDL.

Similarly, I do not see any value in changing `ups.test.result` to `device.test.result`. Most of the supported devices are UPSes, anyway.

aquette, 2018-12-11:
hi @clepple, thanks for your answer and helping clarifying the topic.

indeed for the new flag. This question to self came out from the netxml-ups driver pushing in ups.test.result and not using the CAL flag to expose that the current OB - on battery is tied to a test and not to a line failure.
I'll try to make a PR to illustrate this.

as for moving test.result from ups collection to device, it's due to the scope that I / we (42ITy / Eaton) are using NUT. For example, ATS have ups.test.result, which seems strange as any ups.* data published with devices that are not UPSs