networkupstools/nut #645: Wrong values for Bluewalker PowerWalker VI 750 R1U

Th3M1k3y, 2018-12-28:
I am using a raspberry pi to monitor my UPS. NUT has detected the model number, and some other things, but battery charge is constantr at 50, no matter if battery is full or not, output.voltage is always 123.4 too, where it should be 230-231, input voltage is 102, but should be 230 too.

I tried to increase poolfreq and poolinterval, as it seems to have fixed problems for someone else, but it didnt change anything for me.

```
pi@raspberrypi:~ $ upsc ups
Init SSL without certificate database
battery.charge: 50
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: 1
battery.runtime: 300
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 72.0
battery.voltage.nominal: 12
device.mfr: 1
device.model: 750R
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 240
driver.parameter.pollinterval: 120
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 14
input.transfer.low: 9
input.voltage: 25.2
input.voltage.nominal: 102
output.voltage: 123.4
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 5
ups.mfr: 1
ups.model: 750R
ups.productid: 0601
ups.realpower.nominal: 999
ups.status: OL BOOST
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

```
maxretry = 3

[ups]
    driver = usbhid-ups
    port = auto
    desc = "Bluewalker PowerWalker VI 750 R1U"
    pollfreq=240
    pollinterval=120
```

Could this be a config problem, or does NUT just not work with this UPS?

clepple, 2018-12-28:
It isn't clear whether the UPS is just not providing those values, or if NUT is incorrectly applying fixes intended for CyberPower UPSes.

We will need a debug log from the driver.

1. Stop the NUT daemons: `sudo service nut-client stop; sudo service nut-server stop`
2. Start the driver in debug mode: `sudo /lib/nut/usbhid-ups -a ups -DDD 2>&1 | tee /tmp/powerwalker-VI-750.txt`
3. Let it run for ~ 30 seconds, then stop with Control-C.
4. Restart the daemons: `sudo service nut-server start; sudo service nut-client-start`
5. Attach the log file here.

Th3M1k3y, 2018-12-28:
Thank you for your reply, here is the log generated.

[powerwalker-VI-750.txt](https://github.com/networkupstools/nut/files/2715271/powerwalker-VI-750.txt)

Easy to notice this...
```
  30.450556	upsdrv_updateinfo...
  30.701636	libusb_get_interrupt: Connection timed out
  30.702692	Got 0 HID objects...
```

clepple, 2018-12-28:
I keep seeing things like this:
```
   0.192872	Report[get]: (3 bytes) => 0e e6 00
   0.192919	Path: UPS.Input.ConfigVoltage, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 16, Value: 102
```

Report ID is 0x0e, and the value is 0x00e6 (little-endian). 0xe6 is 230 decimal, but NUT is picking up a scaling factor from somewhere else in the report descriptor, and gets 102 instead. (It's like leaving a parenthesis open by mistake.)

Theoretically, we can fix this after #169 is done, but I would also recommend trying the vendor's software (maybe on a laptop or in a VM) to see if the values are actually reported properly there. The value 123.4 looks suspiciously like a placeholder.

The timeout is not an actual error - some UPSes use USB interrupt transfers (where NUT asks the UPS if *any* data is ready), and others only respond to polling (where NUT asks what the value is for specific HID reports). You could set the `pollonly` option if you wanted to prevent the interrupt transfers (if it was spamming the kernel log, etc). Reference: https://networkupstools.org/docs/man/usbhid-ups.html

Th3M1k3y, 2018-12-28:
It is working fine with their own software.

[image]

That problem was opened 4 years ago, I assume there isnt anything close to being fixed yet?

IndrekHaav, 2019-01-23:
I have a VI 500 R1U and am seeing the same issue - wrong values in `upsc` output when using the usbhid-ups driver.

I initially tried the blazer_usb and nutdrv_qx drivers, but they didn't work. `lsusb` identified the UPS as a "Cyber Power System, Inc. PR1500LCDRT2U UPS" so I tried usbhid-ups on a whim, and it managed to at least communicate with the UPS. Also, some basic commands like enabling/disabling the beeper and running a light battery test with `upscmd` seem to work.

hoch1009, 2019-04-15:
@Th3M1k3y @IndrekHaav do you have any workarounds or are you using nut as it is (with wrong values)?

IndrekHaav, 2019-04-15:
> @Th3M1k3y @IndrekHaav do you have any workarounds or are you using nut as it is (with wrong values)?

I ended up getting the management card for the UPS, so I'm not using NUT at all.

psychogun, 2020-04-25:
Any news on this?

Th3M1k3y, 2020-04-26:
> Any news on this?

Only news from my end, is that the UPS is useless. Every 4-5 months it switches the output off. To get it to work again, I have to pull mains power out, which causes it to shut down immediately, instead of continuing on battery power. Plugging mains back in makes it start up again, and showing fully charged battery. Then it can run another 4-5 months... So the UPS is more unreliable than my mains supply. Can't RMA is, as it work just fine when you power it up, and if it works then, they will send a bill for the time spend testing...

psychogun, 2020-04-26:
What is the management card for this UPS? Is it the SNMP cards? They are almost the same price as the UPS itself. .

IndrekHaav, 2020-04-26:
> What is the management card for this UPS? Is it the SNMP cards? They are almost the same price as the UPS itself. .

This one: https://powerwalker.com/?page=product&item=10131008&lang=en

And yes, it costs about as much as the UPS itself.

FWIW, my VI 500 R1U has been rock-solid for over a year.

psychogun, 2020-04-26:
Yeah, well- I am looking for a 1U UPS in a similar price range of the VI 750 which could power off my pfsense / Proxmox environment. But if it does not work, yet, with NUT .. Still searching, do not know yet if there are any alternatives. Thanks

jimklimov, 2024-12-18:
PR #2718 is presumed to have fixed the broken CPS input/output voltage reports discussed in this issue.

I am not sure if `battery.charge` got fixed by any other changes, keeping open for now. Testing of current codebase on the device is welcome.

jimklimov, 2025-03-19:
Presuming fixed in current code base, until proven otherwise.

tux86, 2026-10-01:
Tested NUT v2.8.5 (usbhid-ups 0.71, cps-hid) on a PowerWalker VI 750 R1U (0764:0601, bcdDevice 2.00), Debian 13, against the distro's 2.8.1:

- `output.voltage`: 25.2 on 2.8.1, **230.0 on 2.8.5**: the fix from #2718 works on this unit.
- `battery.charge`: correct on both (tracks a recharge from 81 to 98%), so the "stuck at 50%" from the original report does not reproduce here.
- `input.voltage.nominal`: 0 on both. The device itself reports `UPS.Input.ConfigVoltage = 0` (report 0x0e), so this looks like a firmware limitation rather than a driver bug.
- Minor: `battery.type` (`PbAcid` on 2.8.1) and `battery.mfr.date` read empty on 2.8.5.

One side effect worth noting: after running the 2.8.5 driver once with `-d 1`, the device was left without an active configuration (`bConfigurationValue` empty, kernel: `can't set config #1, error -71` on re-authorization), and the 2.8.1 driver failed with "insufficient permissions" / iManufacturer read errors until `1` was written to `bConfigurationValue`. I have not isolated whether this is the driver's exit path or this unit's USB firmware.

florian-asche, 2026-10-21:
Hi all,
just added this device to my inventory. Here is a full output in case it may helps:
```
battery.charge: 92
battery.charge.low: 0
battery.charge.warning: 20
battery.mfr.date: 1 
battery.runtime: 1800
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.6
battery.voltage.nominal: 12
device.mfr: 1 
device.model: 750R
device.serial: 320024FQ30000060
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 10
driver.parameter.port: auto
driver.parameter.synchronous: yes
driver.state: quiet
driver.version: 2.8.1
driver.version.data: CyberPower HID 0.8
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.frequency: 49.9
input.transfer.high: 290
input.transfer.low: 165
input.voltage: 230.4
input.voltage.nominal: 0
output.frequency: 49.9
output.voltage: 25.2
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 20
ups.mfr: 1 
ups.model: 750R
ups.productid: 0601
ups.realpower.nominal: 450
ups.serial: 320024FQ30000060
ups.status: OL CHRG
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```