networkupstools/nut #648: HCL PowerWalker VFI 1000 CG PF1 supported by nutdrv_qx

bkvamme, 2019-01-15:
PowerWalker VFI 1000 CG PF1 tested successfully with nutdrv_qx

Also works with blazer_usb, do you need an additional issue for this? 

Link to manufacturer: https://powerwalker.com/?page=product&item=10122108&lang=en

```
Init SSL without certificate database
battery.charge: 91
battery.energysave: no
battery.packs: 1
battery.protection: yes
battery.runtime: 240
battery.voltage: 41.00
battery.voltage.nominal: 36.0
device.model: OLHV1K0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Voltronic 0.06
driver.version.internal: 0.28
input.current.nominal: 4.0
input.frequency: 50.1
input.frequency.high: 53.0
input.frequency.low: 47.0
input.frequency.nominal: 50.0
input.phases: 1
input.transfer.high: 242
input.transfer.high.max: 254
input.transfer.high.min: 237
input.transfer.low: 218
input.transfer.low.max: 223
input.transfer.low.min: 206
input.voltage: 229.2
input.voltage.nominal: 230.0
outlet.0.switchable: no
output.current: 0.0
output.current.nominal: 4
output.frequency: 0.0
output.frequency.nominal: 50.0
output.phases: 1
output.power.maximum.percent: 0
output.power.minimum.percent: 0
output.powerfactor: 1.0
output.voltage: 0.0
output.voltage.nominal: 230.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 01574.04
ups.firmware.aux: P01
ups.load: 0
ups.power.nominal: 1000
ups.productid: 5161
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL OFF
ups.temperature: 25.8
ups.type: online
ups.vendorid: 0665
```

bkvamme, 2019-01-18:
'pi@pinut:~ $ upsrw ups
[battery.energysave]
Switch off when running on battery and no/low load
Type: ENUM
Option: "no" SELECTED
Option: "yes"

[battery.packs]
Number of battery packs
Type: RANGE
Option: "1-99" SELECTED

[battery.protection]
Prevent deep discharge of battery
Type: ENUM
Option: "no"
Option: "yes" SELECTED

[input.frequency.high]
Maximum input line frequency (Hz)
Type: RANGE
Option: "53-60" SELECTED

[input.frequency.low]
Minimum input line frequency (Hz)
Type: RANGE
Option: "40-47" SELECTED

[input.transfer.high]
High voltage transfer point (V)
Type: RANGE
Option: "237-254" SELECTED

[input.transfer.low]
Low voltage transfer point (V)
Type: RANGE
Option: "206-223" SELECTED

[outlet.0.switchable]
Description unavailable
Type: ENUM
Option: "no" SELECTED
Option: "yes"

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: RANGE
Option: "12-5940" SELECTED

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: RANGE
Option: "0-599940" SELECTED

[ups.start.auto]
UPS starts when mains is (re)applied
Type: ENUM
Option: "no"
Option: "yes" SELECTED

[ups.start.battery]
Allow to start UPS from battery
Type: ENUM
Option: "no"
Option: "yes" SELECTED

pi@pinut:~ $ upscmd -l ups
Instant commands supported on UPS [ups]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.toggle - Toggle the UPS beeper
bypass.start - Put the UPS in bypass mode
bypass.stop - Take the UPS out of bypass mode
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test'

Beeper disable works. Will test more tomorrow, and update the ticket.

bkvamme, 2019-01-18:
How do you test the shutdown process? Tried pulling the cable, but it didn't shutdown until low battery. Big 73Ah battery, so haven't tried it yet.

zykh, 2019-01-19:
> Tried pulling the cable, but it didn't shutdown until low battery.

Yeah, exactly that... (but I'd use a power strip or the like, to keep things grounded)

Meanwhile, if you don't have time to let the battery discharge, you could at least try `upsdrvctl shutdown`, which should send the shutdown command to the device (but some devices don't accept the command, if they're not on battery).

clepple, 2019-11-06:
@bkvamme feel free to reopen if you have additional details on shutdown, or discrepancies in the variables.

In particular, is `ups.status: OL OFF` correct?

closed 2019-11-06 by clepple