networkupstools/nut #657: Augeas support completion

aquette, 2019-02-01:
while looking back at Augeas for #656, I remembered that there were a number of things still to be done:

* [ ] add a "check" target in scripts/augeas/Makefile.am using 
> augparse -I ./ ./tests/test_nut.aug
* [ ] complete nutupsdconf.aug.in with CERTPATH CERTIDENT CERTREQUEST (Mozilla NSS sec)
* audit for other lenses and new features TBD

aquette, 2019-02-01:
side note: tests are failing on my system:

>$ augparse -I ./ ./tests/test_nut.aug
>Test failure:./tests/test_nut.aug:20.0-25.37:
> Expected:
>{ 
>  {  }
>  { "testups"
>    { "driver" = "dummy-ups" }
>    { "port" = "auto" }
>    { "desc" = ""Dummy UPS"" }
>  }
>}
>
> Actual:
>{ 
>  {  }
>  { "testups"
>    { "driver" = "dummy-ups" }
>    { "port" = "auto" }
>    { "desc" = "Dummy UPS" }
>  }
>}
>
>Syntax error in lens definition
>Failed to load ./tests/test_nut.aug

aquette, 2019-02-26:
on the test failure:
I just had news from Raphael Pinson. He confirmed what I saw, that is
`IniFile.indented_entry` handles quotes (and double quotes) in a specific way that can't be modified. The best would be to look elsewhere (Shellvars) to see how it is managed.

Ref: https://github.com/hercules-team/augeas/blob/master/lenses/shellvars.aug#L39

jimklimov, 2024-04-26:
@aquette : wrapping my head around the test as it exists today:
````
let ups_conf  = "
[testups]
	driver = dummy-ups
	port = auto
	desc = \"Dummy UPS\"
"

test NutUpsConf.ups_lns get ups_conf = 
	{ }
	{ "testups" 
		{ "driver" = "dummy-ups"   }
		{ "port"   = "auto" }
		{ "desc"   = "\"Dummy UPS\""    } }
````

still yielding the discrepancy you saw:
````
:;  make ; augparse -I ./ ./tests/test_nut.aug
cd ../.. && /bin/bash ./config.status scripts/augeas/nutupsmonconf.aug
config.status: creating scripts/augeas/nutupsmonconf.aug
Test failure:./tests/test_nut.aug:20.0-25.37:
 Expected:
{
  {  }
  { "testups"
    { "driver" = "dummy-ups" }
    { "port" = "auto" }
    { "desc" = ""Dummy UPS"" }
  }
}

 Actual:
{
  {  }
  { "testups"
    { "driver" = "dummy-ups" }
    { "port" = "auto" }
    { "desc" = "Dummy UPS" }
  }
}

Syntax error in lens definition
Failed to load ./tests/test_nut.aug
````

BUT: Why do we expect the double quotes inside the `desc` value? If they were to be escaped, there should be more slashes. If they were to be a single token of two words, augeas already does the right thing... no?

jimklimov, 2024-04-27:
Generally it seems quite complicated to conjure up lenses that would support several quotes in the same line, whether "plaintext" lines with multiple tokens like `CERTIDENT "My server" "Db Pass W0rd"` or escaped inside single text tokens like `desc = "A \"Dummy\" UPS"` which NUT parser allows but augeas struggles with.

jimklimov, 2024-04-29:
With the PR above, I've added a `scripts/augeas/tests/test_nut_fixme.aug` file with some test cases that currently fail for existing NUT augeas lens definitions and inputs that should be valid for NUT.

Conversely, I think one case in the main test file is passing despite being invalid:
````
desc "\"Some text\"
````
The trailing `\"` would be part of the token enclosed into the first opening double-quote. There is no ending quote so the line is... unfinished? Multi-line? Aborted by default?