networkupstools/nut #661: UPSD too eager to report Data stale

jimklimov, 2019-02-11:
On our relatively busy NAS server there are occasionally console messages that `Data for UPS X is stale`. In original default settings, the messages for `data is no longer stale` do not get to the console (which looked worrysome, to see only lots of failures, until I updated logging settings).

Now that both types of messages are here, I see that stale and not-stale often happen in the same second, so with a bit of patience much of this spam could be avoided :)

To me it seems like some off-by-one math error in patience calculation, or perhaps a hint to induce one to avoid spurious messages on systems a bit too busy to context-switch fast?

From cases I've seen before in different codebases, maybe there is a rounding error of sec/msec/usec (including granularity of time reports from an OS - more pronounced for filesystem objects which may be off by 1-2 seconds) and comparisons of those numbers assuming infinite (or at least same) precision?

aquette, 2019-02-26:
privet @jimklimov 
I'd like to make a setup to reproduce this, since the fix may not be that trivial.
Questions:
* which driver are you using?
* are you using the default MAXAGE in upsd.conf (15 seconds)?

AndiHe, 2019-03-19:
We are plagued by the same problem. In our case we would like to monitor a Eaton 93PS and tried netxml-ups and snmp-ups and get the stale/no longer stale cycle with both.
I would be able to provide tcpdump data and we are able to test/compile patches on that machine if someone is interested in solving the problem.

AndiHe, 2019-03-19:
Might this be solved with https://github.com/networkupstools/nut/issues/542 ??

jimklimov, 2019-03-21:
@aquette : for the device in that issue, it was a generic `usbhid-ups`:

````
[eaton]
    driver = usbhid-ups
    synchronous = yes
    port = /dev/usb/463.ffff/0
    desc = "EATON Protection Station AN2E49008"
#    pollonly = true
````

The `MAXAGE` is not customized, so should be 15 sec per comments in the config./

jimklimov, 2025-11-09:
In `usbhid-ups.c` the `dstate_datastale()` is called in case of a failed reconnection.