networkupstools/nut #674: UPS DigiTECH Computer 650VA Not Supported

guzzim, 2019-03-05:
This UPS appears to be Megatec like.  If I run the NUT driver nutdrv_qx first, it fails to start.  However, if I run the manufacturers software first (a linux binary supplied on a cd), close it, then start nutdrv_qx, it starts but displays the last value the OEM software retrieved.  I have experimented with other combinations of driver and subdriver, all without success.

The OEM software is called UPSmart by Guangdong IDBK.  The software allows you to fire a battery test and mute the beeper (only if it's already sounding).

I'd be happy with just the ability to read the the values.  Mute the beeper can be solved with a soldering iron.

./nutdrv_qx -s nutdev1 -x protocol="megatec" -x subdriver="fabula" -x vendorid="0001" -x productid="0000" -x port="auto"
 0.010091	[D2] Checking device (0001/0000) (001/011)
   0.011510	[D2] - VendorID: 0001
   0.011527	[D2] - ProductID: 0000
   0.011533	[D2] - Manufacturer: unknown
   0.011537	[D2] - Product: MEC0003
   0.011542	[D2] - Serial Number: unknown
   0.011546	[D2] - Bus: 001
   0.011551	[D2] - Device release number: 0100
   0.011555	[D2] Trying to match device
   0.011597	[D2] Device matches
   0.011649	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.011668	[D2] Skipping protocol Voltronic 0.06
   0.011674	[D2] Skipping protocol Voltronic-QS 0.07
   0.011682	[D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.011688	[D2] Skipping protocol Mustek 0.07
   0.011694	[D2] Skipping protocol Megatec/old 0.07
   0.011700	[D2] Skipping protocol BestUPS 0.06
   0.011708	[D2] Skipping protocol Mecer 0.07
   0.011716	[D3] send: Q1
   0.014763	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.014836	[D3] send: I
   0.017451	[D3] read: #                                     
   0.017474	[D1] ups_infoval_set: non significant value [ups.firmware]
   0.017480	[D3] send: Q1
   0.020399	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.020417	[D3] send: I
   0.023125	[D3] read: #                                     
   0.023150	[D1] ups_infoval_set: non significant value [ups.firmware]
   0.023166	[D3] send: Q1
   0.026382	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.026418	[D3] send: I
   0.029145	[D3] read: #                                     
   0.029169	[D1] ups_infoval_set: non significant value [ups.firmware]
   0.029180	[D2] Skipping protocol Zinto 0.06
   0.029186	[D2] Skipping protocol Q1 0.07
   0.029192	Device not supported!
   0.029200	Device not supported!

After opening and then closing the OEM software, then starting the nutdrv_qx:

   0.008463	[D2] Checking device (0001/0000) (001/011)
   0.009833	[D2] - VendorID: 0001
   0.009860	[D2] - ProductID: 0000
   0.009868	[D2] - Manufacturer: unknown
   0.009875	[D2] - Product: MEC0003
   0.009882	[D2] - Serial Number: unknown
   0.009889	[D2] - Bus: 001
   0.009911	[D2] - Device release number: 0100
   0.009914	[D2] Trying to match device
   0.009941	[D2] Device matches
   0.009989	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.010000	[D2] Skipping protocol Voltronic 0.06
   0.010004	[D2] Skipping protocol Voltronic-QS 0.07
   0.010009	[D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.010017	[D2] Skipping protocol Mustek 0.07
   0.010023	[D2] Skipping protocol Megatec/old 0.07
   0.010027	[D2] Skipping protocol BestUPS 0.06
   0.010030	[D2] Skipping protocol Mecer 0.07
   0.010035	[D3] send: Q1
   0.013097	[D3] read: (248.0 000.0 250.0 000 50.1 13.7 29.0 00001000
   0.013131	[D3] send: I
   0.015728	[D3] read: #                           V3.7      
   0.015781	Using protocol: Megatec 0.06
[... 773 lines cut ...]
	0x0050:  3000 3000 3000 2e00 3000 2000 3200 3500  0.0.0...0...2.5.
	0x0060:  3000 2e00 3000 2000 3000 3000 3000 2000  0...0...0.0.0...
	0x0070:  3400 3900 2e00 3700 2000 3100 3300 2e00  4.9...7...1.3...
	0x0080:  3700 2000 3200 3900 2e00 3000 2000 3000  7...2.9...0...0.
	0x0090:  3000 3000 3000 3100 3000 3000 3000 0d00  0.0.0.1.0.0.0...
00:23:18.832097 CONTROL SUBMIT to 1:13:0
	0x0000:  8024 ddc4 f59a ffff 5302 800d 0100 003c  .$......S......<
	0x0010:  76a2 7e5c 0000 0000 61b2 0c00 8dff ffff  v.~\....a.......
	0x0020:  6600 0000 0000 0000 8006 0c03 0904 6600  f.............f.
	0x0030:  0000 0000 0000 0000 0002 0000 0000 0000  ................
00:23:18.834195 CONTROL COMPLETE from 1:13:0
	0x0000:  8024 ddc4 f59a ffff 4302 800d 0100 2d00  .$......C.....-.
	0x0010:  76a2 7e5c 0000 0000 93ba 0c00 0000 0000  v.~\............
	0x0020:  5000 0000 5000 0000 0000 0000 0000 0000  P...P...........
	0x0030:  0000 0000 0000 0000 0002 0000 0000 0000  ................
	0x0040:  5003 2300 2000 2000 2000 2000 2000 2000  P.#.............
	0x0050:  2000 2000 2000 2000 2000 2000 2000 2000  ................
	0x0060:  2000 2000 2000 2000 2000 2000 2000 2000  ................
	0x0070:  2000 2000 2000 2000 2000 5600 3300 2e00  ..........V.3...
	0x0080:  3700 2000 2000 2000 2000 2000 2000 0d00  7...............
00:23:18.834692 INTERRUPT COMPLETE to 1:13:1
	0x0000:  0039 a134 f69a ffff 4301 810d 0100 2d00  .9.4....C.....-.
	0x0010:  76a2 7e5c 0000 0000 84bc 0c00 feff ffff  v.~\............
	0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0030:  0800 0000 0000 0000 0002 0000 0000 0000  ................

zykh, 2019-03-18:
@guzzim: could you test #638, please?

guzzim, 2019-03-18:
@zykh Thank you for taking a look at this onen.  I'm a bit of a n00b with git, but I checked out github.com/marianojan/nut.git and aftering making the drivers I'm happy to report that it's working.  Readings from the driver match and change with the LCD display on the UPS.

Is there a way to git clone the master and integrate Mariano hunnox drivers?

zykh, 2019-03-19:
@guzzim: could you please post the output of `nutdrv_qx` launched with a debug level of 5 (let it run for a couple of minutes)?

> I checked out github.com/marianojan/nut.git

At this point, you should have also switched to the `hunnox-hnx850` branch (not that this would have made a big difference, given that Mariano's `master` is very similar to that branch).

> Is there a way to git clone the master and integrate Mariano hunnox drivers?

Not sure I understand what you mean.

If you want to have your local copy of Mariano's branch updated to an 'official' `master` branch, assuming your repository is paired (and updated) to https://github.com/networkupstools/nut (i.e. originated with `git clone https://github.com/networkupstools/nut.git <localdir>`), and that you're on its `master` branch, you could do something like this:
1. add (and update) a new 'remote' to it (e.g. `git remote add -f marianojan https://github.com/marianojan/nut.git`),
2. create/checkout a new branch (`git checkout -b <branchname>`),
3. merge Mariano's branch into this new branch (`git merge marianojan/hunnox-hnx850`), resolving any possible conflict.

guzzim, 2019-03-20:
Thanks for that @zykh .  I think i've done this correctly.

1. `git clone https://github.com/networkupstools/nut.git`
1. cd nut
1. `git remote add -f marianojan https://github.com/marianojan/nut.git`
1. `git merge marianojan/hunnox-hnx850`
I had a conflict with the drivers/Makefile.am, which i was able to clear up (add in the hunnox driver)

Settings used in ups.conf
```
[myups]
  driver = "nutdrv_qx"
  desc = "DigiTech 650VA UPS"
  port = "auto"
  vendorid = "0001"
  productid = "0000"
  protocol = "hunnox"
  langid_fix = "0x0409"
  novendor
  noscanlangid
```

Debug output from the driver.  Part way through i disconnect the mains so you can see the results while on battery.
```
$ ./nutdrv_qx -a myups -DDDDD
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4-701-gf745a34c)
USB communication driver 0.33
   0.000000	[D1] debug level is '5'
   0.003621	[D5] send_to_all: SETINFO device.type "ups"
   0.003645	[D1] upsdrv_initups...
   0.003677	[D2] Language ID workaround enabled (using '0x409')
   0.016095	[D3] usb_busses=0xd3e040
   0.016111	[D2] Checking device (413C/2513) (002/004)
   0.016127	[D2] Failed to open device, skipping. (Permission denied)
   0.016131	[D2] Checking device (413C/818E) (002/003)
   0.016138	[D2] Failed to open device, skipping. (Permission denied)
   0.016149	[D2] Checking device (8087/0024) (002/002)
   0.016156	[D2] Failed to open device, skipping. (Permission denied)
   0.016161	[D2] Checking device (1D6B/0002) (002/001)
   0.016168	[D2] Failed to open device, skipping. (Permission denied)
   0.016172	[D2] Checking device (0C45/648B) (001/005)
   0.016179	[D2] Failed to open device, skipping. (Permission denied)
   0.016182	[D2] Checking device (413C/8197) (001/004)
   0.016188	[D2] Failed to open device, skipping. (Permission denied)
   0.016192	[D2] Checking device (0001/0000) (001/006)
   0.017824	[D2] - VendorID: 0001
   0.017851	[D2] - ProductID: 0000
   0.017874	[D2] - Manufacturer: unknown
   0.017883	[D2] - Product: MEC0003
   0.017905	[D2] - Serial Number: unknown
   0.017907	[D2] - Bus: 001
   0.017910	[D2] - Device release number: 0100
   0.017913	[D2] Trying to match device
   0.017936	[D2] Device matches
   0.017945	[D2] failed to claim USB device: Device or resource busy
   0.018318	[D2] detached kernel driver from USB device...
   0.018362	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.018380	[D5] send_to_all: SETINFO ups.vendorid "0001"
   0.018391	[D5] send_to_all: SETINFO ups.productid "0000"
   0.018915	[D1] First supported language ID: 0x409 (please report to the NUT maintainer!)
   0.018933	[D2] Skipping protocol Voltronic 0.06
   0.018940	[D2] Skipping protocol Voltronic-QS 0.07
   0.018947	[D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.018953	[D2] Skipping protocol Mustek 0.07
   0.018959	[D2] Skipping protocol Megatec/old 0.07
   0.018965	[D2] Skipping protocol BestUPS 0.06
   0.018971	[D2] Skipping protocol Mecer 0.07
   0.018977	[D2] Skipping protocol Megatec 0.06
   0.018984	[D2] Skipping protocol Zinto 0.06
   0.018993	[D3] send: Q1
[... 847 lines cut ...]
 110.167824	[D5]  30 30 30 0d
 110.167834	[D3] read: (246.0 000.0 248.0 000 50.0 13.5 29.0 00001000
 110.167888	[D5] send_to_all: SETINFO output.voltage "248.0"
 110.167913	[D5] send_to_all: SETINFO input.frequency "50.0"
 110.167929	[D5] send_to_all: SETINFO battery.voltage "13.50"
 110.167946	[D5] update_status: OL
 110.167960	[D5] update_status: !LB
 110.167985	[D5] update_status: !CAL
 110.168008	[D5] update_status: !FSD
 112.167335	[D1] upsdrv_updateinfo...
 112.167384	[D1] Quick update...
 112.167398	[D3] send: Q1
 112.167410	[D4] command index: 0x03
 112.167425	[D3] asking for: 0C
 112.172482	[D5] read: (47 bytes) => 28 32 34 36 2e 30 20 30 30 30 2e 30 20 32 34 38 2e 30
 112.172531	[D5]  20 30 30 30 20 34 39 2e 39 20 31 33 2e 35 20 32 39 2e 30 20 30 30 30 30 31
 112.172543	[D5]  30 30 30 0d
 112.172554	[D3] read: (246.0 000.0 248.0 000 49.9 13.5 29.0 00001000
 112.172609	[D5] send_to_all: SETINFO input.frequency "49.9"
 112.172642	[D5] update_status: OL
 112.172656	[D5] update_status: !LB
 112.172671	[D5] update_status: !CAL
 112.172688	[D5] update_status: !FSD

```

guzzim, 2019-03-20:
This error seems to occur on some polls.

```
 102.142383     [D1] upsdrv_updateinfo...
 102.142434     [D1] Quick update...
 102.142448     [D3] send: Q1
 102.142460     [D4] command index: 0x03
 102.142478     [D3] asking for: 0D
 102.146433     [D5] read: (47 bytes) => 2a 32 34 34 2e 30 20 30 30 30 2e 30 20 32 34 38 2e 30
 102.146482     [D5]  20 30 30 30 20 35 30 2e 30 20 31 33 2e 37 20 32 39 2e 30 20 30 30 30 30 31
 102.146494     [D5]  30 30 30 0d
 102.146517     [D3] read: *244.0 000.0 248.0 000 50.0 13.7 29.0 00001000
 102.146535     [D2] qx_process_answer: input.voltage - invalid start character [2a], expected [28]
 102.146550     [D1] Communications with the UPS lost: status read failed!

 106.146519     [D1] upsdrv_updateinfo...
 106.146561     [D1] Quick update...
 106.146576     [D3] send: Q1
 106.146586     [D4] command index: 0x03
 106.146598     [D3] asking for: 0C
 106.151299     [D5] read: (47 bytes) => 2a 32 34 35 2e 30 20 30 30 30 2e 30 20 32 34 38 2e 30
 106.151333     [D5]  20 30 30 30 20 35 30 2e 30 20 31 33 2e 37 20 32 39 2e 30 20 30 30 30 30 31
 106.151344     [D5]  30 30 30 0d
 106.151353     [D3] read: *245.0 000.0 248.0 000 50.0 13.7 29.0 00001000
 106.151367     [D2] qx_process_answer: input.voltage - invalid start character [2a], expected [28]
 106.151382     [D1] Communications with the UPS lost: status read failed!

 230.272058     [D1] upsdrv_updateinfo...
 230.272111     [D1] Quick update...
 230.272125     [D3] send: Q1
 230.272137     [D4] command index: 0x03
 230.274811     [D5] read: (47 bytes) => 2a 32 34 35 2e 30 20 30 30 30 2e 30 20 32 34 38 2e 30
 230.274873     [D5]  20 30 30 30 20 34 39 2e 39 20 31 33 2e 37 20 32 39 2e 30 20 30 30 30 30 31
 230.274900     [D5]  30 30 30 0d
 230.274931     [D3] read: *245.0 000.0 248.0 000 49.9 13.7 29.0 00001000
 230.274960     [D2] qx_process_answer: input.voltage - invalid start character [2a], expected [28]
 230.274981     [D1] Communications with the UPS lost: status read failed!

 258.300110     [D1] upsdrv_updateinfo...
 258.300163     [D1] Quick update...
 258.300177     [D3] send: Q1
 258.300189     [D4] command index: 0x03
 258.300208     [D3] asking for: 0D
 258.304477     [D5] read: (47 bytes) => 15 32 34 35 2e 30 20 30 30 30 2e 30 20 32 34 38 2e 30
 258.304540     [D5]  20 30 30 30 20 35 30 2e 30 20 31 33 2e 37 20 32 39 2e 30 20 30 30 30 30 31
 258.304569     [D5]  30 30 30 0d
 258.304605     [D3] read: ^U245.0 000.0 248.0 000 50.0 13.7 29.0 00001000
 258.304639     [D2] qx_process_answer: input.voltage - invalid start character [15], expected [28]
 258.304702     [D1] Communications with the UPS lost: status read failed!

 270.312460     [D1] upsdrv_updateinfo...
 270.312503     [D1] Quick update...
 270.312518     [D3] send: Q1
 270.312528     [D4] command index: 0x03
 270.312541     [D3] asking for: 0D
 270.316696     [D5] read: (47 bytes) => 2a 32 34 35 2e 30 20 30 30 30 2e 30 20 32 34 38 2e 30
 270.316733     [D5]  20 30 30 30 20 34 39 2e 39 20 31 33 2e 37 20 32 39 2e 30 20 30 30 30 30 31
 270.316744     [D5]  30 30 30 0d
 270.316754     [D3] read: *245.0 000.0 248.0 000 49.9 13.7 29.0 00001000
 270.316768     [D2] qx_process_answer: input.voltage - invalid start character [2a], expected [28]
 270.316781     [D1] Communications with the UPS lost: status read failed!
```

zykh, 2019-03-31:
@guzzim: glad #638 worked and thanks for the logs (and `ups.conf` options), really useful.

As for the errors, you could try increasing `pollinterval`, but if they are really frequent, we could consider ignoring the first character of the reply (as we already do, for example, in the `krauler` USB subdriver).

germanbour, 2019-07-15:
@guzzim Hola, tengo la misma ups y no logro hacerla funcionar, soy nuevo con git no se como compilar el proyecto menos como uniste las dos ramas. Si me explicas te agradecere me seria de mucha ayuda. Saludos.

commit dc0ae878 (2022-04-25): driver.list.in: Add HCL info for DigiTECH 650VA (USB)

Closes: #674
closed 2022-04-25 by jimklimov