networkupstools/nut #676: fix invocation of isdigit(), or use str* API

clepple, 2019-03-13:
`isdigit()` takes a char/int, not a char*.

[Originally mentioned here](
https://github.com/networkupstools/nut/commit/9a40084a5c756fa47c436819c2b15c41fa91d93a#commitcomment-32558936), but we need to fix this before the next release.

clepple, 2019-04-29:
I'm bumping the milestone from 2.7.5 to 2.7.6 because we merged #690, which fixes the misuse of `isdigit()`. I still think that a slightly reworked version of #677 (with signed/unsigned expanded out into separate variables, etc.) should be merged into the following release.

user01010111, 2026-09-01:
PR #677 remains conflicting and changes-requested. Would you be happy for me to prepare a fresh parsing-only PR from current master that supersedes it, preserves the existing public types, and reuses the current `str_to_*` helpers?

jimklimov, 2026-09-01:
@user01010111 : generally yes, but maybe a bit later - the CI resources are already overwhelmed and will be for a few days to come, with currently open PRs (which is generally a good thing) :)

I also hoped to tackle different C-str, C-bool etc. cleanups after finally merging the FTY branch into main code, to avoid fixing those troubles twice.

user01010111, 2026-09-01:
@jimklimov, just keeping the machines feed 😂

Have mostly just been trying to clear the lower-hanging fruit for 2.8.6 and addressing the other items where I can be helpful without (hopefully) treading on anyone's toes.

Ping me in here when you'd like me to pick it up, and I can continue if so. Thanks, Jim.

closed 2026-09-14 by jimklimov