networkupstools/nut #695: Each time I reboot my computer, nut-driver.service fails to start:

kevindd992002, 2019-05-12:
So everytime I reboot my Debian machine, I see the nut-driver.service failing to start. When I check through systemctl, it is loaded but not active. I'm using this Debian box as a simple slave monitoring client that is pointed to a NUT master in my pfsense firewall/gateway box (where my actual UPS is connected). In that scenario, I would imagine that the driver service shouldn't be loaded at all because it is not being used anyway. Do I need to do something else to keep this driver from loading in the first place, so that my Debian box isn't in a "degraded" state?

clepple, 2019-05-12:
> I would imagine that the driver service shouldn't be loaded at all because it is not being used anyway.

Correct, the driver service is not being used in a monitoring-only setup.

You didn't specify which version of the NUT package you are using (or the version of Debian), but the most likely way to avoid this for a client-only machine is to remove the `nut-server` package. If you installed the `nut` meta-package, you might need to reinstall `nut-client`, or mark it as not automatically installed.

There is also an `/etc/nut/nut.conf` file which is intended to help manage the various client/server/standalone scenarios, but it is from the SysV init days, and I haven't been tracking how that works in the presence of systemd. It might still leave the system in a degraded state.

kevindd992002, 2019-05-14:
Right. Before I read your response I noticed that there is a nut-server and a nut-client package installed. So I removed nut and installed only the nut-client, which solved the problem.

Before doing all these, I also did configure nut.conf to mode=netclient. So that should've disabled the driver service altogether but I guess you're right that is not working with systemd.

clepple, 2019-05-14:
Since it sounds like your setup is working, I'm going to close this issue.

Ultimately, the startup scripts are handled by the packagers, so I would recommend filing a bug with Debian if you need `nut.conf` to interact differently with the systemd units.

closed 2019-05-14 by clepple