networkupstools/nut #697: Correct Solis driver voltage readings - APC BackUPS only

rpvelloso, 2019-05-16:
Solis driver outputs voltage at 100% at all times and hence, there is no way to detected low battery. This behaviour was observed in APC BackUPS BR. The voltage readings for other Solis models are "probably" correct since the driver was originally coded for it.

I have the source code from the manufacturer for this specific model.

rpvelloso, 2019-05-16:
As soon as I have the time I'll start doing this, opened the issue so I don't forget it or if anybody wants the source to transcribe it to NUT/Solis just send me an email.

fernandolcx, 2020-01-05:
Tried solis driver from master branch with my new Back-UPS 1500
but the readings seems unusable:

```
battery.charge: 100.0
battery.voltage: 25.6
device.mfr: Microsol
device.model: Solis 2.0
device.type: ups
driver.name: solis
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyACM0
driver.parameter.synchronous: no
driver.version: 2.7.4-735-gd56ac771
driver.version.internal: 0.67
input.frequency: 0.0
input.transfer.high: 300.0
input.transfer.low: 150.0
input.voltage: 200.1
output.current: 0.0
output.voltage: 206.2
ups.load: 0.0
ups.mfr: Microsol
ups.model: Solis 2.0
ups.status: OL
ups.temperature: 0.00
```

From SGM:
[image]

Is this related or is another problem?

fernandolcx, 2020-01-09:
As there is really something wrong with solis driver when using this model (0xBE), I've sniffed four packet samples and readings from SGM (official APC's software for brazillian Back-UPS models).

My C skills are very weak, can't fix this myself, but I hope this helps whoever.

---

Packet `be 78 72 b0 00 2e a4 95 15 29 30 14 00 00 00 00 01 00 89 16 49 60 60 ea fe`:

Input voltage: 207,0 V
Input current: 3,6 A
Input freq: 59,6 Hz

Output power: 681,9 VA
Output real power: 672,0 W
Output power factor: 0,98
Output voltage: 107,6 V
Output current: 6,3 A
Output freq: 59,6Hz

Battery voltage: 27,3V
Autonomy: 1min

Status: 
* Line ON
 
---
 
Packet `be 7d 73 b0 00 03 a4 ef 01 3b 35 14 00 00 00 00 01 00 89 16 49 60 60 22 fe`:

Input voltage: 209,0 V
Input current: 0,6 A
Input freq: 59,6 Hz
 
Output power: 105,0 VA
Output real power: 100,3 W
Output power factor: 0,95
Output voltage: 111,3 V
Output current: 0,9A
Output freq: 59,6Hz
 
Battery voltage: 27,3V
Autonomy: 60 min
 
 Status: 
* Line ON
 
 ---
 
Packet `be 19 00 9f 00 09 05 76 04 08 3a 14 00 00 00 00 01 00 89 16 6b 60 60 1f fe`

Input voltage: 0,0V
Input current: 0,0 A
Input freq: 0,0 Hz
 
Output power: 187,8 VA
Output real power: 178,0 W
Output power factor: 0,94
Output voltage: 114,5 V
Output current: 1,6 A
Output freq: 60,0 Hz
 
Battery voltage: 24,8V
Autonomy: 30 min

 Status: 
* Line OFF
* On battery
 
 ---

Packet `be 7d 74 ae 00 00 a0 00 00 15 07 15 00 00 00 00 01 00 89 16 4b 60 60 d9 fe`:

Input voltage: 211,0V
Input current: 0,0 A
Input freq: 59,6 Hz
 
Output power: 0,0 VA
Output real power: 0,0 W
Output power factor: 0,0
Output voltage: 112,3 V
Output current: 0,0 A
Output freq: 59,6 Hz
 
Battery voltage: 27,0V
Autonomy: 60 min

 Status: 
* Line ON
* charging batteries

oseiasslompo, 2020-01-17:
Hello guys! It really would be simply awesome if you guys corrected it. I've looking for a long time for a drive to rely on. I'm still using driver solis 2.0, but as you told, it's not working properly. There are a lot of people looking for that. I wish you the best luck.

fernandolcx, 2020-01-22:
[image]

Not sure if this helps, but opening SGM's classes in JD-GUI may be useful as source for corect protocol implementation.

rpvelloso, 2020-01-23:
@fernandolcx can you send me this decompiled code? As soon as I have som time I'll fix the driver. Thanks.

ygorre, 2021-03-18:
@rpvelloso @fernandolcx Can you send me a copy of this manufacturer source code, so I can try to make a fix? Using an APC BZ2200BI here.

rpvelloso, 2021-03-18:
@ygorre sure: https://drive.google.com/file/d/1QpuRddyCXwnjbAJFc4EIiu_grO0D6j3u/view?usp=sharing

ygorre, 2021-03-21:
I have submitted pull request #994 with a new driver (mostly based on solis.c code). The code was written to share code with solis.c, as many calculations and operations are the same for both, with only some of them (voltages, currents and real power) being model-specific.

I did it on a separate driver to avoid breakages on old, Microsol-branded, Solis models. This way, the newer, APC-branded, models should on the new driver.

Tested it on a BZ2200BI, with results looking correct (no more too high voltage or current reads, but no way to do proper testing here). Also tested its formulas with @fernandolcx data for BZ1500.