networkupstools/nut #699: New releases needed

robbat2, 2019-05-26:
It's been 3 years since the last release, and 700+ commits to the master branch.

Please issue a new formal release, esp. since the OpenSSL 1.1 fix was merged to master recently, and is now crucial on many distributions.

zoon01, 2019-06-07:
I like to see a new release version too!

not everyone/project can/wil use sources from master
For this stable releases are needed!

jimklimov, 2019-09-13:
Yes, maintained NUT is, actively developed also. There are several regular contributors and quite a few individuals submitting a handful of changes to "scratch their itch", in a good sense.

There is a semblance of a roadmap for the release (some of the issues are tagged as 2.7.5 goals), as well as quite a bit of development and cross-pollination in forks that bubble back up to this project as they mature in practical products (e.g. newer vendor devices support).

I agree that a formal new release would be quite welcome, since not only certain users but also many distributions (packagers) reasonably do not like to pick random stepping stones from third-party developments and want something that matter experts said is worthy of being named a stable release.

On my side, a large part of changes since last release were about the maintainability and quality of the project itself, something behind the curtains and not directly a functional end-user benefit (CI integrations, makefile dependency chains, spellchecks, various clean-ups and optimizations...)
Of course as time went there were some bug fixes and device support improvements. 
Something that users on Linux and Solaris/illumos systems can see, added integration for systemd and SMF services for more granular and robust setup than one init-script where either all works or something fails and then everything does. There are still some rough edges with this integration that are being gradually improved, hopefully committed code for this would get better before a formal release.

Something I look forward to seeing in common master as part of the release is the improved libusb support per https://github.com/networkupstools/nut/tree/libusb-1.0+0.1 - I believe it was not merged yet, but many of my builds on various OSes include those changes, e.g. they are practically required for USB UPS connections on modern Solaris-ish illumos (OpenIndiana and OmniOS) systems and the changeset looks pretty solid to me.

OpenSSL-1.1.0 is already in master by now.

DMF (loadable rather than statically compiled mappings for SNMP, eventually maybe other driver types) likely won't get in before the next release, mostly because it is a very big change for extensive review by independent contributors (some of us from the NUT team worked on it for a forked project so are sort of biased), and there are also still some unsolved architectural puzzles on fitting some nuances together neatly. But still I look forward to getting it into the main tree sometime, both for its own merits and also to simplify fork maintenance :-)

vitlav, 2019-12-10:
still wait when #486 will fixed...

zoon01, 2020-03-23:
Where are we standing now? 
jimklimov, are you able to give us some more light on this subject?

jimklimov, 2021-09-21:
Closing this issue in favor of #819 to track it in one place.

In short: there sadly were lags due to real life, and to CI systems changed in the middle of code clean-up effort, which protracted things. But we are getting closer, especially with a recent streak of "green" builds on new CI farm, that fix a number of long-standing warnings.

closed 2021-09-21 by jimklimov