networkupstools/nut #702: Best Fortress LI660 not respond

tomiisp, 2019-06-08:
Hi,

I have two of this serial UPS's, I configured it as: 
driver = bestfortress
baudrate = 9600
and seems they not respond to NUT and upsc show "Error: Driver not connected".

I try simple serial program in python:
port = serial.Serial("/dev/ttyUSB1", baudrate=9600, timeout=3.0)
port.write("f\r")
rcv = port.read(128)
print (rcv)

And respond is:
000000000001000000000000023302330000001000023300000264499000230033000000000101B9

This response is correct but not know why nut ups wont detect it. I quickly look in driver code and seems he awaiting this response but it doesnt work in real.
I tested also old APC SmartUPS600 on the same serial cable and it work without problem.

I running on:
nut-server/bionic,now 2.7.4-5.1ubuntu2 arm64
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Best Fortress UPS driver 0.05 (2.7.4)

Thanks for any help

/Tomi

closed 2019-06-08 by tomiisp

clepple, 2019-06-08:
I’m curious: how did you fix this? I was going to look later and see if we had a proposed fix for this on another branch.

tomiisp, 2019-06-08:
In truth I have no idea, it fixes itself somehow, I spent about 2 hours with trying to establish communication with this UPS and nothing, when I connect cable to APC and change config - APC works, if I switch back to Fortress - nothing - Error. Py script works always. Then I try run :
 ./bestfortress -a upsli660 -DDDDDD
and it start responding and also upsc see parameters.
(I also try to run windows port of nut, APC ok, Fortress no response but CheckUPS II works fine).

clepple, 2019-06-08:
If CheckUPS II works, it might need the “bestups” driver, superseded by “blazer_ser” and “nutdrv_qx” in recent versions of NUT.

clepple, 2019-06-08:
(Ignore my last comment - the other protocol for the Fortresses is more like contact-closure, and isn’t software selectable)