networkupstools/nut #706: usbhid-ups unable to disable beeper with Eaton 3S550

optimaldev, 2019-06-19:
Had a very similar issue as issue #9, and reading through the comments it would seem that it was resolved? I did however encounter the same issue:
Installed NUT on CentOS system, have everything configured properly and working. 
In /etc/ups/upsd.users I have

`[admin]
     password = PASS
     actions = SET
     instcmds = ALL`

and I am able to issue a beeper.disable command to the UPS device (Eaton 3S 550)
The device responds with `OK`
After issuing the `beeper.disable` command I query the UPS for `ups.beeper.status` 
The device responds with `enabled`
I then can validate that the beeper is still enabled by unplugging the device. 

I decided to try running one of the Eaton provided tools on a windows machine, and I was able to set the UPS Beeper to disabled from the Eaton UPS Companion tool. 

When returning the UPS usb device to my linux machine, the `ups.beeper.status` remains `enabled`, even though it is not. 

Not necessarily the most important bug, but thought I'd bring it to your attention

optimaldev, 2019-06-19:
Could this be related to the NUT package not being a recent release (I saw the other issue that was talking about a new formal release, and thought 'hmmm... that could be it maybe I should build from source to get the updated code...') 

not something that even occurred to me!

Thanks, 
Omar

clepple, 2019-06-20:
Probably related, since 2.7.5 has not been released yet. If you have a Debian- or Ubuntu-based system, we have instructions for building packages here: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

I don't remember off the top of my head if any of the RPM repositories build from the Git master branch, but you might want to check the mailing list archives if that applies to you. I seem to remember some discussion of prerequisites for building from a Git checkout on RedHat derivatives. https://networkupstools.org/support.html#_mailing_lists

optimaldev, 2019-06-20:
Thanks, I'm pretty certain that is the issue! Good luck with the release!

closed 2019-06-20 by optimaldev