networkupstools/nut #707: American Wise Power ARC2000RT

japurillo, 2019-06-21:
Hi All,

I tried just the drivers and can't seem to get it to work with any. From testing the Windows application itself, it looks like it works perfectly using UPSmart application on singlephase USB driver.

On a linux box, it displays an HID device on usb but I cant seem to get it to work. Anyone got this to work?

clepple, 2019-06-21:
I don't think we've seen that vendor before. As such, if you don't start the driver with "-u root", the default permissions on the USB /dev nodes won't allow the driver to see UPS. (We don't recommend running the driver as root all the time, but it's quicker and less error-prone to test that way than to change the udev files.)

What does `lsusb -v` return? (You can trim the output to just the UPS, and you can get more information from lsusb if you run it as root.) Also, the more information you can provide about the OS and version of NUT, the quicker we can help get things working.

clepple, 2019-06-21:
#674 mentions UPSmart

ClarkAlmazan, 2026-11-10:
Hello, I have a similar model to this one, an ARC1500RT. Here's the output for `lsusb -v`

```
Bus 001 Device 002: ID 0001:0000 Fry's Electronics ATCL FOR UPS
Negotiated speed: Low Speed (1Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000 ATCL FOR UPS
  bcdDevice            0.00
  iManufacturer           1 ATCL FOR UPS
  iProduct                1 ATCL FOR UPS
  iSerial                 1 ATCL FOR UPS
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      32
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```

ClarkAlmazan, 2026-11-11:
I was able to connect with these settings (based on [this comment](https://github.com/networkupstools/nut/issues/674#issuecomment-474840605))
```
[arc1500rt]
    driver = "nutdrv_qx"
    protocol = hunnox
    port = "auto"
    vendorid = "0001"
    productid = "0000"
    novendor
    langid_fix = "0x0409"
    desc = "ARC1500RT"
    noscanlangid
```

Output:
```
upsc arc1500rt
Init SSL without certificate database
battery.charge: 100
battery.voltage: 26.8
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.flag.noscanlangid: enabled
driver.flag.novendor: enabled
driver.name: nutdrv_qx
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.protocol: hunnox
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.state: quiet
driver.version: 2.8.1
driver.version.data: Hunnox 0.02
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.28 (API: 0x100010a)
input.current.nominal: 5.4
input.frequency: 60.0
input.frequency.nominal: 60
input.voltage: 237.0
input.voltage.fault: 0.0
input.voltage.nominal: 220
output.voltage: 204.0
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 0
ups.load: 14
ups.productid: 0000
ups.status: OL TRIM
ups.temperature: 37.0
ups.type: offline / line interactive
ups.vendorid: 0001
```

Not a lot of data, though I've yet to check in a while what UPSmart reports.

jimklimov, 2026-11-11:
@ClarkAlmazan : thanks for the report!