networkupstools/nut #710: Please use PKG_CONFIG environment variable instead of hardcoding pkg-config path/name

bigon, 2019-07-09:
Hello,

It seems that multiple/all m4 files are not using the PKG_CONFIG environment variable and are instead just using pkg-config from the path

AFAICT, this can break cross-compilation.

The build system should obey the PKG_CONFIG environment variable with the default to the one on the path

An other solution might be to use the PKG_CHECK_MODULES macro. I'm not sure that this is available on all supported platform

jimklimov, 2021-11-14:
Much of this should have got addressed recently with #1179 - at least, it should no longer use `pkg-config` verbatim (and assume it exists at all) but I'm not sure if `PKG_CONFIG` default from caller environment is used. This variable is used to carry the name of the program detected or optionally specified on command line (as seems more proper for `configure` script).

There were earlier PRs like #1007 (and search shows up more) that dealt with pkg-config and its absence on a build agent.

@bigon : can you please take a look whether current NUT master branch seems fit in this regard now?

jimklimov, 2022-03-12:
In recent search across the codebase, `pkg-config` string was found in comments, docs and sample code. Of near-production things, it was in `ci_build.sh` which should now be addressed in PR #1334

closed 2022-03-12 by jimklimov