networkupstools/nut #721: weird behaviour when switching PC's

tazhate, 2019-07-27:
I have ups dexp mix 850va.

Here is my config:

```
[ups1]
       desc="dexp"
       port = auto
       driver = blazer_usb
       default.battery.voltage.high = 2.27
       default.battery.voltage.low = 1.72
       runtimecal = 240,100,720,50
       langid_fix=0x0409
```
It is seems working ok:

```
root@tazthink:~# upsc ups1
Init SSL without certificate database
battery.charge: 100
battery.runtime: 2400
battery.voltage: 13.60
battery.voltage.high: 2.27
battery.voltage.low: 1.72
battery.voltage.nominal: 0.0
device.mfr: 
device.model: 
device.type: ups
driver.name: blazer_usb
driver.parameter.langid_fix: 0x0409
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.runtimecal: 240,100,720,50
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 0.0
input.frequency: 49.9
input.voltage: 224.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 226.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V3.7
ups.load: 13
ups.mfr: 
ups.model: 
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001
```

but when i put usb cable into another machine - all values become zeroes, reboots and restarts doesnt help. all configs and versions of the nut is the same, the only difference is that second PS have only usb 2 instead of usb 3 at machine where everything works ok.

nut version: 2.7.4-5.1ubuntu2

```
root@eeepc:~# upsc ups1
Init SSL without certificate database
battery.charge: 1
battery.runtime: 33
battery.voltage: 0.00
battery.voltage.high: 2.27
battery.voltage.low: 1.72
battery.voltage.nominal: 0.0
device.mfr: 
device.model: 
device.type: ups
driver.name: blazer_usb
driver.parameter.langid_fix: 0x0409
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.runtimecal: 240,100,720,50
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 0.0
input.frequency: 0.0
input.frequency.nominal: 0
input.voltage: 0.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 0.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 
ups.load: 0
ups.mfr: 
ups.model: 
ups.productid: 0000
ups.status: OL
ups.temperature: 0.0
ups.type: online
ups.vendorid: 0001
```

is there any way to fix it?

zykh, 2019-07-30:
@tazhate: could you please post the logs of the `blazer_usb` driver launched with a debug level of 3? (i.e. `-DDD`)
Also, do you notice any difference with the `nutdrv_qx` driver?

tazhate, 2019-07-30:
```
root@eeepc:~# /lib/nut/blazer_usb -DDD -a ups1
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
   0.000000	debug level is '3'
   0.004334	language ID workaround enabled (using '0x409')
   0.115898	Checking device (1D6B/0001) (005/001)
   0.143599	- VendorID: 1d6b
   0.143670	- ProductID: 0001
   0.143701	- Manufacturer: unknown
   0.143732	- Product: unknown
   0.143764	- Serial Number: unknown
   0.143796	- Bus: 005
   0.143827	- Device release number: 0415
   0.143859	Trying to match device
   0.143896	Device does not match - skipping
   0.143953	Checking device (1D6B/0001) (004/001)
   0.171706	- VendorID: 1d6b
   0.171784	- ProductID: 0001
   0.171814	- Manufacturer: unknown
   0.171843	- Product: unknown
   0.171872	- Serial Number: unknown
   0.171902	- Bus: 004
   0.171930	- Device release number: 0415
   0.171960	Trying to match device
   0.171995	Device does not match - skipping
   0.172056	Checking device (1D6B/0001) (003/001)
   0.199652	- VendorID: 1d6b
   0.199724	- ProductID: 0001
   0.199753	- Manufacturer: unknown
   0.199781	- Product: unknown
   0.199809	- Serial Number: unknown
   0.199837	- Bus: 003
   0.199866	- Device release number: 0415
   0.199894	Trying to match device
   0.199925	Device does not match - skipping
   0.199982	Checking device (0001/0000) (002/003)
   0.211976	- VendorID: 0001
   0.212063	- ProductID: 0000
   0.212093	- Manufacturer: unknown
   0.212123	- Product: MEC0003
   0.212152	- Serial Number: unknown
   0.212177	- Bus: 002
   0.212208	- Device release number: 0100
   0.212238	Trying to match device
   0.212271	Device matches
   0.212467	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.216937	First supported language ID: 0x409 (please report to the NUT maintainer!)
   0.217060	Trying megatec protocol...
   0.217098	send: Q1
   0.233946	received 96 (96)
   0.234029	read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.234197	Status read in 1 tries
   0.234227	Supported UPS detected with megatec protocol
   0.234255	send: F
   0.243919	received 46 (46)
   0.243995	read: #000.0 0.0 00.00 00.0
   0.244087	Ratings read in 1 tries
   0.244121	send: I
   0.258911	received 80 (80)
   0.258995	read: #                                     
   0.259069	Vendor information read in 1 tries
   0.259156	battery runtime exponent : 1.585
   0.259201	battery runtime nominal  : 240.0
   0.259276	battery runtime estimate : 0.0
   0.259312	No charge time specified, using built in default [43200 seconds]
   0.259562	No idle load specified, using built in default [10.0 %]
   0.259670	send: Q1
   0.275917	received 96 (96)
   0.276001	read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.276270	dstate_init: sock /var/run/nut/blazer_usb-ups1 open on fd 5
[... 191 lines cut ...]
   0.532344	qx_initbattery: battery runtime estimate: 0.0
   0.532552	No charge time specified, using built in default [43200 seconds]
   0.533149	No idle load specified, using built in default [10.0 %]
   0.533375	upsdrv_updateinfo...
   0.533761	Quick update...
   0.533962	send: Q1
   0.551171	received 96 (96)
   0.551388	read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.552048	dstate_init: sock /var/run/nut/nutdrv_qx-ups1 open on fd 5
   0.552581	upsdrv_updateinfo...
   0.552905	Quick update...
   0.553345	send: Q1
   0.570139	received 96 (96)
   0.570349	read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   2.552703	upsdrv_updateinfo...
   2.552774	Quick update...
   2.552805	send: Q1
   2.569019	received 96 (96)
   2.569133	read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   4.553811	upsdrv_updateinfo...
   4.553891	Quick update...
   4.553933	send: Q1
   4.570921	received 96 (96)
   4.571048	read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
```

zykh, 2019-07-30:
hmm...
a few questions:
1. if you replug the UPS to PC 1 (the one where data was retrieved correctly), does the driver get the missing data again?
2. did PC 1 have perhaps the original software running before launching NUT?
3. using `nutdrv_qx` from #638 does any difference?

tazhate, 2019-08-01:
1. yeah, after some time of uptime
2. yeah, but it is not "original". it is upsmart, in windows vm. 
3. i'll check, thanks

closed 2022-04-25 by jimklimov

CipherGeek, 2022-07-30:
Hello.

I faced a similar problem on debian 7 (NUT version 2.6.4-2.3), recently fixed it on debian 9 (NUT version 2.7.4-5.1) in the same way, but with an additional patch that allows you to build the project.

I had the following problems:
1) the values ​​were shown only 1 time, and became static, which changed only after rebooting the PC and sometimes when removing and inserting the USB cable back;
2) when trying to run a self-written script to test for receiving values ​​from the UPS, sometimes a data reading error occurred on the port (in a script that was also not tailored for my UPS) due to the length of the buffer;
3) the values ​​were sometimes shown as zeros, which were updated to real ones, but also static ones when the script was launched (1-2 times) from point 2.

The problem is that the entire line of UPS from DEXP operates on a buffer length of 102 bytes, while NUT operates on 512 bytes. Both drivers suggested here did not help me.

Changing the buffer length to 102 bytes completely solved all the problems. NUT instantly picks up the transition to the battery and any changes in external power.

My configuration ups.conf:
`[dexp]`
	`driver = blazer_usb`
	`port = auto`
	`desc = "DEXP IEC LCD ONLINE 3000VA"`
	`vendorid = 0001`
	`productid = 0000`
	`override.battery.charge.low = 10`
	`default.battery.voltage.low = 63.6`
	`default.battery.voltage.high = 81.7`
	`override.battery.voltage.nominal = 72`
	`override.battery.packs = 36`
	`chargetime = 43200`
	`idleload = 1%`
	`langid_fix = 0x0409`

My patches:
[nut_2.7.4-5.1_(debian9.12-stretch)-patches.zip](https://github.com/networkupstools/nut/files/9225631/nut_2.7.4-5.1_.debian9.12-stretch.-patches.zip)

commit 2b239931 (2022-04-25): driver.list.in: Add HCL info for DEXP MIX 850VA (USB)

Closes: #721