networkupstools/nut #723: upsd dies if upsd.conf contains LISTEN line with nonexistant ip

bigon, 2019-07-27:
The following bug has been open in debian (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=926329)

===
Dear Maintainer,

After reconfiguring 1 network interface to a new range, but forgetting to remove the last line, then one with 192.168.123.9 from /etc/nut/upsd.conf

LISTEN 127.0.0.1 3493
LISTEN 192.168.119.253 3493
LISTEN 192.168.123.9 3493

The nut daemon refused to run after it at a later date was restarted. Syslog just reported:

Apr  2 20:54:11 dkfitlet upsd[16941]: Network UPS Tools upsd 2.7.4
Apr  2 20:54:11 dkfitlet upsd[16941]: not listening on 192.168.123.9 port 3493
Apr  2 20:54:11 dkfitlet upsd[16941]: listening on 192.168.119.253 port 3493
Apr  2 20:54:11 dkfitlet upsd[16941]: listening on 127.0.0.1 port 3493
Apr  2 20:54:11 dkfitlet upsd[16941]: no listening interface available

Notice the last line message: "no listening interface available" which is not true, as 2 out of 3 configured LISTEN addresses was available. Notice also that syslog does not say that upsd exitted.

After removing the bad line from upsd.conf "LISTEN 192.168.123.9 3493" everything started working fine.

I would expect that upsd listens to the addresses it can find and not just exit.
===

lienrag35, 2023-11-04:
I had the same issue and find why it refused to start :) of this can be fixed

jimklimov, 2023-11-04:
@lienrag35 : which NUT version?

jimklimov, 2023-11-04:
Ok, can confirm with current too:
````
Nov 04 17:59:14 pve nut-server[770211]: not listening on 12.34.56.78 port 3493
Nov 04 17:59:14 pve nut-server[770211]: listening on 127.0.0.1 port 3493
Nov 04 17:59:14 pve nut-server[770211]: not listening on 1.2.3.4 port 3493
Nov 04 17:59:14 pve nut-server[770211]: no listening interface available
````

And curiously (and inconsistently) if a working address is first, it is "okay"; if a failing one is first it aborts:
````
Nov 04 18:00:32 pve nut-server[790607]: Could not find PID file '/var/state/ups/upsd.pid' to see if previous upsd instance is already running!
Nov 04 18:00:32 pve nut-server[790607]: listening on 127.0.0.1 port 3493
Nov 04 18:00:32 pve nut-server[790607]: not listening on 1.2.3.4 port 3493
Nov 04 18:00:32 pve nut-server[790607]: Connected to UPS [eco650]: usbhid-ups-eco650
````

vs.
````
Nov 04 18:00:16 pve nut-server[788715]: Could not find PID file '/var/state/ups/upsd.pid' to see if previous upsd instance is already running!
Nov 04 18:00:16 pve nut-server[788715]: not listening on 12.34.56.78 port 3493
Nov 04 18:00:16 pve nut-server[788715]: listening on 127.0.0.1 port 3493
Nov 04 18:00:16 pve nut-server[788715]: no listening interface available
````

Not fully sure, pedantically, which one is right (e.g. from security standpoint). What do apache/nginx/etc. do?

jimklimov, 2023-11-04:
I picked asking the audience :) (Mailing list)

lienrag35, 2023-11-04:
> @lienrag35 : which NUT version?

2.8.0

jimklimov, 2023-11-04:
> I picked asking the audience :) (Mailing list)

https://alioth-lists.debian.net/pipermail/nut-upsuser/2023-November/013479.html

jimklimov, 2023-11-05:
So, it seems there are basically two approaches to this (NUT's non-deterministic one is wrong, better to ensure one of these two happens):
* Listen to whatever interfaces we can bind to, as long as at least one is okay - proceed;
* Require all listed interfaces to be usable (they were listed for a reason, right? not listening on some would get those clients not served, which is a problem best detected and fixed early).

There are corner cases though:
* Listen on `*` "any address" (IPv4 and/or IPv6) AND on specific address+port endpoints - e.g. per https://forum.nginx.org/read.php?2,293645,293649#msg-293649 nginx apparently allows to proceed if specific-address listeners failed, but a catch-all on the same port succeeded. This may mean that some other program would respond on that failed address+port (same situation is possible and sometimes desired when one program listens on `*:PORT` only, and another listens on specific `IP:PORT`). 
* Listen on string-named `localhost` (resolve it? or hardcode to `127.0.0.1` and/or `::1`?)
* See also #2012 and #2013 regarding recent NUT progress in this area.

jimklimov, 2023-11-06:
Any takers here to walk the walk? ;)

* [ ] Refine tracking how many IP:Port sockets we have requested vs. actually managed to listen on, regardless of who was first in success/fail state
* [ ] Clarify the emitted messages, if needed
* [ ] Support some way of configuration to switch between "B" (default) and "A" behaviors

lienrag35, 2023-11-06:
B with an explicit message as for example, some listeners are invalid

jimklimov, 2023-11-06:
I am pressed on time lately, got tech debts in dayjob after recent crunch to release NUT - so would rather review and direct a PR than post one. At least this change should be relatively straightforward, of the "99% perspiration and 1% inspiration" type :)

jimklimov, 2023-11-09:
Oh well, PR posted :)

PR #2160 (2023-11-13): upsd: introduce support for `ALLOW_NOT_ALL_LISTENERS` configuration toggle
...and rectify condition checks for "no listening interface available" diagnosis in any case.

Closes: #723
closed 2023-11-13 by jimklimov