networkupstools/nut #726: Suffix rules in Makefile

bigon, 2019-07-28:
Hello,

If I'm not wrong, the Makefiles in docs/ are using [Suffix rules](https://www.gnu.org/software/make/manual/html_node/Suffix-Rules.html)

Something like:
```Makefile
txt.html: common.xsl xhtml.xsl
    foo
```

If I understand the documentation well, you cannot add prerequisites to such rules, documentation says: `Suffix rules cannot have any prerequisites of their own. If they have any, they are treated as normal files with funny names, not as suffix rules.`

These rules can be changed to use pattern rules, but I _think_ they are GNU make only

zykh, 2019-07-30:
Unless you're using a not-yet-released version of GNU make which incorporates [this commit](http://git.savannah.gnu.org/cgit/make.git/commit/?id=67e6d81256486cbf160208c5b7ea164964c49dd4), apparently, they are supported since version 3.58 (see also http://savannah.gnu.org/bugs/?40657).
I guess we should fix them.

jimklimov, 2021-09-21:
Recent PR #1104 should have addressed it the other way around, to improve portability of the Makefiles (tested with FreeBSD 12's make and GNU make). Use of suffix rules seems to preclude use of prerequisites; I added the wildcard-rule prereqs just in case but not sure they do fire.

On the other hand, for these few cases, the referenced resources are not something that changes every year...

closed 2021-11-07 by jimklimov