networkupstools/nut #731: Battery.runtime maximum at 65535

NightDragon1, 2019-09-20:
Hi!

I'm using a Dell UPS 2700W with an extended Battery Modul (EBM).
As i have a quite low load on the UPS (about 5%) it's easy to reach a very high runtime (in sec)

For now it seems like i reached the max of a 16Bit variable.
Is this limit given by NUT? if so, could you please increase it?
Thanks!,

Rob

PS: Here some more data about the UPS:

```
battery.capacity: 27.00
battery.charge: 100
battery.charge.low: 30
battery.charge.restart: 0
battery.charger.status: resting
battery.energysave: no
battery.runtime: 65535
battery.runtime.low: 180
battery.type: PbAc
device.mfr: DELL
device.model: Dell UPS Rack/Tower 2700W HV - 230V
device.serial: CN-0J727N-(rest removed)
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 007
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Dell UPS Rack/Tower 2700W HV - 230V
driver.parameter.productid: FFFF
driver.parameter.serial: CN-0J727N-(rest removed)
driver.parameter.synchronous: no
driver.parameter.vendor: DELL
driver.parameter.vendorid: 047C
driver.version: 2.7.4
driver.version.data: MGE HID 1.39
driver.version.internal: 0.41
input.frequency: 49.9
input.frequency.nominal: 50
input.transfer.boost.low: 210
input.transfer.high: 286
input.transfer.low: 160
input.transfer.trim.high: 250
input.voltage: 229.0
input.voltage.nominal: 230
outlet.1.autoswitch.charge.low: 0
outlet.1.delay.shutdown: 2147483647
outlet.1.delay.start: 0
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.autoswitch.charge.low: 0
outlet.2.delay.shutdown: 2147483647
outlet.2.delay.start: 1
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 2
outlet.2.status: on
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.current: 0.60
output.frequency: 49.9
output.frequency.nominal: 50
output.voltage: 229.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.date: 2019/09/20
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 01.12.0001
ups.load: 5
ups.load.high: 110
ups.mfr: DELL
ups.model: Dell UPS Rack/Tower 2700W HV - 230V
ups.power: 138
ups.power.nominal: 2700
ups.productid: ffff
ups.realpower: 132
ups.realpower.nominal: 2700
ups.serial: CN-0J727N-(rest removed)
ups.shutdown: enabled
ups.start.battery: yes
ups.start.reboot: yes
ups.status: OL
ups.test.interval: 7776000
ups.test.result: Aborted
ups.time: 13:18:30
ups.timer.shutdown: 0
ups.timer.start: 0
ups.type: offline / line interactive
ups.vendorid: 047c
```

clepple, 2019-09-20:
> ups.test.result: Aborted
NUT gets the size of the variables from the USB HID descriptor. The 65535 is probably intended to be a 16-bit signed value of -1 for "unknown". I would recommend running a battery test, and seeing if the "ups.test.result" variable changes.

NightDragon1, 2019-09-27:
It does change as i have seen it loading the batteries and reaching the limit.
By now i run a test:
```
battery.capacity: 27.00
battery.charge: 97
battery.charge.low: 30
battery.charge.restart: 0
battery.charger.status: discharging
battery.energysave: no
battery.runtime: 22617
battery.runtime.low: 180
battery.type: PbAc
device.mfr: DELL
device.model: Dell UPS Rack/Tower 2700W HV - 230V
device.serial: CN-0J727N-(rest removed)
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 007
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Dell UPS Rack/Tower 2700W HV - 230V
driver.parameter.productid: FFFF
driver.parameter.serial: CN-0J727N-(rest removed)
driver.parameter.synchronous: no
driver.parameter.vendor: DELL
driver.parameter.vendorid: 047C
driver.version: 2.7.4
driver.version.data: MGE HID 1.39
driver.version.internal: 0.41
input.frequency: 50.0
input.frequency.nominal: 50
input.transfer.boost.low: 210
input.transfer.high: 286
input.transfer.low: 160
input.transfer.trim.high: 250
input.voltage: 228.0
input.voltage.nominal: 230
outlet.1.autoswitch.charge.low: 0
outlet.1.delay.shutdown: 2147483647
outlet.1.delay.start: 0
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.autoswitch.charge.low: 0
outlet.2.delay.shutdown: 2147483647
outlet.2.delay.start: 1
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 2
outlet.2.status: on
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.current: 1.20
output.frequency: 50.1
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.date: 2019/09/27
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 01.12.0001
ups.load: 10
ups.load.high: 110
ups.mfr: DELL
ups.model: Dell UPS Rack/Tower 2700W HV - 230V
ups.power: 278
ups.power.nominal: 2700
ups.productid: ffff
ups.realpower: 245
ups.realpower.nominal: 2700
ups.serial: CN-0J727N-(rest removed)
ups.shutdown: enabled
ups.start.battery: yes
ups.start.reboot: yes
ups.status: OL DISCHRG
ups.test.interval: 7776000
ups.test.result: In progress
ups.time: 20:56:31
ups.timer.shutdown: 0
ups.timer.start: 0
ups.type: offline / line interactive
ups.vendorid: 047c
```

clepple, 2020-06-22:
Closing since test seems to have fixed the reported value.

closed 2020-06-22 by clepple

NightDragon1, 2020-06-22:
After i saw you closed the bugfix I tried to investigate a bit more. Result: If the UPS load is below or equal to 10%, it does not calculate the runtime properly. There is also a paragraph in the manual saying, that a battery test can only be done with a load above 10%. However, last time it still started properly. So we can assume it's a Firmware issue.

**Some data**
`TIME:16:52:22
battery.runtime: 65535
battery.runtime.low: 180
ups.load: 7
ups.realpower: 161
TIME:16:52:24
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:26
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:28
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:30
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:32
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:34
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:36
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:38
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:40
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:42
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:44
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:46
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:48
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:50
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:53
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:55
battery.runtime: 18358
battery.runtime.low: 180
ups.load: 11
ups.realpower: 276
TIME:16:52:57
battery.runtime: 65535
battery.runtime.low: 180
ups.load: 10
ups.realpower: 260
TIME:16:52:59
battery.runtime: 65535
battery.runtime.low: 180
ups.load: 10
ups.realpower: 260
TIME:16:53:01
battery.runtime: 65535
battery.runtime.low: 180
ups.load: 10
ups.realpower: 260`

NightDragon1, 2020-06-22:
***Some addon...***

Once the UPS is online again, it seems to do some calulations, but I doubt one can trust the values.

`battery.runtime: 65181
battery.runtime.low: 180
ups.load: 6
ups.realpower: 133
`